stack-of-tasks/pinocchio #1633: Cannot install through pip with pip 22.0.4 & python 3.9

leo4life2, 2022-04-11:
I'm getting the following error message:

`ERROR: Could not find a version that satisfies the requirement pin (from versions: none)`
`ERROR: No matching distribution found for pin`

nim65s, 2022-04-11:
Hi @leo4life2,

I don't understand your error.

This is working for me, in a clean docker with python 3.9 and pip 22.0.4
```Dockerfile
FROM python:3.9
RUN python -m pip install pin
```

What are your setuptools and wheel versions ?

I have the following
```bash
root@2345b293f9da:/# pip freeze --all
eigenpy==2.6.9
hpp_fcl==1.7.8
numpy==1.22.3
pin==2.6.4
pip==22.0.4
setuptools==58.1.0
wheel==0.37.1
```

nim65s, 2022-04-11:
Oh, wait, what is your operating system ?

leo4life2, 2022-04-11:
Sorry, just saw the line saying that pip only works for Linux. I'm using a mac and I'll install with conda.

leo4life2, 2022-04-11:
Tried installing with conda, but it didn't work and was stuck with the following messages (conda 4.10.3):
```
conda install pinocchio -c conda-forge
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: - failed
```

Then I tried installing with brew and brew tap failed:

```brew tap gepetto/homebrew-gepetto
==> Tapping gepetto/gepetto
Cloning into '/usr/local/Homebrew/Library/Taps/gepetto/homebrew-gepetto'...
remote: Enumerating objects: 745, done.
remote: Counting objects: 100% (80/80), done.
remote: Compressing objects: 100% (55/55), done.
remote: Total 745 (delta 32), reused 68 (delta 25), pack-reused 665
Receiving objects: 100% (745/745), 220.55 KiB | 2.00 MiB/s, done.
Resolving deltas: 100% (347/347), done.
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/gepetto/homebrew-gepetto/Formula/gepetto-viewer-corba.rb
gepetto-viewer-corba: Invalid bottle tag symbol
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/gepetto/homebrew-gepetto/Formula/omniorbpy.rb
omniorbpy: undefined method 'cellar' for #<BottleSpecification:0x00007fafc2866c48>
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/gepetto/homebrew-gepetto/Formula/eigenpy@2.rb
eigenpy@2: Invalid bottle tag symbol
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/gepetto/homebrew-gepetto/Formula/gepetto-viewer.rb
gepetto-viewer: Invalid bottle tag symbol
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/gepetto/homebrew-gepetto/Formula/osgqt.rb
osgqt: undefined method 'cellar' for #<BottleSpecification:0x00007fafc28bc1e8>
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/gepetto/homebrew-gepetto/Formula/open-scene-graph-with-colladadom.rb
open-scene-graph-with-colladadom: Invalid bottle tag symbol
Error: Cannot tap gepetto/gepetto: invalid syntax in tap!

nim65s, 2022-04-11:
wheels for mac will be available, eventually, but not yet, sorry.

And I can't help with conda :/

closed 2022-04-11 by leo4life2