stack-of-tasks/pinocchio #1634: Can the input of pinocchio.forwardKinematics() be a set of joint angles?

WangXingfang, 2022-04-12:
For a 6-DOF robot, we have a set of joint angles as follows.
q_all = np.array([[0, 0, 0, 0, 0, 0],[1, 1, 1, 1, 1, 1],[2, 2, 2, 2, 2, 2]]), which is a 3x6 matrix.
So can we take “q_all” directly as an input “q” to pinocchio.forwardKinematics(model, data, q)? 
Because the method used _for_ loop is time-consuming, let's say "q_all" is a 1000* 6 matrix.

jcarpent, 2022-04-12:
Batch computation is not yet supported by Pinocchio. What is your application?

jcarpent, 2022-04-13:
What is the application for? Which computations do you want to perform after this batch computation?

WangXingfang, 2022-04-13:
> What is the application for? Which computations do you want to perform after this batch computation?
I want to use it in the reinforcement learning application of 7-DOF robot. For example, input 1000 joint angle samples to calculate their forward kinematics. If the _for_ loop is used for serial calculation, it takes a long time. Therefore, I want to use parallel computing to improve computing efficiency.

WangXingfang, 2022-04-13:
> What is the application for? Which computations do you want to perform after this batch computation?

I want to use it in the reinforcement learning application of 7-DOF robot. For example, input 1000 joint angle samples to calculate their forward kinematics. If the for loop is used for serial calculation, it takes a long time. Therefore, I want to use parallel computing to improve computing efficiency.

WangXingfang, 2022-04-15:
> 

Thanks for your advice! 
I'm working in Python. And I will consider the multi-threading computation.

WangXingfang, 2022-04-15:
> Small question: what do you expect from the forward kinematics results? You can see to the pool structure implemented in Pinocchio to use them for multi-threading evaluation.
> 
> Do you work in C++?

Thanks for your advice!
I'm working in Python. And I will consider the multi-threading computation.

closed 2022-04-15 by WangXingfang