stack-of-tasks/pinocchio #1638: Fixed joint in URDF seems being merged when loading model

DavidYaonanZhu, 2022-04-26:
Fixed joint in URDF seems to be merged when loading model.

I loaded a tiago dual arm robot URDF, 

`model    = pinocchio.buildModelFromUrdf(self.urdf_filename)`
then,
`model.njoints`
it gives 34,
however If I,
`model.getJointId("arm_left_tool_joint")`

**some of the joints give the same ID.**

In addition, does this library support Jacobian acquisition of different base_link and tool_link, like KDL?

jcarpent, 2022-04-26:
> loading model (Issue #1638)

Could you provide a full example? I will provide a complete answer then

jcarpent, 2022-04-26:
> In addition, does this library support Jacobian acquisition of different base_link and tool_link, like KDL?

I don't the API of KDL, but for sure, you can retrieve the Jacobian of any link with Pinocchio.

DavidYaonanZhu, 2022-04-26:
> 

How to do that? can you give me some examples?

DavidYaonanZhu, 2022-04-26:
Yes, can you give some hints on how to do that indirectly?

DavidYaonanZhu, 2022-04-26:
I will update information of KDL tool shortly

jcarpent, 2022-04-26:
> 

You just need to collect the column indexes which are related to the joints supporting the path between the two links in questions
For a given joint, you can retrieve its starting index and the size of the motion subspace by using `model.joints[joint_id].idx_v, model.joints[joint_id].nv`