stack-of-tasks/pinocchio #1639: Mesh package could not be found

Vassil17, 2022-04-26:
Hello,

I've been trying to get pinocchio (and gepetto-viewer) to work on a custom robot model (that of the Go1), but while the URDF gets loaded without a problem, when I try loading the mesh model get the following error: 
```
ValueError: Mesh package://go1_description/meshes/trunk.dae could not be found.
```
For reference, I have sourced everything and added the "/opt/openrobots/share" to the ROS_PACKAGE_PATH. The other robots in the example-robot-data package load just fine. The meshes/ folder also contains all the necessary files (I believe):
```
ls /opt/openrobots/share/example-robot-data/robots/go1_description/meshes
>> calf.dae  depthCamera.dae  hip.dae  thigh.dae  thigh_mirror.dae  trunk.dae  ultraSound.dae
```

Do you know what could be causing this or how I could fix it? Is there anything else I need to set up to make it recognise the mesh files? Thanks!

jcarpent, 2022-04-26:
You should export ROS_PACKAGE_PATH as `/opt/openrobots/share/example-robot-data/robots` or set 'mesh_dir=/opt/openrobots/share/example-robot-data/robots`

closed 2022-04-26 by jcarpent

cmastalli, 2022-04-26:
> Thanks, I'll take a look at that!
> 
> @cmastalli As a disclaimer I'm just a master's student working with the Go1, however I would still be happy to help!

That's fine. We are happy to review your PR.