stack-of-tasks/pinocchio #1641: Compilation issue with OSX Monterey M1

cmastalli, 2022-04-26:
Hi guys,

I am starting to move all my development to an OSX Monterey PC, and I am still learning how to do it.
I wish to compile Pinocchio from source, and this is the error that I encountered
```bash
/Users/cm2099/Devel/pinocchio/build/include/pinocchio/parsers/urdf/types.hpp:31:3: error: typedef redefinition with different types ('boost::shared_ptr<Box>' vs 'std::shared_ptr<Box>')
  PINOCCHIO_URDF_TYPEDEF_CLASS_POINTER(Box);
  ^
/Users/cm2099/Devel/pinocchio/build/include/pinocchio/parsers/urdf/types.hpp:25:42: note: expanded from macro 'PINOCCHIO_URDF_TYPEDEF_CLASS_POINTER'
/Users/cm2099/Devel/pinocchio/src/parsers/urdf/model.cpp:11:10: typedef PINOCCHIO_URDF_SHARED_PTR(Class) Class##SharedPtr; \
                                         ^
<scratch space>:174fatal error: :1:'urdf_parser/urdf_parser.h' file not found
note: expanded from here
BoxSharedPtr
#include <urdf_parser/urdf_parser.h>^

         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/homebrew/include/urdf_model/types.h:53:1: note: previous definition is here
URDF_TYPEDEF_CLASS_POINTER(Box);
^
/opt/homebrew/include/urdf_model/types.h:44:32: note: expanded from macro 'URDF_TYPEDEF_CLASS_POINTER'
typedef std::shared_ptr<Class> Class##SharedPtr; \
                               ^
<scratch space>:126:1: note: expanded from here
BoxSharedPtr
^
In file included from /Users/cm2099/Devel/pinocchio/src/parsers/urdf/geometry.cpp:6:
/Users/cm2099/Devel/pinocchio/build/include/pinocchio/parsers/urdf/types.hpp:31:3: error: typedef redefinition with different types ('boost::shared_ptr<const Box>' vs 'std::shared_ptr<const Box>')
  PINOCCHIO_URDF_TYPEDEF_CLASS_POINTER(Box);
  ^
/Users/cm2099/Devel/pinocchio/build/include/pinocchio/parsers/urdf/types.hpp:26:48: note: expanded from macro 'PINOCCHIO_URDF_TYPEDEF_CLASS_POINTER'
typedef PINOCCHIO_URDF_SHARED_PTR(const Class) Class##ConstSharedPtr; \
                                               ^
<scratch space>:175:1: note: expanded from here
BoxConstSharedPtr
^
/opt/homebrew/include/urdf_model/types.h:53:1: note: previous definition is here
URDF_TYPEDEF_CLASS_POINTER(Box);
^
/opt/homebrew/include/urdf_model/types.h:45:38: note: expanded from macro 'URDF_TYPEDEF_CLASS_POINTER'
typedef std::shared_ptr<const Class> Class##ConstSharedPtr; \
```

I have installed Boost from source, since the homebrew formulae doesn't work well.
I have also installed `hpp-fcl` and `eigenpy` from source.

I have installed the following packages using homebrew:
  - octomap
  - urdfodom_headers

Finally, I install Python 3.10 from python.org

Please let me know what I am doing wrong.

jcarpent, 2022-04-26:
Please use conda, it works pretty well for Mac M1.

closed 2022-04-26 by jcarpent

traversaro, 2022-04-26:
I think the problem is related the issue https://github.com/ros/urdfdom_headers/issues/73 . Basically recent versions of urdfdom/urdfdom_headers changed the location where headers are installed, but that breaks existing binaries with cmake config files that expect the headers in a different location.

jcarpent, 2022-04-26:
> The issue is with urdfdom_headers. I installed it from conda, and it went away.
> 
> But I am still curious to understand deeply this problem. Could you tell me if there is a particular version restrictions with Pinocchio?

From your log, your are using the one from brew.

jcarpent, 2022-04-26:
You can install pinocchio for M1 with conda without any problem

cmastalli, 2022-04-26:
> > The issue is with urdfdom_headers. I installed it from conda, and it went away.
> > But I am still curious to understand deeply this problem. Could you tell me if there is a particular version restrictions with Pinocchio?
> 
> From your log, your are using the one from brew.

Yes, I was indeed using it.
And the issue still stands with brew.

 But now I sorted it out using the Conda one.