stack-of-tasks/pinocchio #1642: Compilation failure in unit tests

bulvara, 2022-05-03:
I'm using GCC 12 and boost 1.78:
```
./pinocchio-2.6.7/unittest/cppad-joint-configurations.cpp:86:26:   von hier erfordert
/usr/include/boost/math/constants/constants.hpp:259:3: Fehler: keine passende Funktion für Aufruf von »boost::math::constants::detail::constant_pi<CppAD::AD<double> >::get(boost::math::constants::construction_traits<CppAD::AD<double>, boost::math::policies::policy<boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy, boost::math::policies::default_policy> >::type)«
  259 |   BOOST_DEFINE_MATH_CONSTANT(pi, 3.141592653589793238462643383279502884e+00, "3.14159265358979323846264338327950288419716939937510582097494459230781640628620899862803482534211706798214808651e+00")
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~

```

jcarpent, 2022-05-04:
What is your OS?

jcarpent, 2022-05-04:
@proyan could you handle this issue please?

proyan, 2022-05-04:
Could you post the full compile log?

jcarpent, 2022-05-06:
I will close this issue as there is no answer from the reporter.

closed 2022-05-06 by jcarpent