stack-of-tasks/pinocchio #1644: AttributeError: module 'pinocchio' has no attribute 'rpy'

Senwei-Huang, 2022-05-06:
代码：
`R_HB = pin.rpy.rpyToMatrix(rpy)`

报错：
```
Traceback (most recent call last):
  File "E:\虚拟机\PaddleRobotics-main\QuadrupedalRobots\ETGRL\RRP_Hopper_simulation-main\RRP_Hopper.py", line 397, in <module>
    a1_sim.run()
  File "E:\虚拟机\PaddleRobotics-main\QuadrupedalRobots\ETGRL\RRP_Hopper_simulation-main\RRP_Hopper.py", line 358, in run
    tau_d = self.robot_SLIP_controller(vel_d, dir_d, F_thrust)
  File "E:\虚拟机\PaddleRobotics-main\QuadrupedalRobots\ETGRL\RRP_Hopper_simulation-main\RRP_Hopper.py", line 259, in robot_SLIP_controller
    tau_d = self.controller.solve(self.q_state, self.dq_state, 
  File "E:\虚拟机\PaddleRobotics-main\QuadrupedalRobots\ETGRL\RRP_Hopper_simulation-main\controller.py", line 88, in solve
    R_HB = pin.rpy.rpyToMatrix(rpy)
AttributeError: module 'pinocchio' has no attribute 'rpy'
```

环境：
```
window 10
python 3.9
pinocchio 0.4.3
```

@jcarpent
@proyan

jcarpent, 2022-05-06:
Where did you get pinocchio?
I don't think you are using the right way to install it: `pip install pin` as Pinocchio corresponds to another project.

Senwei-Huang, 2022-05-06:
噢噢，我知道了，是我安装错误了，我是这样安装的
`pip install pinocchio`

window系统不能安装python版本的吗？我这样安装不行诶
`pip install pin`

jcarpent, 2022-05-06:
Please use conda for installing Python bindings of Pinocchio for windows. We do not support yet pip on Windows.

closed 2022-05-06 by jcarpent

Senwei-Huang, 2022-05-06:
Thank you again. I used this to solve the problem
`conda install pinocchio -c conda-forge`
@jcarpent