stack-of-tasks/pinocchio #1647: Log3 corner case bug

jmirabel, 2022-05-09:
Hi,

I encountered what looks like a bug in the `log3` implementation to me.

```python
import pinocchio
import numpy as np

np.set_printoptions(3)

R = np.array([[0.0, -1.0, 0.0], [-1.0, 0.0, 0.0], [0.0, 0.0, -1.0]])
R2 = pinocchio.exp3(pinocchio.log3(R))

print("rotation matrix")
print("original", R)
print("reconstructed", R2)
```
which gives
```
rotation matrix
original [[ 0. -1.  0.]
 [-1.  0.  0.]
 [ 0.  0. -1.]]
reconstructed [[ 0.00e+00  1.00e+00 -8.66e-17]
 [ 1.00e+00  0.00e+00  8.66e-17]
 [ 8.66e-17 -8.66e-17 -1.00e+00]]
```

Comparing with `scipy`, it seems it is `log3` fault.
```python

from scipy.spatial.transform import Rotation
r_pin = pinocchio.log3(R)
r_scipy = Rotation.from_matrix(R).as_rotvec()

print("rotation vector")
print("pinocchio", r_pin)
print("scipy", r_scipy)
```

I tried this on a fresh conda install (pinocchio 2.6.7)

jmirabel, 2022-05-09:
The bug comes from https://github.com/stack-of-tasks/pinocchio/blob/5c93f4e043886ec43659a10a79701263a1e8fa18/src/spatial/log.hxx#L47

The equations implemented in this function come from https://github.com/humanoid-path-planner/hpp-constraints/blob/master/doc/rotation.tex

Indeed when `R(i,j) == R(j,i)`, the sign of `r[k]` is ill-defined.

jcarpent, 2022-05-09:
Could you have a look at pinocchio-3x. You should have access to.

jmirabel, 2022-05-09:
`quaternion::log3` does not suffer this bug. Haven't tried to compile `pinocchio-3x` but the implementation of `pinocchio::log3` *looks like* `pinocchio::quaternion::log3` so the bug may not be there anymore.

closed 2022-06-16 by jcarpent