stack-of-tasks/pinocchio #1650: Set fext when computing inverse dynamics with rnea()

FenglongSong, 2022-05-11:
Hi, first of all, thanks for developing such a great tool. I'm trying to use the rnea(model, data, q, v, a, fext) function to compute inverse dynamics of a quadruped robot. The documentation says fext is: 

> "vector of external forces expressed in the local frame of the joints (dim model.njoints)". 

In my case, the external forces are contact forces at the foot tips and there are no joints there. How should I do to deal with this? Should I transform the force to the knee joint frame? This seems to be feasible way but not straightforward. Could you give me some advice on that? Thanks.

jcarpent, 2022-05-11:
If you know the relative placement of the tip frame wrt the knee joint frame, let's say `kMt`, you can get the force expressed in the local frame of the tip to the local frame of the knee joint as `f_k = kMt.act(f_t)`.

FenglongSong, 2022-05-12:
Hi, thanks for the reply, but I'm not quite clear how to get `kMt`. I assume `kMt` should be a `pinocchio::SE3` object representing the homogeneous transformation from frame k to frame t, and `f_t` is a `pinocchio::Force` object.

Therefore, what I did was: 
```cpp
const pinocchio::FrameIndex bodyIdLFFoot = model.getFrameId("LF_FOOT");
const pinocchio::SE3Tpl<SCALAR_T> T_wf = data.oMf[model.getFrameId("LF_FOOT")]; // transformation from world to foot
const pinocchio::SE3Tpl<SCALAR_T> T_wk = data.oMf[model.getFrameId("LF_KFE")]; // transformation from world to knee
const pinocchio::SE3Tpl<SCALAR_T> T_fk = T_wk * T_wf.inverse(); // transformation from world to knee
const pinocchio::container::aligned_vector<pinocchio::ForceTpl<SCALAR_T>> force_in_knee = T_fk.act(force_at_foot);
```
 but the compiling failed with the following error:
```
error: ‘const struct pinocchio::container::aligned_vector<pinocchio::ForceTpl<double, 0> >’ has no member named ‘se3Action’
```

How should I fix this? I will be much appreciated!

FenglongSong, 2022-05-12:
I found that I actually have exactly the same questions as #1544, @jcarpent  your answer is quite helpful and I would be grateful if you can provide a bit more detail about the implementation.

jcarpent, 2022-05-12:
Your code is not complete, but you should work at the level of Force element and not std::vector<Force>. This is what the compilation info gives you.

manumerous, 2022-05-15:
@jcarpent May I ask if you know whether and why my method would be faster/slower than using the rnea function from Pinocchio? Your insights would be much appreciated!

cmastalli, 2022-05-15:
@manumerous you are not exploiting the  sparsity induced by the kinematic tree. This is what the Recursive Newton-Euler algorithm does.

If I remember correctly, the algorithm complexity of CRBA is also higher than RNEA. In short, it is terribly inefficient to do the computation yourself.

For more details, I advise you to read Rigid Body Dynamics from Featherstone.

manumerous, 2022-05-15:
Thanks for your answer! I guess I should have been more precise, but I think your comments and the book reference guided me in the right direction. 

Although i was aware of the way RNEA leverages the kinematic sparsity for fixed-base kinematic trees I was unsure what it would do in the case of several fixed contact points. Please correct me if I am wrong, but according to my current understanding there is no smart way to enforce zero velocity/acceleration at several points in the kinematic chain in the RNEA algorithm?

If that is true the best way forward depends on whether a contact constraint consistent motion plan (q, u, a, f_ext) is present. 

- If yes the floating base dynamics could be reformulated to have one of the contact points as the root and adding a corresponding joint depending on the contact. I guess this would be much very fast but requires the root joint to change upon contact. 

- If not it might be better to use [the TSID library](https://github.com/stack-of-tasks/tsid) to solve the problem via Hierarchical Least Square Optimization instead of using Pinocchios RNEA. 

In that sense my manual computation just has the advantage of being easy to implement for a floating base robot model (where the root joint can move) at the disadvantage of being much slower than RNEA. 

I hope this thoughts can be useful for others. So please correct me if anything is wrong. Thanks! :)

cmastalli, 2022-05-16:
You could artificially define the root of your system as a single contact point, and then use RNEA for that. However, you cannot do this for multiple contacts (you only have one root). This would be the most efficient way, but it is SO restrictive that (I might say) it is out of interest.

Imposing acceleration-based holonomic constraints can be done with TSID or any general-purpose QP algorithm. If we think in terms of QP, what we can solve these problems via the Schur-complement approach, for instance.
Such techniques requires matrix-matrix multiplications between the inertia matrix and contact Jacobians.
Therefore, you can leverage the structure induced by the kinematic tree during those operations.
This is something that you can do with a general-purpose approach.

As a conclusion, exploiting or not the completely structure in your algo might be extremely expensive to develop in your project. As a pragmatic person, I would do it if you really need it.

closed 2022-05-17 by jcarpent

FenglongSong, 2022-05-17:
Thanks for the great insights too from my side! Back to my original question, it's finally solved and in case anyone is also looking for a solution, I would refer to this example [here](https://github.com/leggedrobotics/ocs2/blob/main/ocs2_pinocchio/ocs2_centroidal_model/src/CentroidalModelRbdConversions.cpp) (line 189-225) which used `rnea()` to solve inverse dynamics for quadruped robot. This example might give people some hints to see how to correctly transform the contact forces in world frame to the joint frame.

And for Manuel @manumerous, thanks for your advice and your method to directly manipulate the floating-base equation of motion can absolutely give a correct result.  And I've compared the results given by your method and directly using `rnea`, they're very close. The only thing that we should notice is `rnea` returns a vector with dimension of generalized velocity. In our case, since the base is actuated, we just need to ignore the first six elements.

shubhamsingh91, 2022-07-19:
> 

> Hi, thanks for the reply, but I'm not quite clear how to get `kMt`. I assume `kMt` should be a `pinocchio::SE3` object representing the homogeneous transformation from frame k to frame t, and `f_t` is a `pinocchio::Force` object.
> 
> Therefore, what I did was:
> 
> ```c++
> const pinocchio::FrameIndex bodyIdLFFoot = model.getFrameId("LF_FOOT");
> const pinocchio::SE3Tpl<SCALAR_T> T_wf = data.oMf[model.getFrameId("LF_FOOT")]; // transformation from world to foot
> const pinocchio::SE3Tpl<SCALAR_T> T_wk = data.oMf[model.getFrameId("LF_KFE")]; // transformation from world to knee
> const pinocchio::SE3Tpl<SCALAR_T> T_fk = T_wk * T_wf.inverse(); // transformation from world to knee
> const pinocchio::container::aligned_vector<pinocchio::ForceTpl<SCALAR_T>> force_in_knee = T_fk.act(force_at_foot);
> ```
> 
> but the compiling failed with the following error:
> 
> ```
> error: ‘const struct pinocchio::container::aligned_vector<pinocchio::ForceTpl<double, 0> >’ has no member named ‘se3Action’
> ```
> 
> How should I fix this? I will be much appreciated!

Hi FenglongSong,
@FenglongSong Thanks for the issue. @jcarpent thanks for the suggestion here. 
@FenglongSong  You could essentially use your approach here for computing the force on the knee, right? But for that, don't you need T_kf instead? And then it could be T_kf.act(force_on_foot). is that correct? Also, did you instead use the OCS2 method here instead of yours? I find Justin's suggestion (also what you implemented above) quite straightforward to do.

FenglongSong, 2022-07-20:
> 

Hi shubhamsingh91 @shubhamsingh91, yes I think it's my mistake and it should be T_kf.act(force_on_foot). 

I did use these [lines](https://github.com/leggedrobotics/ocs2/blob/3566993b9fc5162f4c57ccb6f3d93aab92c5c2b1/ocs2_pinocchio/ocs2_centroidal_model/src/CentroidalModelRbdConversions.cpp#L190) in OCS2 because they match my requirements perfectly and just take copy and paste.  And I trid to simply transform external forces on foot tips into joint torques by multiplying the Jacobian. It also works well.