stack-of-tasks/pinocchio #1651: Error when integrating SO2 with the same vector given as input and output

Toefinder, 2022-05-12:
The integration method `integrate_impl` for SO2 group fails when the same vector reference is used both for input and for storing output. The code and the documentation do not indicate that they must be different. 
- version: `v2.6.3`
- how to reproduce: pass the same reference into `integrate` method of SO2 like `SpecialOrthogonalOperationTpl<2,_Scalar,_Options>().integrate(q, v, q)`

The snippet of [code of concern](https://github.com/stack-of-tasks/pinocchio/blob/5ed15458193bf255a29f6888067d887a963c0275/src/multibody/liegroup/special-orthogonal.hpp#L159) is shown below:

```c++
template <class ConfigIn_t, class Velocity_t, class ConfigOut_t>
static void integrate_impl(const Eigen::MatrixBase<ConfigIn_t> & q,
                           const Eigen::MatrixBase<Velocity_t> & v,
                           const Eigen::MatrixBase<ConfigOut_t> & qout)
{
  ConfigOut_t & out = PINOCCHIO_EIGEN_CONST_CAST(ConfigOut_t,qout);

  const Scalar & ca = q(0);
  const Scalar & sa = q(1);
  const Scalar & omega = v(0);

  Scalar cosOmega,sinOmega; SINCOS(omega, &sinOmega, &cosOmega);
  // TODO check the cost of atan2 vs SINCOS

  out << cosOmega * ca - sinOmega * sa,
         sinOmega * ca + cosOmega * sa;
```
When `qout` refers to the same vector as `q`, `out[0]` and `ca` will be referring to the same scalar value, and `out[1]` and `sa` will refer to the same value. Using a combination of double angle brackets `<<` and comma operator DOES NOT evaluate the full right hand side before assignment: instead it evaluates and assigns one by one as separated by the comma. Because of this, in the code for SO2, `cosOmega * ca - sinOmega * sa` is evaluated and assigned to `out[0]`, which means `ca` is changed, before evaluating `sinOmega * ca + cosOmega * sa` to assign to `out[1]`. The result is wrong and not normalized for most of the time. 

To demonstrate this effect with Eigen, consider the example below:
```c++
Eigen::Matrix<double,2,1> v;
v << 1,2;
std::cout << "v=(" << v[0] << ", " << v[1] << ")" << std::endl;
v << v[1], v[0];
std::cout << "v=(" << v[0] << ", " << v[1] << ")" << std::endl;
```
Running this gives
> v=(1, 2)
v=(2, 2)

Toefinder, 2022-05-12:
Removing the reference from `const Scalar & ca = q(0);  const Scalar & sa = q(1);` fixes the issue

jcarpent, 2022-05-13:
SOlved via #1652

PR #1652 (2022-05-13): Fix integrate method for SO2
When the same vector reference is given to the integrate method as both
for input and for storing the output, the result is incorrectly
calculated due to inplace changes. This problem is fixed in this commit.
Close #1651
closed 2022-05-13 by jcarpent