stack-of-tasks/pinocchio #1664: Dimension of aba_partial_dq in computeABAderivatives()

FenglongSong, 2022-05-30:
Hi, I have two questions.
1. By my understanding, aba-derivatives should be providing the derivatives of **generalized acceleration** w.r.t q, v,  and tau. Is the **generalized torque vector** in the following comment a typo?
https://github.com/stack-of-tasks/pinocchio/blob/5c93f4e043886ec43659a10a79701263a1e8fa18/src/algorithm/aba-derivatives.hpp#L29
2. I would expect the size of `aba_partial_dq` to be (nv, nq) since the derivative is w.r.t to joint configuration vector. I don't understand why the size of `aba_partial_dq` is (nv, nv) as I found in the following line:
https://github.com/stack-of-tasks/pinocchio/blob/5c93f4e043886ec43659a10a79701263a1e8fa18/src/algorithm/aba-derivatives.hxx#L346

I will appreciate for your answer :)

jcarpent, 2022-05-30:
> 1. By my understanding, aba-derivatives should be providing the derivatives of **generalized acceleration** w.r.t q, v,  and tau. Is the **generalized torque vector** in the following comment a typo?

This is indeed a copy-pasting mistake. Sorry for that. It refers in fact to the generalized joint acceleration.

> 2\. I would expect the size of `aba_partial_dq` to be (nv, nq) since the derivative is w.r.t to joint configuration vector. I don't understand why the size of `aba_partial_dq` is (nv, nv) as I found in the following line:

You are working on the tangent space of the configuration space. For instance, for the free-flyer, the joint configuration space is SE(3) represented by a translation and a quaternion, while the tangent space is so(3), corresponding to a Motion quantity composed of the linear and angular velocity vectors, both of dimension 3.

Is it answering your question?

FenglongSong, 2022-05-30:
Thanks for your reply. I guess dq is something like as a velocity *time, rather that a numerical difference of quaternion. That will be clear. I will close this issue soon.

And BTW, what I want to get is the analytical derivatives of accelerations w.r.t the joint configuration. In free flyer case, it should contain a submatrix of size (3,4) describing the derivatives of angular acceleration w.r.t the quaternion. Would there be an efficient way to do so?

closed 2022-05-30 by FenglongSong

jcarpent, 2022-05-30:
> I guess dq is something like as a velocity *time, rather that a numerical difference of quaternion. That will be clear. I will close this issue soon.

You can indeed think of it like this.