stack-of-tasks/pinocchio #1666: When I run ctest I received the following failed tests

mahyaret, 2022-05-31:
```
62/67 Test #62: example-cpp-overview-urdf ..................Child aborted***Exception:   0.00 sec
      Start 63: example-cpp-build-reduced-model
63/67 Test #63: example-cpp-build-reduced-model ............Child aborted***Exception:   0.00 sec
      Start 64: example-cpp-geometry-models
64/67 Test #64: example-cpp-geometry-models ................Child aborted***Exception:   0.00 sec
      Start 65: example-cpp-kinematics-derivatives
65/67 Test #65: example-cpp-kinematics-derivatives .........Child aborted***Exception:   0.00 sec
      Start 66: example-cpp-forward-dynamics-derivatives
66/67 Test #66: example-cpp-forward-dynamics-derivatives ...Child aborted***Exception:   0.00 sec
      Start 67: example-cpp-inverse-dynamics-derivatives
67/67 Test #67: example-cpp-inverse-dynamics-derivatives ...Child aborted***Exception:   0.00 sec

90% tests passed, 7 tests failed out of 67

Total Test time (real) =  26.71 sec

The following tests FAILED:
         21 - test-cpp-urdf (Failed)
         62 - example-cpp-overview-urdf (Child aborted)
         63 - example-cpp-build-reduced-model (Child aborted)
         64 - example-cpp-geometry-models (Child aborted)
         65 - example-cpp-kinematics-derivatives (Child aborted)
         66 - example-cpp-forward-dynamics-derivatives (Child aborted)
         67 - example-cpp-inverse-dynamics-derivatives (Child aborted)
Errors while running CTest 
```

jcarpent, 2022-05-31:
Could use the option of ctest output-on-failure to show the issue on the URDF test?

mahyaret, 2022-05-31:
Thanks for your hint! I ran `ctest -V` I've figured the problem was that it could not find the urdf file for the test robot:
```src/models/example-robot-data/robots/ur_description/urdf/ur5_robot.urdf does not contain a valid URDF model.```
For some reason it was not pulled properly during `git clone --recursive https://github.com/stack-of-tasks/pinocchio`.
Any idea what could be the reason or it was simply my error?

jcarpent, 2022-05-31:
Indeed you were missing the sub module. We can try to add some check within cmake to help users. Nice you have found the solution.

closed 2022-05-31 by jcarpent