stack-of-tasks/pinocchio #1675: installing from source documentation is outdated

IvanVnucec, 2022-06-08:
https://stack-of-tasks.github.io/pinocchio/download.html

There are a few issues I've stumbled upon:
1. `Optional dependencies` are not optional rather mandatory because I could not get `cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local` to pass.
2. `Eigenpy 2.6` is hard to install on Ubuntu because you cannot just install it with `apt-get` rather you must first download it from sources [here](https://github.com/stack-of-tasks/eigenpy/releases/download/v2.6.11/eigenpy-2.6.11.tar.gz) and then install it with cmake. Because of that I think that the install procedure should be added to pinnochio documentation.

jcarpent, 2022-06-08:
Do you want to use the Python bindings? Otherwise, you can add to the cmake command line `-DBUILD_PYTHON_INTERFACE=OFF` to deactivate the building of the Python interface.

jcarpent, 2022-06-15:
I'm closing the issue as there is no answer from the original author.

closed 2022-06-15 by jcarpent

cmastalli, 2022-06-15:
Hi @jcarpent 

I don't know if this is the current behaviour. However, I think it should not compile the Python bindings if eigenpy (and perhaps boost Python) is not detected.

You might also print a message that the python bindings are not going to be generated as it was not detected those dependencies.

IvanVnucec, 2022-06-16:
I think that the problem still persist and closing it will not solve it. I have no intentions to contribute.

jcarpent, 2022-06-21:
> I think that the problem still persist and closing it will not solve it. I have no intentions to contribute.

Have you been able to install Pinocchio finally @IvanVnucec?

IvanVnucec, 2022-06-21:
> > I think that the problem still persist and closing it will not solve it. I have no intentions to contribute.
> 
> Have you been able to install Pinocchio finally @IvanVnucec?

Yes I did but with a lot of fiddling around.

It was really a brute-force approach. Fortunately for me, I have a little bit of experience with programming and building/installing other dependencies was successful, but not easy. 

If someone who is novice in programming tries to install Pinocchio from source, I'm afraid one will not succeed.