stack-of-tasks/pinocchio #1685: Aborted (core dumped) when run example-cpp-collisions

hxrkpib, 2022-06-21:
https://github.com/stack-of-tasks/pinocchio/blob/master/examples/collisions.cpp

version:
ubuntu 18.04
pinocchio master branch(v2.6.8)
hpp-fcl 1.7.3 or 1.8.0

run`./example-cpp-collisions`

I received the following
> example-cpp-collisions: /home/user/hpp-fcl/hpp-fcl/src/narrowphase/gjk.cpp:512: bool hpp::fcl::details::details::getClosestPoints(const hpp::fcl::details::GJK::Simplex&, hpp::fcl::Vec3f&, hpp::fcl::Vec3f&): Assertion `false' failed.
> Aborted (core dumped)

jcarpent, 2022-06-21:
Thanks @hxrkpib for reporting this issue. Do you have several installations of Pinocchio and hpp-fcl (locate libhpp-fcl.so, libpinocchio.so)?

jcarpent, 2022-06-21:
Could you run `ldd /usr/local/lib/libhpp-fcl.so`?

hxrkpib, 2022-06-21:
> ldd /usr/local/lib/libhpp-fcl.so

> Could you run `ldd /usr/local/lib/libhpp-fcl.so`?

[image]

fabinsch, 2022-06-21:
Hi, I created an Ubuntu 18.04 docker image installing hpp-fcl (v.1.8.0) and pinocchio master branch (v.2.6.8) that you can access here: [Pinocchio-example-collisions](https://hub.docker.com/layers/fabinsch/pinocchio-example-collisions/latest/images/sha256:7c9a1ffe8b331e1c42b47911ad802bd04d26f1c4aef91e2ec4252d9405158f3c)

`docker run` executes the cpp-example-collisions and everything works fine.

Your problem should be coming from your setup and we advice you to start off with a clean environment.

hxrkpib, 2022-06-22:
@fabinsch @jcarpent 
Thanks a lot for you help! I found the problem that I build hpp-fcl under debug mode.