stack-of-tasks/pinocchio #1686: Collision detection between multi instances of the same model ?

aamorel, 2022-06-23:
First of all thank you for the amazing work on Pinocchio.

I am trying to simulate forward kinematics and collision detection for a multi-robot setup.
Each robot has the same URDF source.

What would be the correct way to implement forward kinematics and collision detection in that case ? 

For now I succeeded for the forward kinematic part with this code, but I don't know how to adapt the collisionDetection example to my case.

`# This examples shows how to load and move a robot in meshcat.
# Note: this feature requires Meshcat to be installed, this can be done using
# pip install --user meshcat

from tkinter import N
from xml.etree.ElementTree import PI
import pinocchio as pin
import numpy as np
import sys
import os
from os.path import dirname, join, abspath
import time
import math

from pinocchio.visualize import MeshcatVisualizer

####################### constants #############################
VIZ = False
PI = math.pi
VOXEL_SIZE = 0.110
NB_ROBOTS = 2

# Load the URDF model.
# Conversion with str seems to be necessary when executing this file with ipython
pinocchio_model_dir = join(
    dirname(str(abspath(__file__))), "roombot_description_no_latch")

model_path = join(pinocchio_model_dir, "urdf")
mesh_dir = pinocchio_model_dir
urdf_filename = "roombot.urdf"
urdf_model_path = join(model_path, urdf_filename)

model, collision_model, visual_model = pin.buildModelsFromUrdf(
    urdf_model_path, mesh_dir, pin.JointModelFreeFlyer())

# create qs, datas, and viz
qs = []
datas = []
vizs = []

# get neutral config
q0 = pin.neutral(model)

# initialize config and data for all robots
# q[0]: x-translation, q[1]: y-translation, q[2]: z-translation,
# q[7]: M0, q[8]: M1, q[9]: M2
for i in range(NB_ROBOTS):
    q = q0.copy()
    q[1] = VOXEL_SIZE * i
    qs.append(q)
    data = model.createData()
    pin.forwardKinematics(model, data, q)
    datas.append(data)

if VIZ:
    # create visualizers
    for _ in range(NB_ROBOTS):
        viz = MeshcatVisualizer(model, collision_model, visual_model)
        vizs.append(viz)

    # init first viz
    # Start a new MeshCat server and client.
    # Note: the server can also be started separately using the "meshcat-server" command in a terminal:
    # this enables the server to remain active after the current script ends.
    #
    # Option open=True pens the visualizer.
    # Note: the visualizer can also be opened seperately by visiting the provided URL.
    try:
        vizs[0].initViewer(open=True)
    except ImportError as err:
        print("Error while initializing the viewer. It seems you should install Python meshcat")
        print(err)
        sys.exit(0)

    # load robots in the visualizers
    vizs[0].loadViewerModel()
    for i in range(NB_ROBOTS - 1):
        vizs[i + 1].initViewer(vizs[0].viewer)
        vizs[i + 1].loadViewerModel(rootNodeName="pinocchio2")

    # display all robots
    for i in range(NB_ROBOTS):
        vizs[i].display(qs[i])
        vizs[i].displayCollisions(True)
        vizs[i].displayVisuals(False)
        # Print out the placement of each joint of the kinematic tree
        for name, oMi in zip(model.names, datas[i].oMi):
            print(("{:<24} : {: .6f} {: .6f} {: .6f}"
                   .format(name, *oMi.translation.T.flat)))

# moving robot 0
idx = 0
IT_MAX = 1000
i = 0
while True:
    pin.forwardKinematics(model, datas[idx], qs[idx])
    qs[idx][7] = qs[idx][7] + 0.01
    if VIZ:
        vizs[idx].display(qs[idx])
        time.sleep(0.01)
    if i >= IT_MAX or qs[idx][7] > 2 * PI/3:
        break
    i += 1

# moving robot 1
idx = 1
IT_MAX = 1000
i = 0
while True:
    pin.forwardKinematics(model, datas[idx], qs[idx])
    qs[idx][7] = qs[idx][7] + 0.01
    if VIZ:
        vizs[idx].display(qs[idx])
        time.sleep(0.01)
    if i >= IT_MAX or qs[idx][7] > 2 * PI/3:
        break
    i += 1
`

jcarpent, 2022-06-23:
@aamorel Could you explain shortly what do want to do?

jcarpent, 2022-06-23:
@aamorel Could you also tell us who are you?

aamorel, 2022-06-23:
Thank you for your fast answer !

I am a new PhD student at EPFL (Lausanne, Switzerland) and I will work on self-reconfigurable modular robotics. 

I am working on self-reconfiguration planning: having the modules locomote individually to eventually form a shape. I already have a MDP representation of the problem and have a search algorithm. However, in order to check that an action is valid, I need to verify that is does not imply a collision between the current active module and the others. 

In that sense, I want to use Pinocchio to have a simple kinematic simulation of several modules behaving in the same environment in order to check for potential collision between two modules during an action.

In the example code that I gave above, I first move robot 1 (which does not collide 2) and then move robot 2 (which now collides with 1). I would like to be able to detect that collision during the second action. 

I'm new to Pinocchio so I probably do some things wrong. I got inspired by the example _Display a model using Meshcat_ to code this setup with multiple robots but I'm not sure it's the right way to do it. 

Please ask me if you need more details. 
Aurélien

jcarpent, 2022-06-23:
Thanks for the complete details and happy to see that you are using Pinocchio for your research. For collision detection, you have this tiny example: https://github.com/stack-of-tasks/pinocchio/blob/master/examples/collisions.py.
I hope it will be sufficient for you to move forward.

aamorel, 2022-06-23:
Actually, I have seen this example but it is not sufficient to implement collision detection between two robots. 
I've tried many tricks but I can't find a solution.

My best bet so far was to create a "global geometry model" and add a copy of all geometryObjects inside it. 
However, in that case I don't know how to properly set the geometryData as I can't use the UpdateGeometryPlacements() method. 

Do you have a clue on how to perform this ?

jcarpent, 2022-06-23:
The best would be to merge the two robots into a single one ... 
Or simply to add all the geoms of a given robot into the first one, as attached to the world frame. And then updating manually the placement of these geometries by changing the placement field of the geometry objects contained in the geometry model.

aamorel, 2022-06-23:
I'm not sure if I understand well what is inside geometryModel and what is inside geometryData.

In the second solution so you mention, how would you generate the geometryData for the new global geometryModel ?

jcarpent, 2022-06-23:
geom_data = pin.GeometryData(geometryModel)

aamorel, 2022-06-23:
Ok so if I understand well you suggest to create a global geometryModel such as:
```
# add all objects to global model
for i in range(NB_ROBOTS):
    for geom_obj in collision_global_model.geometryObjects:
        collision_global_model.addGeometryObject(geom_obj)
```
giving me such structure if we make the hypothesis that I have two robots:
`geometryModel: {geomObject1, geomObject2, geomObject1, geomObject2}`

I write 1 and 2 two times because the two robots are identical and composed of two geomObject

then i use:
`geom_data = pin.GeometryData(geometryModel)`

and then I should use the previous geom_data_robot_1 and geom_data_robot_2 and manually set the `geom_data.oMg` field  ? 

Thanks a lot for your time !

jcarpent, 2022-06-23:
Yes. You can use `appendModel` to append two models together and create a joint and unified model.

jcarpent, 2022-06-23:
I will close this issue as it seems to have been correctly answered.