stack-of-tasks/pinocchio #1687: Python scripts stops and no error when using pin.appendModel()

aamorel, 2022-06-24:
I'm trying to merge two kinematic chains into one. For this I am trying to use pin.appendModel() to create a new model from the two that represent the two initial kinematic chains. The kinematic chains are identical (coming from the same URDF). 
However, when I use this method, the Python script just stops without any error message. 

Here is a minimal example ('succeeded' is not printed):
```
import pinocchio as pin
from os.path import dirname, join, abspath
import copy

# Load the URDF model.
# Conversion with str seems to be necessary when executing this file with ipython
pinocchio_model_dir = join(
    dirname(str(abspath(__file__))), "roombot_description_no_latch")

model_path = join(pinocchio_model_dir, "urdf")
mesh_dir = pinocchio_model_dir
urdf_filename = "roombot.urdf"
urdf_model_path = join(model_path, urdf_filename)

# build reference model
model, collision_model, visual_model = pin.buildModelsFromUrdf(
    urdf_model_path, mesh_dir, pin.JointModelFreeFlyer())

# changing the names of frames and joints to avoid name conflicts
# build custom model for robot 1
custom_model_1 = copy.deepcopy(model)
for i, name in enumerate(custom_model_1.names):
    custom_model_1.names[i] = name + str(1)
for i, frame in enumerate(custom_model_1.frames):
    custom_model_1.frames[i].name = frame.name + str(1)

# build custom model for robot 2
custom_model_2 = copy.deepcopy(model)
for i, name in enumerate(custom_model_2.names):
    custom_model_2.names[i] = name + str(2)
for i, frame in enumerate(custom_model_2.frames):
    custom_model_2.frames[i].name = frame.name + str(2)

# merge models
new_model = pin.appendModel(
    custom_model_1, custom_model_2, 0, pin.SE3.Identity())
print('succeeded')
```
I probably do something wrong but I don't understand what. Thank you in advance for your time.

jcarpent, 2022-06-24:
Don't use deepcopy as the objects are managed by C++.
Here is the working example:
```
import pinocchio as pin
from os.path import dirname, join, abspath
import copy

# Load the URDF model.
# Conversion with str seems to be necessary when executing this file with ipython
pinocchio_model_dir = join(
    dirname(str(abspath(__file__))), "roombot_description_no_latch")

model_path = join(pinocchio_model_dir, "urdf")
mesh_dir = pinocchio_model_dir
urdf_filename = "roombot.urdf"
urdf_model_path = join(model_path, urdf_filename)

# build reference model
model, collision_model, visual_model = pin.buildModelsFromUrdf(
    urdf_model_path, mesh_dir, pin.JointModelFreeFlyer())

# changing the names of frames and joints to avoid name conflicts
# build custom model for robot 1
custom_model_1 = model.copy()
for i, name in enumerate(custom_model_1.names):
    custom_model_1.names[i] = name + str(1)
for i, frame in enumerate(custom_model_1.frames):
    custom_model_1.frames[i].name = frame.name + str(1)

# build custom model for robot 2
custom_model_2 = model.copy()
for i, name in enumerate(custom_model_2.names):
    custom_model_2.names[i] = name + str(2)
for i, frame in enumerate(custom_model_2.frames):
    custom_model_2.frames[i].name = frame.name + str(2)

# merge models
new_model = pin.appendModel(
    custom_model_1, custom_model_2, 0, pin.SE3.Identity())
print('succeeded')
```

PS: next time, you should provide the URDF model of your robot.

aamorel, 2022-06-24:
There is still a problem I think. I updated the code (and now use a pre-given URDF).
Now, the script succeeds approximately once in 3 times. The other times, it does the same as before and exits without any error. 
```
import pinocchio as pin
from os.path import dirname, join, abspath
import copy

# Load the URDF model.
# Conversion with str seems to be necessary when executing this file with ipython
pinocchio_model_dir = join(dirname(dirname(str(abspath(__file__)))), "models")

model_path = join(pinocchio_model_dir, "example-robot-data/robots")
mesh_dir = pinocchio_model_dir
urdf_filename = "talos_reduced.urdf"
urdf_model_path = join(join(model_path, "talos_data/robots"), urdf_filename)

model, collision_model, visual_model = pin.buildModelsFromUrdf(
    urdf_model_path, mesh_dir, pin.JointModelFreeFlyer())

# changing the names of frames and joints to avoid name conflicts
# build custom model for robot 1
custom_model_1 = model.copy()
for i, name in enumerate(custom_model_1.names):
    custom_model_1.names[i] = name + str(1)
for i, frame in enumerate(custom_model_1.frames):
    custom_model_1.frames[i].name = frame.name + str(1)

# build custom model for robot 2
custom_model_2 = model.copy()
for i, name in enumerate(custom_model_2.names):
    custom_model_2.names[i] = name + str(2)
for i, frame in enumerate(custom_model_2.frames):
    custom_model_2.frames[i].name = frame.name + str(2)

# merge models
new_model = pin.appendModel(
    custom_model_1, custom_model_2, 0, pin.SE3.Identity())
print('succeeded')
```

jcarpent, 2022-06-24:
I will handle it on Monday.

jcarpent, 2022-06-27:
@aamorel Thanks for raising this issue. We have found the bug with @fabinsch and the solution is here #1693.
You can install it manually or I will try to make a new release soon.