stack-of-tasks/pinocchio #1694: Unable to compile pinocchio3-preview branch

GiulioRomualdi, 2022-06-27:
Hi all! First of all thanks for developing and maintaining Pinocchio! It's a great library! 
I'm trying to compile the `pinocchio3-preview` branch since I would like to test the casadi + python feature https://github.com/stack-of-tasks/pinocchio/issues/1569#issuecomment-987974103.

I tried to compile Pinocchio from sources and I installed all the dependencies through `conda`.
Enabling the casadi support with `BUILD_WITH_CASADI_SUPPORT` I got the following error
```
In file included from /home/gromualdi/mambaforge/envs/pinocchio-3-env/include/boost/python/object/class_wrapper.hpp:8,
                 from /home/gromualdi/mambaforge/envs/pinocchio-3-env/include/boost/python/object/class_metadata.hpp:12,
                 from /home/gromualdi/mambaforge/envs/pinocchio-3-env/include/boost/python/class.hpp:23,
                 from /home/gromualdi/mambaforge/envs/pinocchio-3-env/include/boost/python.hpp:18,
                 from /home/gromualdi/robot-code/pinocchio/bindings/python/math/expose-eigen-types.cpp:5:
/home/gromualdi/mambaforge/envs/pinocchio-3-env/include/boost/python/to_python_converter.hpp: In instantiation of 'static const PyTypeObject* boost::python::to_python_converter<T, Conversion, has_get_pytype>::get_pytype_1(mpl_::true_*) [with T = Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 1, 1, 0, 1, 1>; Conversion = eigenpy::EigenToPy<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 1, 1, 0, 1, 1>, casadi::Matrix<casadi::SXElem> >; bool has_get_pytype = true; PyTypeObject = _typeobject; mpl_::true_ = mpl_::bool_<true>]':
/home/gromualdi/mambaforge/envs/pinocchio-3-env/include/boost/python/to_python_converter.hpp:64:28:   required from 'static const PyTypeObject* boost::python::to_python_converter<T, Conversion, has_get_pytype>::get_pytype_impl() [with T = Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 1, 1, 0, 1, 1>; Conversion = eigenpy::EigenToPy<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 1, 1, 0, 1, 1>, casadi::Matrix<casadi::SXElem> >; bool has_get_pytype = true; PyTypeObject = _typeobject]'
/home/gromualdi/mambaforge/envs/pinocchio-3-env/include/boost/python/to_python_converter.hpp:86:11:   required from 'boost::python::to_python_converter<T, Conversion, has_get_pytype>::to_python_converter() [with T = Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 1, 1, 0, 1, 1>; Conversion = eigenpy::EigenToPy<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 1, 1, 0, 1, 1>, casadi::Matrix<casadi::SXElem> >; bool has_get_pytype = true]'
/home/gromualdi/mambaforge/envs/pinocchio-3-env/include/eigenpy/eigen-to-python.hpp:133:9:   required from 'static void eigenpy::EigenToPyConverter<MatType>::registration() [with MatType = Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 1, 1, 0, 1, 1>]'
/home/gromualdi/mambaforge/envs/pinocchio-3-env/include/eigenpy/details.hpp:68:44:   required from 'void eigenpy::enableEigenPySpecific() [with MatType = Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 1, 1, 0, 1, 1>]'
/home/gromualdi/robot-code/pinocchio/bindings/python/math/expose-eigen-types.cpp:75:57:   required from here
/home/gromualdi/mambaforge/envs/pinocchio-3-env/include/boost/python/to_python_converter.hpp:55:38: error: 'get_pytype' is not a member of 'eigenpy::EigenToPy<Eigen::Matrix<casadi::Matrix<casadi::SXElem>, 1, 1, 0, 1, 1>, casadi::Matrix<casadi::SXElem> >'
   55 |         return Conversion::get_pytype();
      |                ~~~~~~~~~~~~~~~~~~~~~~^~
```

Accordingly to the error, there seems that I'm using the wrong version of `eigenpy`. In my case, conda installed the latest version `2.7.6`. 

Could you please help me to solve this issue? 

<details><summary>Click me to check the outcome of `conda list` </summary>
<p>

```
# packages in environment at /home/gromualdi/mambaforge/envs/pinocchio-3-env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
assimp                    5.2.4                h4c92fa2_0    conda-forge
binutils                  2.36.1               hdd6e379_2    conda-forge
binutils_impl_linux-64    2.36.1               h193b22a_2    conda-forge
binutils_linux-64         2.36                hf3e587d_10    conda-forge
boost                     1.74.0          py310h7c3ba0c_5    conda-forge
boost-cpp                 1.74.0               h75c5d50_8    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
c-compiler                1.4.2                h166bdaf_0    conda-forge
ca-certificates           2022.6.15            ha878542_0    conda-forge
casadi                    3.5.5           py310h597d7ca_9    conda-forge
cmake                     3.23.2               h5432695_0    conda-forge
compilers                 1.4.2                ha770c72_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
cxx-compiler              1.4.2                h924138e_0    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
eigenpy                   2.7.6           py310hf3e5c9c_0    conda-forge
elfutils                  0.186                he364ef2_0    conda-forge
expat                     2.4.8                h27087fc_0    conda-forge
expat-cos6-x86_64         2.0.1             h9b0a68f_1105    conda-forge
fortran-compiler          1.4.2                h2a4ca65_0    conda-forge
gcc                       10.3.0              he2824d0_10    conda-forge
gcc_impl_linux-64         10.3.0              hf2f2afa_16    conda-forge
gcc_linux-64              10.3.0              hc39de41_10    conda-forge
gettext                   0.19.8.1          h73d1719_1008    conda-forge
gfortran                  10.3.0              h18518b4_10    conda-forge
gfortran_impl_linux-64    10.3.0              h73f4979_16    conda-forge
gfortran_linux-64         10.3.0              hb09a455_10    conda-forge
gnutls                    3.7.6                hf3e180e_5    conda-forge
gxx                       10.3.0              he2824d0_10    conda-forge
gxx_impl_linux-64         10.3.0              hf2f2afa_16    conda-forge
gxx_linux-64              10.3.0              h2593f52_10    conda-forge
hpp-fcl                   2.0.1           py310h2ff80d2_2    conda-forge
icu                       70.1                 h27087fc_0    conda-forge
[... 87 lines cut ...]
xorg-damageproto          1.2.1             h7f98852_1002    conda-forge
xorg-fixesproto           5.0               h7f98852_1002    conda-forge
xorg-glproto              1.4.17            h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxdamage           1.1.5                h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.10            h7f98852_1003    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-util-macros          1.19.3               h7f98852_0    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
zlib                      1.2.12               h166bdaf_1    conda-forge
zstd                      1.5.2                h8a70e8d_1    conda-forge
```

</p>
</details>

jcarpent, 2022-06-27:
@GiulioRomualdi Thanks for using Pinocchio at IIT, we really appreciate it.
I've pushed a fix on the `pinocchio3-preview` branch.
It should solve your issue.

GiulioRomualdi, 2022-06-28:
Hi @jcarpent! Thanks for the quick replay! The code you pushed fixes the problem! Thanks for helping.

closed 2022-06-28 by GiulioRomualdi

jcarpent, 2022-06-28:
> Hi @jcarpent! Thanks for the quick replay! The code you pushed fixes the problem! Thanks for helping.

With great pleasure. We have also a more advanced version of pinocchio3 which we plan to release soon (we are late as usual).