stack-of-tasks/pinocchio #1708: Why do I get extra number of frames when I do model.buildModelFromUrdf()?

winkhoo, 2022-07-06:
Hello all,

I am fairly new to pinocchio. Pardon me if the question is considerably naive. According to RVIZ,

It should have only 3 frames as shown below:
[image]

But as pinocchio returns 7 frames, namely with the additional frames named "universe", "root_joint", and also including all the joints and links. Instead of just three frames as showed by RVIZ.

[image]

Thank you in advance.

jcarpent, 2022-07-06:
universe is a default frame in Pinocchio related to the universe.
Then, for each joint, the URDF parser in Pinocchio adds a Frame in such a way you can operate some easy operations just by using frames.

closed 2022-07-06 by jcarpent

winkhoo, 2022-07-06:
Thank you mate, this is helpful to me.