stack-of-tasks/pinocchio #1715: Hermeticity of "pip install pin" for Bazel

stephane-caron, 2022-07-14:
Using Pinocchio as a Python dependency via the ``pip_install`` rule of Bazel's [rules_python](https://github.com/bazelbuild/rules_python) does not work.

I haven't figured out a working solution yet. Here are some ongoing notes, all suggestions welcome :slightly_smiling_face: 

### Symptom

The ``pip_install`` rule itself does not fail at build time, but trying to import Pinocchio then fails at run time:

```
Traceback (most recent call last):
  File [redacted], line 24, in <module>
    import pinocchio
  File "[runfiles]/pip_upkie_locomotion/pypi__pin/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ImportError: libboost_python38.so.1.76.0: cannot open shared object file: No such file or directory
```

### Diagnosis

Installing Pinocchio as a PyPI package creates not only a ``pinocchio`` module to the library folder, but also ``openrobots.libs`` which is shared with other Pinocchio dependencies like eigenpy. The idea is that ``site-packages`` will look like this:

```console
$ ls lib/python3.8/site-packages
_distutils_hack/          pin-2.6.4.dist-info/          wheel-0.37.1.dist-info/
eigenpy/                  pinocchio/                    distutils-precedence.pth
eigenpy-2.6.9.dist-info/  pip/                          pip-22.1.2.virtualenv
hppfcl/                   pip-22.1.2.dist-info/         setuptools-62.4.0.virtualenv
hpp_fcl-1.7.8.dist-info/  pkg_resources/                _virtualenv.pth
numpy/                    __pycache__/                  _virtualenv.py
numpy-1.23.1.dist-info/   setuptools/                   wheel-0.37.1.virtualenv
numpy.libs/               setuptools-62.4.0.dist-info/
openrobots.libs/          wheel/
```

But Bazel is [strict about isolation](https://bazel.build/concepts/hermeticity), so it will actually produce something like this:

```console
$ tree -L 2 -d /path/to/pip/runfiles 
/path/to/pip/runfiles
├── ...
├── pypi__eigenpy
│   ├── ...
│   └── openrobots.libs
│       ├── __init__.py
│       ├── libboost_python38.so.1.76.0
│       └── libeigenpy.so
├── pypi__hpp_fcl
│   ├── ...
│   └── openrobots.libs
│       ├── __init__.py
│       ├── libassimp.so.3
│       ├── libboost_chrono.so.1.76.0
│       ├── libboost_serialization.so.1.76.0
│       ├── libboost_system.so.1.76.0
│       ├── libhpp-fcl.so
│       ├── liboctomap.so.1.9
│       └── liboctomath.so.1.9
├── pypi__pin
│   ├── ...
│   ├── openrobots.libs
│   │   ├── __init__.py
│   │   ├── libboost_filesystem.so.1.76.0
│   │   ├── libconsole_bridge.so.0.4
│   │   ├── libpinocchio.so
│   │   ├── libtinyxml.so.2
│   │   ├── liburdfdom_model.so.1.0
│   │   ├── liburdfdom_model_state.so.1.0
│   │   ├── liburdfdom_sensor.so.1.0
│   │   └── liburdfdom_world.so.1.0
│   ├── ...
├── ...
```

When importing ``pinocchio``, the shared libs in ``pypi__pin/openrobots.libs`` are found, but ``libboost_python38.so.1.76.0`` is not found since it is in ``pypi_eigenpy``.

### Leads / Not tried yet

* Try splitting ``openrobots.libs`` into separate libraries to avoid a name clash?

stephane-caron, 2022-07-15:
Sounds good, I'll keep an eye on https://pypi.org/project/pin/#history and try again when the new version comes out.

jcarpent, 2022-10-24:
@stephane-caron Should we keep it open?

stephane-caron, 2022-10-28:
It can be tested again now that Pinocchio 2.6.10 is on PyPI https://github.com/stack-of-tasks/pinocchio/issues/1745#issuecomment-1288679235. I will take a look at it next week.

stephane-caron, 2022-11-07:
OK, so I've tried and getting it to work with Bazel may be a low-hanging fruit. Although, I don't understand what's happening with cmeel and cmeel.prefix, so it may also not be, you tell me :wink: 

Here is a reproduction repo: [bazel_pinocchio](https://github.com/stephane-caron/bazel_pinocchio). This should run the overview Python example from Pinocchio:

```console
$ bazel run //examples/python:overview-simple
```

Currently it results in:

```
  File "/.../examples/python/overview-simple.py", line 13, in <module>
    import pinocchio
ModuleNotFoundError: No module named 'pinocchio'
```

As far as I can tell, the reason for it is that Bazel populates ``sys.path`` as follows:

```
sys.path = [
    ...
    "bazel-out/k8-fastbuild/bin/examples/python/overview-simple.runfiles/pip_bazel_pinocchio/pypi__cmeel/cmeel.prefix/lib/python3.8/site-packages",
    ...
]
```

The problem is that the ``cmeel.prefix`` that contains pinocchio is located in ``pypi__pin`` rather than in ``pypi__cmeel``, so the sys.path is incomplete.

@nim65s how is ``cmeel.prefix`` added to ``sys.path``?

nim65s, 2022-11-07:
> @nim65s how is `cmeel.prefix` added to `sys.path` in CMake Wheels?

through `cmeel.pth`. 

Thanks for the example, I'll have a look

stephane-caron, 2022-11-07:
OK sounds good :smiley: I've added bazelisk to the repo so that you can test it without installing anything: from the cloned working directory, just run

```
./tools/bazelisk run //examples/python:overview-simple
```

jcarpent, 2023-01-23:
@stephane-caron May we close this issue?

stephane-caron, 2023-01-23:
Well the issue is still open, but Bazel is not officially supported by Pinocchio so I suggest we move it to https://github.com/stephane-caron/bazel_pinocchio/issues/1 and close it here.

closed 2023-01-23 by stephane-caron