stack-of-tasks/pinocchio #1724: Couldn't Catkin_make pinocchio_ros_example

hh2564, 2022-08-01:
When I cloned the pinocchio_ros_example to my local working catkin workspace and trying to catkin_make without changing anything, it says: 
home/crrl/catkin_ws/src/pinocchio_ros_example/src/example.cpp:1:10: fatal error: pinocchio/parsers/sample-models.hpp: No such file or directory
    1 | #include <pinocchio/parsers/sample-models.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [pinocchio_ros_example/CMakeFiles/example.dir/build.make:63: pinocchio_ros_example/CMakeFiles/example.dir/src/example.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:404: pinocchio_ros_example/CMakeFiles/example.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j8 -l8" failed

I installed Pinocchio on Ubuntu 20.04, and I think the installation doesn't have any problem because I could import Pinocchio while using python. 

Thank you very much in advance!

wxmerkt, 2022-08-01:
I just tried to reproduce using the ROS buildfarm released Pinocchio (`ros-noetic-pinocchio`) and both catkin_make and catkin-tools. It did not trigger this issue. How did you install Pinocchio? How is your workspace configured?

Can you provide us with the output of `locate sample-models.hpp` and `ls /opt/ros/noetic/include/pinocchio/parsers`?

hh2564, 2022-08-01:
crrl@crrl-HP-Z440-Workstation:~/catkin_ws$ locate sample-models.hpp
/opt/openrobots/include/pinocchio/bindings/python/parsers/sample-models.hpp
/opt/openrobots/include/pinocchio/parsers/sample-models.hpp
/opt/ros/noetic/include/pinocchio/bindings/python/parsers/sample-models.hpp
/opt/ros/noetic/include/pinocchio/parsers/sample-models.hpp

crrl@crrl-HP-Z440-Workstation:~/catkin_ws$ ls /opt/ros/noetic/include/pinocchio/parsers
python.hpp  sample-models.hpp  sample-models.hxx  srdf.hpp  srdf.hxx  urdf  urdf.hpp  utils.hpp

hh2564, 2022-08-01:
I installed Pinocchio using 
 sudo apt install -qqy robotpkg-py38-pinocchio

wxmerkt, 2022-08-01:
You most likely have a conflict between the two installations - robotpkg & the ROS provided distributions are unlikely to be compatible, I would recommend using one or the other. While I am not sure why this would lead to the above configuration error, can you please make sure to only have one sourced on your paths (i.e. no export ...=/opt/openrobots). You can also add a `message(WARNING ${pincchio_INCLUDE_DIRS}` or a corresponding message extracting the INTERFACE_INCLUDE_DIRS from the pinocchio::pinocchio target to the CMakeLists?

hh2564, 2022-08-01:
> message(WARNING ${pincchio_INCLUDE_DIRS}

I deleted ros-noetic-pinocchio. Now my `CMakeLists.txt` is:

```cmake
cmake_minimum_required(VERSION 3.0.2)
project(pinocchio_ros_example)

find_package(catkin REQUIRED)

find_package(pinocchio REQUIRED PATHS /opt/openrobots/include)
message(WARNING ${pincchio_INCLUDE_DIRS})
catkin_package()

add_executable(example src/example.cpp)
target_link_libraries(example pinocchio::pinocchio)

install(TARGETS example
  RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)
```

When running `catkin_make --only-pkg-with-deps pinocchio_ros_example` I get the following error

```console
Base path: /home/crrl/catkin_ws
Source space: /home/crrl/catkin_ws/src
Build space: /home/crrl/catkin_ws/build
Devel space: /home/crrl/catkin_ws/devel
Install space: /home/crrl/catkin_ws/install
Whitelisted packages: pinocchio_ros_example
####
#### Running command: "make cmake_check_build_system" in "/home/crrl/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/crrl/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/openrobots/include;/opt/ros/noetic;/opt/openrobots/include
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/crrl/catkin_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- Using CATKIN_WHITELIST_PACKAGES: pinocchio_ros_example
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - pinocchio_ros_example
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'pinocchio_ros_example'
-- ==> add_subdirectory(pinocchio_ros_example)
CMake Warning at pinocchio_ros_example/CMakeLists.txt:7 (message):

-- Configuring done
CMake Error at pinocchio_ros_example/CMakeLists.txt:10 (add_executable):
  Target "example" links to target "pinocchio::pinocchio" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
make: *** [Makefile:348: cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```

wxmerkt, 2022-08-01:
Did you delete the build directory / clean prior to rebuilding? It may still point to the old configuration. Also I noticed my earlier message has a typo, it's missing an "o" in pinocchio_INCLUDE_DIRS

hh2564, 2022-08-01:
Hi I was able to solve this problem by creating another workspace and have the example compiled and running in the new workspace. I'm wondering if you know what might be the problem with the old workspace that could cause this error?  Thank you so much for your help!

jcarpent, 2022-08-02:
Thanks @wxmerkt for solving this issue.

closed 2022-08-02 by jcarpent

wxmerkt, 2022-08-02:
Hi @hh2564, happy to hear you have been able to resolve things! I dont have a way of telling what was wrong with the other workspace based on the available information