stack-of-tasks/pinocchio #1725: Segmentation Fault during Collision Checking

arjung128, 2022-08-06:
Hi, I have been experiencing strange behavior during collision checking. For some specific qpos-point combinations, I have been getting a segmentation fault . An example is the following:

```
import hppfcl
import pinocchio as pin
import numpy as np

# collison checking
model, geom_model, visual_model = pin.buildModelsFromUrdf('misc_data/panda/panda.urdf',
                                                          'misc_data/panda/',
                                                          pin.JointModelFreeFlyer())

# collision
geom_model.addAllCollisionPairs()
srdf_filename = 'misc_data/panda/pandafix.srdf'
pin.removeCollisionPairs(model, geom_model, srdf_filename)
data, geom_data, visual_data = pin.createDatas(model, geom_model, visual_model)

# robot
q = np.array([-0.75, 0.01, 0.5, 0., 0., -0.70710678, 0.70710678, -1.44568566, -1.26939072, -2.89194326, -0.8701567, -1.66512304, 1.27484547, -1.78646917, 0.03999, 0.03999])

# env
new_pts = np.array([[-0.69438231, -0.09979469,  0.83098379]])
oct_tree  = hppfcl.makeOctree(new_pts, 0.001)
t1 = hppfcl.Transform3f(np.eye(3), np.array([0., 0., 0.]))
world_shapes = [oct_tree]
world_transforms = [t1]

# collision checking
pin.updateGeometryPlacements(model, data, geom_model, geom_data, q)
collisions = np.zeros((len(world_shapes), len(geom_model.geometryObjects)), dtype=bool)
for i, (ws, wt) in enumerate(zip(world_shapes, world_transforms)):
    for k in range(len(geom_model.geometryObjects)):
        oM = geom_data.oMg[k]
        o = geom_model.geometryObjects[k].geometry

        res = hppfcl.CollisionResult()
        req = hppfcl.CollisionRequest()
        hppfcl.collide(o, oM, ws, wt, req, res)
        collision = res.isCollision()
        collisions[i,k] = collision

if np.any(collisions):
    print('collision')
```

Here is the error I am getting:

```
python: /projects/bbhm/saurabhg/drawers/fresh-3/libs/hpp-fcl/src/narrowphase/gjk.cpp:357: void hpp::fcl::details::getSupportFuncTpl(const hpp::fcl::details::MinkowskiDiff&, const Vec3f&, bool, hpp::fcl::Vec3f&, hpp::fcl::Vec3f&, hpp::fcl::support_func_guess_t&, hpp::fcl::details::MinkowskiDiff::ShapeData*) [with Shape0 = hpp::fcl::Box; Shape1 = hpp::fcl::TriangleP; bool TransformIsIdentity = true; hpp::fcl::Vec3f = Eigen::Matrix<double, 3, 1>; hpp::fcl::support_func_guess_t = Eigen::Matrix<int, 2, 1>]: Assertion `NeedNormalizedDir || dir.cwiseAbs().maxCoeff() >= 1e-6' failed.
Aborted (core dumped)
```

Interestingly, if either q[8] changes by a little bit, or the point changes by a little bit, the segmentation fault goes away. When I visualize the qpos and the point, the point seems to lie exactly on the surface of one of the links of the robot arm.

Any ideas why this may be happening, and what can be done to resolve it? Thanks in advance!

jcarpent, 2022-08-06:
Could you provide all the scripts and data needed to reproduce this potential bug and the versions of hppfcl and Pinocchio used?

jcarpent, 2022-08-06:
Certainly, some mesh of your robots are not convex and then, leading to a failure of the GJK algorithm

arjung128, 2022-08-06:
Here is a zip file containing the contents of misc_data/panda, this should have all the data needed to reproduce the bug
[panda.zip](https://github.com/stack-of-tasks/pinocchio/files/9274538/panda.zip).

Running the script above with this data should reproduce the bug.

I am on pinocchio version `2.5.6` and hppfcl version `1.7.8`.

jcarpent, 2022-08-06:
You are working with old versions of both packages

jcarpent, 2022-08-06:
Could you use Pinocchio 2.6.8 and hppfcl 2.1.2?

jcarpent, 2022-08-06:
And then report the new issues if any?

arjung128, 2022-08-06:
Are the two latest versions compatible? When I try to build using conda using the following commands:

```
conda create -n hppfcl python=3.9
conda activate hppfcl
conda install -c conda-forge hpp-fcl # installs hpp-fcl-2.1.2
conda install -c conda-forge pinocchio
```

I get:

```
The following packages will be DOWNGRADED:
    hpp-fcl                              2.1.2-py39h76655d1_0 --> 1.6.0-py39hbcdfc36_4
```

Is this expected? Or do I have to build from source to get the latest versions of both?

jcarpent, 2022-08-06:
There is something weird on your install. It should move down to hpp-fcl 2.01, not 1.6.0. On which OS are you?

arjung128, 2022-08-06:
Ubuntu 18.04

arjung128, 2022-08-06:
I just tried Ubuntu 20.04, and I'm getting conflicts there as well:

```
conda install -c conda-forge pinocchio

Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: |
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                                                                                                  \

UnsatisfiableError: The following specifications were found to be incompatible with a past
explicit spec that is not an explicit spec in this operation (hpp-fcl):

  - pinocchio -> hpp-fcl[version='>=1.1.3,<1.2.0a0|>=1.2.0,<1.3.0a0|>=1.2.1,<1.3.0a0|>=1.2.2,<1.3.0a0|>=1.3.0,<1.4.0a0|>=1.4.3,<1.5.0a0|>=1.4.4,<1.5.0a0|>=1.4.5,<1.5.0a0|>=1.4.6,<1.5.0a0|>=1.5.1,<1.6.0a0|>=1.5.2,<1.6.0a0|>=1.5.3,<1.6.0a0|>=1.6.0,<1.7.0a0|>=1.7.1,<1.8.0a0|>=1.7.3,<1.8.0a0|>=1.7.4,<1.8.0a0|>=1.7.5,<1.8.0a0|>=1.7.7,<1.8.0a0|>=1.8.0,<1.9.0a0|>=1.8.1,<1.9.0a0|>=2.0.1,<2.1.0a0|>=1.7.8,<1.8.0a0']

The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package _libgcc_mutex conflicts for:
tk -> libgcc-ng[version='>=9.4.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
_libgcc_mutex
libopenblas -> _openmp_mutex=[build=*_llvm] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
hpp-fcl==2.1.2 -> libgcc-ng[version='>=12'] -> _libgcc_mutex==0.1=conda_forge
eigenpy -> libgcc-ng[version='>=12'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
libzlib -> libgcc-ng[version='>=12'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
lz4-c -> libgcc-ng[version='>=9.3.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
sqlite -> libgcc-ng[version='>=12'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
libblas -> libgcc-ng[version='>=9.3.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
numpy -> libgcc-ng[version='>=12'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
libcblas -> libgcc-ng[version='>=9.3.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
python=3.9 -> libgcc-ng[version='>=12'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
icu -> libgcc-ng[version='>=10.3.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
ncurses -> libgcc-ng[version='>=10.3.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
libuuid -> libgcc-ng[version='>=9.3.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
xz -> libgcc-ng[version='>=7.5.0'] -> _libgcc_mutex[version='*|0.1|0.1',build='conda_forge|main|main']
eigen -> libgcc-ng[version='>=9.4.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
libgcc-ng -> _libgcc_mutex[version='*|0.1|0.1',build='conda_forge|main|main']
zlib -> libgcc-ng[version='>=12'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
_openmp_mutex -> _libgcc_mutex==0.1[build='conda_forge|main']
boost -> libgcc-ng[version='>=10.3.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
libffi -> libgcc-ng[version='>=9.4.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
bzip2 -> libgcc-ng[version='>=9.3.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
assimp -> libgcc-ng[version='>=12'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
libgomp -> _libgcc_mutex==0.1[build='conda_forge|main']
libnsl -> libgcc-ng[version='>=9.4.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
boost-cpp -> libgcc-ng[version='>=10.3.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
liblapack -> libgcc-ng[version='>=9.3.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
readline -> libgcc-ng[version='>=12'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
openssl -> libgcc-ng[version='>=12'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
octomap -> libgcc-ng[version='>=9.3.0'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
zstd -> libgcc-ng[version='>=12'] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']
pinocchio -> _openmp_mutex=[build=*_llvm] -> _libgcc_mutex[version='*|0.1',build='main|conda_forge|main']

Package ncurses conflicts for:
boost -> python[version='>=3.9,<3.10.0a0'] -> ncurses[version='5.9.*|5.9|>=6.1,<7.0.0a0|>=6.2,<7.0.0a0|>=6.3,<7.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0|>=6.2,<6.3.0a0|>=6.0,<7.0a0|6.0.*']
sqlite -> readline[version='>=8.0,<9.0a0'] -> ncurses[version='6.0.*|>=6.0,<7.0a0|>=6.1,<7.0a0']
numpy -> python[version='>=3.9,<3.10.0a0'] -> ncurses[version='5.9.*|5.9|>=6.1,<7.0.0a0|>=6.2,<6.3.0a0|>=6.2,<7.0.0a0|>=6.3,<7.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0|>=6.0,<7.0a0|6.0.*']
wheel -> python[version='!=3.0,!=3.1,!=3.2,!=3.3,!=3.4'] -> ncurses[version='5.9.*|5.9|>=6.1,<7.0.0a0|>=6.2,<6.3.0a0|>=6.3,<7.0a0|>=6.2,<7.0.0a0|>=6.2,<7.0a0|>=6.1,<7.0a0|>=6.0,<7.0a0|6.0.*']
[... cut ...]
  - readline -> libgcc-ng[version='>=9.3.0'] -> __glibc[version='>=2.17']
  - sqlite -> libgcc-ng[version='>=9.3.0'] -> __glibc[version='>=2.17']
  - tk -> libgcc-ng[version='>=7.5.0'] -> __glibc[version='>=2.17']
  - xz -> libgcc-ng[version='>=7.5.0'] -> __glibc[version='>=2.17']
  - zlib -> libgcc-ng[version='>=7.5.0'] -> __glibc[version='>=2.17']
  - zstd -> libgcc-ng[version='>=9.3.0'] -> __glibc[version='>=2.17']

Your installed version is: 2.31

Note that strict channel priority may have removed packages required for satisfiability.
```

jcarpent, 2022-08-06:
Create a new env. It will be fine.

jcarpent, 2022-08-06:
> Create a new env. It will be fine.

And please update conda, everything seems very old. Check with conda info.

arjung128, 2022-08-06:
Ok, I do not get the segmentation fault with Pinocchio `2.6.8` and hppfcl `2.1.2`. Thank you so much for your help!

closed 2022-08-06 by arjung128

jcarpent, 2022-08-07:
> Ok, I do not get the segmentation fault with Pinocchio `2.6.8` and hppfcl `2.1.2`. Thank you so much for your help!

Sounds great. Happy to see it works well.