stack-of-tasks/pinocchio #1726: Cmake error when building from source

arjung128, 2022-08-06:
Hi, I have been trying to build pinocchio from source but am running into an issue when I cmake. Here are the commands I'm using:

```
git clone --single-branch -b v2.6.8 --depth 1 https://github.com/stack-of-tasks/pinocchio.git
cd pinocchio && git submodule update --init --recursive && mkdir build && cd build
LD_LIBRARY_PATH=/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib/ cmake .. -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=OFF -DBUILD_PYTHON_INTERFACE=ON -DBUILD_WITH_AUTODIFF_SUPPORT=ON -DBUILD_WITH_URDF_SUPPORT=ON -DCMAKE_INSTALL_PREFIX=/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/ -Durdfdom_DIR=/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib/urdfdom/cmake -Durdfdom_headers_DIR=/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib/urdfdom_headers/cmake -Dconsole_bridge_DIR=/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib/console_bridge/cmake -Dcppad_LIBRARY=/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib -Dcppad_INCLUDE_DIR=/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/include -Deigenpy_DIR=/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib/cmake/eigenpy -Doctomap_DIR=/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/share/octomap -Dhpp-fcl_DIR=/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib/cmake/hpp-fcl/ -DBUILD_WITH_COLLISION_SUPPORT=On -DPYTHON_EXECUTABLE=$(which python)
```

And here's the output:

```
-- Package version (ROS package.xml): 2.6.8
-- Since urdfdom >= 1.0.0, the default C++ standard is C++11. The project is then compiled with C++11 standard.
-- Found Boost: /usr/include (found version "1.71.0") found components: filesystem serialization system
-- The Python bindings of Pinocchio will be compiled along the main library. If you want to disable this feature, please set the option BUILD_PYTHON_INTERFACE to OFF.
-- PythonLibraryDirs: /home/arjung2/anaconda3/lib
-- PythonLibVersionString: 3.8.3
-- Python site lib: lib/python3.8/site-packages
-- Found Boost: /usr/include (found version "1.71.0") found components: python38
-- Boost_PYTHON_LIBRARY: /usr/lib/x86_64-linux-gnu/libboost_python38.so
-- eigenpy FOUND. eigenpy at /data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib/libeigenpy.so
CMake Warning (dev) at /data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib/cmake/eigenpy/python.cmake:68 (message):
  Macro FINDPYTHON has already been called.  Several call to FINDPYTHON may
  not find the same Python version (for a yet unknown reason).
Call Stack (most recent call first):
  /data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib/cmake/eigenpy/eigenpyConfig.cmake:142 (FINDPYTHON)
  cmake/package-config.cmake:97 (find_package)
  CMakeLists.txt:155 (ADD_PROJECT_DEPENDENCY)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- PythonLibraryDirs: /home/arjung2/anaconda3/lib
-- PythonLibVersionString: 3.8.3
-- Python site lib: lib/python3.8/site-packages
-- Python include dirs: /home/arjung2/anaconda3/include/python3.8
-- checking for numpy
--   NUMPY_INCLUDE_DIRS=/home/arjung2/anaconda3/lib/python3.8/site-packages/numpy/core/include
--   NUMPY_VERSION=1.21.2
-- Python compiler: CPython
-- hpp-fcl FOUND. hpp-fcl at /data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib/libhpp-fcl.so
-- Found Boost: /usr/include (found version "1.71.0") found components: chrono serialization
-- Default C++ standard: 201402
-- C++ standard sufficient: Minimal required 11, currently defined: 11
-- hpp-fcl Python bindings NOT found.
-- Found Boost: /usr/include (found version "1.71.0") found components: unit_test_framework
-- pybind11 v2.8.0
-- CXX_STANDARD for cpp2pybind11 changed from 11 to 14
-- Doxygen rendering: using MathJax backend
-- Configuring done
CMake Error at bindings/python/CMakeLists.txt:134 (ADD_LIBRARY):
  Target "pinocchio_pywrap" links to target "Python3::NumPy" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at cmake/test.cmake:84 (ADD_EXECUTABLE):
  Target "test-cpp-python_parser" links to target "Python3::NumPy" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  unittest/CMakeLists.txt:24 (ADD_UNIT_TEST)
  unittest/CMakeLists.txt:99 (ADD_PINOCCHIO_UNIT_TEST)

CMake Error at unittest/python/pybind11/CMakeLists.txt:12 (add_library):
  Target "cpp2pybind11" links to target "Python3::NumPy" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at bindings/python/CMakeLists.txt:134 (ADD_LIBRARY):
  Target "pinocchio_pywrap" links to target "Python3::NumPy" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

WARNING: Target "test-cpp-cppad-joint-configurations" requests linking to directory "/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test-cpp-cppad-spatial" requests linking to directory "/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test-cpp-cppad-basic" requests linking to directory "/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test-cpp-cppad-algo" requests linking to directory "/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test-cpp-cppad-joints" requests linking to directory "/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test-cpp-cppad-algo-derivatives" requests linking to directory "/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "timings-derivatives" requests linking to directory "/data02/arjung2/drawers/fresh-yosemite-v4/libs-install/lib".  Targets may link only to libraries.  CMake is dropping the item.
-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

This is on Ubuntu 20.04. Any ideas what might be happening and what can be done to resolve this? Thanks in advance!

jcarpent, 2022-08-07:
Please use the devel branch for that, which contains the fix.

closed 2022-08-07 by jcarpent

FenglongSong, 2022-08-08:
Hi Justin @jcarpent thanks for the reply. I also encountered this issue when I tried to build the pinocchio3-preview branch since I'd like to try the casadi support in pinocchio3. Will the fix for pinocchio3 be released recently?

khaninger, 2022-08-29:
Hello, I am having the same issue and this is blocking using Casadi on Pinocchio. I have tried adding find_package calls for numpy to the named CMakeLists, but no success yet.

jcarpent, 2022-08-29:
I've just made a fix on the dedicated branch

khaninger, 2022-08-29:
It built! Thanks a lot, I really appreciate both the library and how responsive you all are to the community!

jcarpent, 2022-08-29:
Thanks @khaninger for this nice and positive feedback. 
We really appreciate it.