stack-of-tasks/pinocchio #1728: in conda (spyder) No module named 'pinocchio.pinocchio_pywrap'

fenaux, 2022-08-09:
I installed Pinocchio with
sudo apt-get install robotpkg-py38-pinocchio

then add the path to my conda environment with
conda develop /opt/openrobots/lib/python3.8/site-packages

but when running "import pinocchio" I get
   import pinocchio

  File /opt/openrobots/lib/python3.8/site-packages/pinocchio/__init__.py:6 in <module>
    from .pinocchio_pywrap import *

ModuleNotFoundError: No module named 'pinocchio.pinocchio_pywrap'

Here is the content of  opt/openrobots/lib/python3.8/site-packages/pinocchio :
deprecated.py   __init__.py                                            __pycache__       shortcuts.py
deprecation.py  pinocchio_pywrap.cpython-38-x86_64-linux-gnu.so        robot_wrapper.py  utils.py
explog.py       pinocchio_pywrap.cpython-38-x86_64-linux-gnu.so.2.6.8  romeo_wrapper.py  visualize

I guess the issue is about an equivalent of 
export PATH=/opt/openrobots/bin:$PATH
export PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig:$PKG_CONFIG_PATH
export LD_LIBRARY_PATH=/opt/openrobots/lib:$LD_LIBRARY_PATH
export CMAKE_PREFIX_PATH=/opt/openrobots:$CMAKE_PREFIX_PATH
in conda environment I tried with conda develop but always got the same error message

I use Ubuntu 20-04 and Python 3.9

Thanks for your help

jcarpent, 2022-08-09:
You should install Pinocchio via conda and not via apt. These are two different package managers and as far I understand, you want to use spyder with conda integration.

jcarpent, 2022-08-09:
In addition, be careful, that you are using Python 3.9 while you have installed Pinocchio bindings for Python 3.8 (the default version on your Ubuntu)

fenaux, 2022-08-09:
Many Thanks for your prompt answer.

It's not my wish to use Pinocchio bindings for Python 3.8. It's just the version name I saw when reading some of the previous issues. If there is a better match with Python 3.9 I will be happy to use it
If you are kind enough to give me the command line to install pinocchio with conda ;)...

fenaux, 2022-08-10:
Thanks, I indeed find the right repo
I was directed to this adress https://stack-of-tasks.github.io/pinocchio/ by a link in https://github.com/zongmianli/Estimating-3D-Motion-Forces.
May be you can add a warning in this page as it seems deprecated

Well I removed previous installation with sudo apt remove robotpkg-\*
and the paths with conda-develop -u path_name

But I am afraid something is going wrong as conda still tries to solve the environment for more that 16 hours now. :(

jcarpent, 2022-08-10:
Use mamba then.

jcarpent, 2022-08-10:
> May be you can add a warning in this page as it seems deprecated

Could you open a PR and make some suggestions for further readers?

nim65s, 2022-08-10:
This is not deprecated, just another way to install, using the standard Debian / Ubuntu packages and package manager. 

in your case, as you want Python 3.9, this can't work with the standard packages from your ubuntu 20.04, because ubuntu 20.04 provides Python 3.8.

fenaux, 2022-08-10:
I got the same difficulties to solve the environment to install namba
Then I decided to clean my environment with conda update anaconda
And then ran again conda install pinocchio -c conda-forge and it works :+1 :)

Many thanks for you support(s)