stack-of-tasks/pinocchio #1729: Built from source, encoutered ImportError: libeigenpy.so: cannot open shared object file: No such file or directory

FenglongSong, 2022-08-09:
Dear Pinocchio team,

First of all thanks for the great project. I'm trying to use pinocchio3 casadi-python interface. And when I run a simple script, the first line `import pinocchio` fails.
```console
fenglongsong@ubuntu:~/Desktop/test_algo$ python3 test.py 
Traceback (most recent call last):
  File "test.py", line 1, in <module>
    import pinocchio
  File "/usr/local/lib/python3/dist-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap_default import *
ImportError: libeigenpy.so: cannot open shared object file: No such file or directory
```
I was following the tutorial on Pinocchio website [here](https://stack-of-tasks.github.io/pinocchio/download.html)  and I did add these lines into `~/.bashrc`:
```sh
export PATH=/usr/local/bin:$PATH
export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig:$PKG_CONFIG_PATH
export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
export PYTHONPATH=/usr/local/lib/python3/dist-packages:$PYTHONPATH
export CMAKE_PREFIX_PATH=/usr/local:$CMAKE_PREFIX_PATH

export PATH=/opt/openrobots/bin:$PATH
export PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig:$PKG_CONFIG_PATH
export LD_LIBRARY_PATH=/opt/openrobots/lib:$LD_LIBRARY_PATH
export PYTHONPATH=/opt/openrobots/lib/python3.8/site-packages:$PYTHONPATH
export CMAKE_PREFIX_PATH=/opt/openrobots:$CMAKE_PREFIX_PATH
```
I'm suscepting this is related to the possible error in installing Pinocchio by build from the source. When I was doing the `cmake ..` step, this is what I got:
```console
fenglongsong@ubuntu:~/Documents/pinocchio3/pinocchio/build(pinocchio3-preview)$ cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DPYTHON_EXECUTABLE=/usr/bin/python3
-- Package version (ROS package.xml): 2.9.1
-- Default C++ standard: 201402
-- C++ standard sufficient: Minimal required 11, currently defined: 14
-- Found Boost: /usr/include (found version "1.71.0") found components: filesystem serialization system 
-- The Python bindings of Pinocchio will be compiled along the main library. If you want to disable this feature, please set the option BUILD_PYTHON_INTERFACE to OFF.
-- Found Python3: /usr/bin/python3 (found version "3.8.10") found components: Interpreter Development 
-- PythonLibraryDirs: /usr/lib/x86_64-linux-gnu
-- PythonLibVersionString: 3.8.10
-- Python site lib: lib/python3/dist-packages
-- Found Boost: /usr/include (found version "1.71.0") found components: python38 
-- Boost_PYTHON_LIBRARY: /usr/lib/x86_64-linux-gnu/libboost_python38.so
-- eigenpy FOUND. eigenpy at /opt/openrobots/lib/libeigenpy.so
CMake Warning (dev) at /opt/openrobots/lib/cmake/eigenpy/python.cmake:68 (message):
  Macro FINDPYTHON has already been called.  Several call to FINDPYTHON may
  not find the same Python version (for a yet unknown reason).
Call Stack (most recent call first):
  /opt/openrobots/lib/cmake/eigenpy/eigenpyConfig.cmake:142 (FINDPYTHON)
  cmake/package-config.cmake:97 (find_package)
  CMakeLists.txt:163 (ADD_PROJECT_DEPENDENCY)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found Python3: /usr/bin/python3 (found version "3.8.10") found components: Interpreter NumPy Development 
-- PythonLibraryDirs: /usr/lib/x86_64-linux-gnu
-- PythonLibVersionString: 3.8.10
-- Python site lib: lib/python3/dist-packages
-- Python include dirs: /usr/include/python3.8
-- checking for numpy
--   NUMPY_INCLUDE_DIRS=/usr/local/lib/python3.8/dist-packages/numpy/core/include
--   NUMPY_VERSION=1.23.1
-- NumPy include dir: /usr/local/lib/python3.8/dist-packages/numpy/core/include
-- checking for numpy
--   NUMPY_INCLUDE_DIRS=/usr/local/lib/python3.8/dist-packages/numpy/core/include
--   NUMPY_VERSION=1.23.1
-- Python compiler: CPython
-- Found Boost: /usr/include (found version "1.71.0") found components: unit_test_framework 
-- Doxygen rendering: using MathJax backend
-- Configuring done
-- Generating done
-- Build files have been written to: /home/fenglongsong/Documents/pinocchio3/pinocchio/build
```
Clearly there is a warning information about eigenpy. And when I run `make -j4`, there was long information in the terminal and I'm taking the last a few lines here, since the information has been repeatedly saying `Entering directory XXX` or `Leaving directory XXX`.

```console
[ 98%] Built target example-cpp-geometry-models
make[2]: Entering directory '/home/fenglongsong/Documents/pinocchio3/pinocchio/build'
make[2]: Nothing to be done for 'examples/CMakeFiles/example-cpp-forward-dynamics-derivatives.dir/build'.
make[2]: Leaving directory '/home/fenglongsong/Documents/pinocchio3/pinocchio/build'
make[2]: Entering directory '/home/fenglongsong/Documents/pinocchio3/pinocchio/build'
make[2]: Nothing to be done for 'examples/CMakeFiles/example-cpp-inverse-dynamics-derivatives.dir/build'.
make[2]: Leaving directory '/home/fenglongsong/Documents/pinocchio3/pinocchio/build'
[100%] Built target example-cpp-forward-dynamics-derivatives
[100%] Built target example-cpp-inverse-dynamics-derivatives
make[1]: Leaving directory '/home/fenglongsong/Documents/pinocchio3/pinocchio/build'
/usr/bin/cmake -E cmake_progress_start /home/fenglongsong/Documents/pinocchio3/pinocchio/build/CMakeFiles 0
```
Would you guys have any advice to fix this? Besides, my purpose of build from source is because I want to use the python bindings in pinocchio3, especially `pinocchio.casadi`. Do you think in this case building from source is necessary? Will there be other ways to achieve this? Thanks a lot!

jcarpent, 2022-08-09:
Could you display `echo $LD_LIBRARY_PATH`?

jcarpent, 2022-08-09:
On which version of ubuntu are you?

FenglongSong, 2022-08-09:
Hi Justin, by `echo $LD_LIBRARY_PATH`:
```console
fenglongsong@ubuntu:~/Documents/pinocchio3/pinocchio/build(pinocchio3-preview)$ echo $LD_LIBRARY_PATH
/opt/openrobots/lib:/usr/local/lib:
```
I'm using Ubuntu 20.04

jcarpent, 2022-08-09:
can you show the output of `ls /opt/openrobots/lib`?

FenglongSong, 2022-08-09:
fenglongsong@ubuntu:~/Documents/pinocchio3/pinocchio/build(pinocchio3-preview)$ ls /opt/openrobots/lib
cmake  libeigenpy.so  libhpp-fcl.so  libpinocchio.so  libpinocchio.so.2.6.8  pkgconfig  python3.8

jcarpent, 2022-08-09:
`ldd /usr/local/lib/python3/dist-packages/pinocchio/pinocchio_pywrap_default.so`?

jcarpent, 2022-08-09:
It is weird because you have Pinocchio 2.6.8 also installed.

jcarpent, 2022-08-09:
You need to first deinstall pinocchio 2.6.8 of your system.

jcarpent, 2022-08-09:
ldd /usr/local/lib/python3/dist-packages/pinocchio/pinocchio_pywrap_default.cpython-38-x86_64-linux-gnu.so

FenglongSong, 2022-08-09:
```console
fenglongsong@ubuntu:~/Documents/pinocchio3/pinocchio/build(pinocchio3-preview)$ ldd /usr/local/lib/python3/dist-packages/pinocchio/pinocchio_pywrap_default.cpython-38-x86_64-linux-gnu.so
	linux-vdso.so.1 (0x00007fff067e8000)
	libpinocchio.so.2.9.1 => /usr/local/lib/libpinocchio.so.2.9.1 (0x00007fbd23d17000)
	libeigenpy.so => /opt/openrobots/lib/libeigenpy.so (0x00007fbd235a3000)
	libboost_python38.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_python38.so.1.71.0 (0x00007fbd23523000)
	libboost_serialization.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_serialization.so.1.71.0 (0x00007fbd234df000)
	liburdfdom_world.so.1.0 => /lib/x86_64-linux-gnu/liburdfdom_world.so.1.0 (0x00007fbd234b9000)
	libconsole_bridge.so.0.4 => /lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007fbd234b1000)
	libpython3.8.so.1.0 => /lib/x86_64-linux-gnu/libpython3.8.so.1.0 (0x00007fbd22f5b000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fbd22d79000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fbd22c2a000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fbd22c0f000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fbd22a1d000)
	libboost_filesystem.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007fbd229fd000)
	liburdfdom_model.so.1.0 => /lib/x86_64-linux-gnu/liburdfdom_model.so.1.0 (0x00007fbd229d8000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fbd229b5000)
	libtinyxml.so.2.6.2 => /lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007fbd2299d000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007fbd2296f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fbd22951000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fbd2294b000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007fbd22946000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fbd25470000)
```

jcarpent, 2022-08-10:
@nim65s Do you know what happens?
I'm pretty sure the env used by @FenglongSong is not clean, but I do not know how to fix it.
Yet, I do think @FenglongSong you should instead use conda to be sure to have isolated Python env or simply reinstall your OS to start from a fresh env (or use Docker).

In the meantime, I will close this issue as it is not related to Pinocchio itself, but it is more related to a local user setup.

closed 2022-08-10 by jcarpent

nim65s, 2022-08-10:
I'll double check pinocchio3-preview with our recent packaging changes, and provide a clean procedure here to @FenglongSong in the day.

nim65s, 2022-08-10:
Ok, so here is a Dockerfile which work:
```Dockerfile
FROM ubuntu:20.04

ADD http://robotpkg.openrobots.org/packages/debian/robotpkg.key /

RUN --mount=type=cache,sharing=locked,target=/var/cache/apt \
    --mount=type=cache,sharing=locked,target=/var/lib/apt \
    apt-get update -y && DEBIAN_FRONTEND=noninteractive apt-get install -qqy \
    build-essential \
    cmake \
    git \
    gnupg2 \
    liburdfdom-dev \
    python-is-python3 \
 && echo "deb [arch=amd64] http://robotpkg.openrobots.org/packages/debian/pub focal robotpkg" \
 >> /etc/apt/sources.list.d/robotpkg.list \
 && apt-key add /robotpkg.key \
 && apt-get update -y && DEBIAN_FRONTEND=noninteractive apt-get install -qqy \
    robotpkg-py38-casadi \
    robotpkg-py38-hpp-fcl

ENV CMAKE_PREFIX_PATH=/opt/openrobots

RUN git clone --recursive -b pinocchio3-preview https://github.com/stack-of-tasks/pinocchio \
 && cmake -B build -S pinocchio -DBUILD_WITH_COLLISION_SUPPORT=ON -DBUILD_WITH_CASADI_SUPPORT=ON \
 && cmake --build build -j 4 \
 && cmake --build build -t install

 RUN python -c "import pinocchio"
 ```
 
Is this helping, @FenglongSong ?

FenglongSong, 2022-08-10:
@nim65s Thanks a lot! I will have a try today and let you know if I have any updates.

FenglongSong, 2022-08-10:
Hi @nim65s , thanks for your nice help. The docker file you provided is working prefectly, `import pinocchio` is successful. But when I tired to use `import pinocchio.casadi as cpin`, it fails. Do you have any suggestions for this?
```console
Traceback (most recent call last):
  File "./test.py", line 2, in <module>
    import pinocchio.casadi as cpin
ModuleNotFoundError: No module named 'pinocchio.casadi'
```

nim65s, 2022-08-10:
Ok I forgot that when installing in `/usr/local`, python on ubuntu wants modules in `/usr/local/lib/python3.8/dist-packages`. This can be set with `-DPYTHON_STANDARD_LAYOUT=ON -DPYTHON_DEB_LAYOUT=ON` cmake options.

In other words, I added this at the end of the dockerfile:
```Dockerfile

RUN cmake --build build -t uninstall \
 && cmake -DPYTHON_STANDARD_LAYOUT=ON -DPYTHON_DEB_LAYOUT=ON build \
 && cmake --build build -t install

ENV LD_LIBRARY_PATH /opt/openrobots/lib

RUN python -c "import pinocchio.casadi"
```

But now I have:
```python
> [stage-0 7/7] RUN python -c "import pinocchio.casadi":                                                                                                                                                                                                                        
 Traceback (most recent call last):                                                                                                                                                                                                                                     
   File "<string>", line 1, in <module>                                                                                                                                                                                                                                 
   File "/usr/local/lib/python3.8/dist-packages/pinocchio/casadi/__init__.py", line 5, in <module>                                                                                                                                                                      
     from ..pinocchio_pywrap_casadi import *
 SystemError: initialization of pinocchio_pywrap_casadi raised unreported exception
 ```
 
 And for this issue, I don't have an answer. Maybe someone else can help, otherwise I'm afraid you'll have to wait for pinocchio 3 release.

FenglongSong, 2022-08-10:
I also tried the modified Dockerfile and had exactly the same error as you. It's a little disappointing but anyway thanks for your help!