stack-of-tasks/pinocchio #1733: 'GepettoVisualizer' object has no attribute 'viewer'

fenaux, 2022-08-12:
I continue my journey in pinocchio world

I could run the examples as described in example page of this repo.
But the ones with viewer give this error message :
 command line : python -i gepetto-viewer.py
display
/home/fenaux/Documents/robot/examples/gepetto-viewer.py:26: UserWarning: Error while starting the viewer client. Check whether gepetto-viewer is properly started
  viz.initViewer()
Error while loading the viewer model. It seems you should start gepetto-viewer
'GepettoVisualizer' object has no attribute 'viewer'
Traceback (most recent call last):
  File "/home/fenaux/Documents/robot/examples/gepetto-viewer.py", line 33, in <module> viz.loadViewerModel("pinocchio")
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 137, in loadViewerModel
    gui = self.viewer.gui
AttributeError: 'GepettoVisualizer' object has no attribute 'viewer'

I installed gepetto with conda : conda install gepetto-viewer-corba -c conda-forge as witten here https://github.com/stack-of-tasks/pinocchio/tree/devel/examples
If I run 
gepetto-gui directly in a console I obtain the window interface but I do not know how to load a model with it
while window's opening, this message appears 4 times in the console : libpng warning: iCCP: known incorrect sRGB profile

Thanks for your help

jcarpent, 2022-08-12:
Could you use meshcat solution instead?

fenaux, 2022-08-12:
I installed it with 
pip install meshcat

And error message are longer :(
python -i meshcat-viewer.py leads to

Traceback (most recent call last):
  File "/home/fenaux/anaconda3/lib/python3.9/runpy.py", line 188, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/home/fenaux/anaconda3/lib/python3.9/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/meshcat/__init__.py", line 5, in <module>
    from . import visualizer
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/meshcat/visualizer.py", line 7, in <module>
    from IPython.display import HTML
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/IPython/__init__.py", line 56, in <module>
    from .terminal.embed import embed
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/IPython/terminal/embed.py", line 16, in <module>
    from IPython.terminal.interactiveshell import TerminalInteractiveShell
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/IPython/terminal/interactiveshell.py", line 35, in <module>
    from .debugger import TerminalPdb, Pdb
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/IPython/terminal/debugger.py", line 6, in <module>
    from IPython.core.completer import IPCompleter
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/IPython/core/completer.py", line 146, in <module>
    import jedi
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/jedi/__init__.py", line 32, in <module>
    from jedi.api import Script, Interpreter, set_debug_function, preload_module
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/jedi/api/__init__.py", line 13, in <module>
    import parso
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/parso/__init__.py", line 42, in <module>
    from parso.grammar import Grammar, load_grammar
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/parso/grammar.py", line 13, in <module>
    from parso.cache import parser_cache, load_module, try_to_save_module
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/parso/cache.py", line 76, in <module>
    _default_cache_path = _get_default_cache_path()
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/parso/cache.py", line 73, in _get_default_cache_path
    return dir_.expanduser()
AttributeError: 'PosixPath' object has no attribute 'expanduser'

Traceback (most recent call last):
  File "/home/fenaux/Documents/robot/examples/meshcat-viewer.py", line 33, in <module>
    viz.initViewer(open=True)
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/pinocchio/visualize/meshcat_visualizer.py", line 99, in initViewer
    self.viewer = meshcat.Visualizer() if viewer is None else viewer
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/meshcat/visualizer.py", line 92, in __init__
    self.window = ViewerWindow(zmq_url=zmq_url, start_server=(zmq_url is None), server_args=server_args)
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/meshcat/visualizer.py", line 20, in __init__
    self.server_proc, self.zmq_url, self.web_url = start_zmq_server_as_subprocess(
  File "/home/fenaux/anaconda3/lib/python3.9/site-packages/meshcat/servers/zmqserver.py", line 70, in start_zmq_server_as_subprocess
    raise RuntimeError("the meshcat server process exited prematurely with exit code " + str(server_proc.poll()))
RuntimeError: the meshcat server process exited prematurely with exit code 1

jcarpent, 2022-08-12:
By the way, it seems that you have installed everything in the base env of conda, while you create a new one where you install stuff. Please see with your supervisor to get some tips.

fenaux, 2022-08-12:
Thanks for your prompt answers.
Yes everything is installed in base env of conda.

I am afraid I am my own supervisor ;)

jcarpent, 2022-08-12:
> I am afraid I am my own supervisor ;)

OK. Then just start from a fresh install, without mixing different package managers and use conda env for isolating your developments. This is the only help I can provide due to my limited bandwith.

fabinsch, 2022-08-12:
Hello @fenaux, you should get a clean conda base environment first. There are two options:
1. completely uninstall anaconda and install it again (recommend).
2. Use the conda install --revision (check how to use)

Afterwards, from a clean base environment you should create a specific conda environment for Pinocchio (see Section “Managing environments” 
https://conda.io/projects/conda/en/latest/user-guide/getting-started.html )

Now depending on what you want to do, inside the specific conda environment you do a conda install -c conda-forge pinocchio to get the master branch of Pinocchio installed. 

If you want to use the Pinocchio-3x-preview branch with some new features, you can do conda install -c conda-forge pinocchio --only-deps , then you clone the repo , switch to the desired branch and built from source. In this Pinocchio environment you then install Meshcat using again conda. I hope it helps you.