stack-of-tasks/pinocchio #1735: Pinocchio packages import issue on Pycharm

JackTony123, 2022-08-19:
Hello all! I met a problem recently when I used the Pinocchio library for my robot arm simulation on Mujoco. I downloaded the Pinocchio library through conda. I can find the package on my conda environment as "home/shine/anaconda3/envs/mujoco_py/lib/python3.8/site-packages/pinocchio"
 When I run the demo program on Ubuntu through Pycharm, something wrong shows on the terminal: 
```
from pinocchio.robot_wrapper import RobotWrapper
E   ModuleNotFoundError: No module named 'pinocchio.robot_wrapper'
```
I try to fix my environment variables by "**Run-> Edit Configuration**", Here are my environment variables setting below:
```
LD_LIBRARY_PATH $LD_LIBRARY_PATH:/home/shine/.mujoco/mujoco210/bin:$LD_LIBRARY_PATH:/usr/lib/nvidia:$LD_LIBRARY_PATH:/home/shine/anaconda3/envs/mujoco_py/lib
LD_PRELOAD
/usr/lib/x86_64-linux-gnu/libGLEW.so
PYTHONUNBUFFERED 1
```

Also, when I add "**pinocchio~=2.6.4**" on "requirements.txt", it shows in red color, that makes me feel confused because other packges name such as "ikpy==3.3.3", "numpy~=1.22.4" shows in green color. Does it mean that pycharm cannot find Pinocchio library? How can I figure out this problem? Thank you guys can give me any suggestion for that!

fabinsch, 2022-08-19:
Have a look here : https://www.jetbrains.com/help/pycharm/configuring-python-interpreter.html#interpreter

JackTony123, 2022-08-19:
> Hey, you're saying you installed Pinocchio via conda. In which conda environment did you install it ? Good practice is to create a specific environment for each of your projects and to leave the base environment clean.
> 
> Then you need to tell pycharm about this specific conda environment where you installed Pinocchio. You can do so by specifying the correct python version (from this environment) to run your script. I hope it can help you !

Hi, I installed conda 4.13.0 on my Ubuntu 20.04 system. I create an specific environment name as mujoco_py and use this environment only for mujoco simulation. Each time when I download related packages or libaraies with mujoco simulation, I would choose this environment only to make sure every package on this environment. Let me see your link and know whether is can work well for me. Thanks! I will update soon!

JackTony123, 2022-08-19:
> 
Hi Fabinsch, I made a double check for my Python interpreter on PyCharm. I already choose mujoco_py(conda environment) as my Python Interpreter. Also, I can find Pinocchio package installed on the list. What's the problem for that? Still feel confused about that.....

fabinsch, 2022-08-19:
Can you do an `import pinocchio as pin` before the import of the RobotWrapper ?

fabinsch, 2022-08-20:
Some steps you could do to further debug your problem is to open a terminal. Conda activate your mujoco environment , do
`conda list | grep pinocchio`

`python -c "import pinocchio"`

`python -c "from pinocchio.robot_wrapper import RobotWrapper"`

If this is successful do `which python` to verify you're using the same as in pycharm. 
If it doesn't work uninstall and reinstall Pinocchio again trough conda, make sure to wipe off all the Pinocchio installations that you might have already installed on your disk before reinstalling (you can search using commands like `find <path> -name *pinocchio*`)

JackTony123, 2022-08-22:
> Some steps you could do to further debug your problem is to open a terminal. Conda activate your mujoco environment , do `conda list | grep pinocchio`
>
> `python -c "import pinocchio"`
>
> `python -c [...]

Hi Fabinsch,

Thanks for your suggestion! I followed your suggestion step by step and made a double check for my project. The problem has been solved successfully by changing the python codes. Even though the Pinocchio in the requirment.txt shows in red color, the program can also run well. Thanks for your suggestion and have a nice day!

closed 2022-08-22 by nim65s