stack-of-tasks/pinocchio #1737: Relocatable Library

Booooooosh, 2022-08-22:
Hi, we have a use case where this library is built under one machine (A) and used on another machine (B). Unfortunately, machine A and B have different paths and machine B isn't able to properly load interface libraries since the path under `INTERFACE_INCLUDE_DIRECTORIES` are absolute path from machine A.

One potential root cause of this is during `TARGET_LINK_LIBRARIES` or `TARGET_INCLUDE_DIRECTORIES` we are getting library path from variables such as `${EIGEN3_INCLUDE_DIR}` [here](https://github.com/stack-of-tasks/pinocchio/blob/master/src/CMakeLists.txt#L56). While I'm digging into this I'm not able to find the definition of `MODERNIZE_TARGET_LINK_LIBRARIES`. Wonder if this is an known issue.

Thank you!

nim65s, 2022-08-23:
Hi,
This library should be easily relocatable.

Could you provide your:
- OS
- eigen version
- cmake version
- which absolute paths and where are they in the packages build on (A)

nim65s, 2022-08-23:
Well, clearly, if you still have `TARGET_INCLUDE_DIRECTORIES(${PROJECT_NAME}_headers SYSTEM INTERFACE ${EIGEN3_INCLUDE_DIR})` somewhere, this is can't be relocatable. You should update your version.

jcarpent, 2022-08-24:
@Booooooosh COuld check whether https://github.com/stack-of-tasks/pinocchio/pull/1738 fixes your issue?

nim65s, 2022-08-25:
If I understand correctly, #1737is already fixed by #1718, but @Booooooosh is not up to date.

jcarpent, 2022-08-25:
Not really, as the target exports were containing hard coded path. #1738 Solves this.

closed 2022-08-28 by jcarpent