stack-of-tasks/pinocchio #1739: "buildModelFromXML()" cannot be used to import .xml file for Pinocchino 2.6.4

JackTony123, 2022-08-26:
Hello all! Recently I met a new problem when I use the function "buildModelFromXML()" to import my model from .XML file. Here is my code below:
`xml_path = "/home/Tony/RR_robots/robot_description/UR5/UR5.xml"
model = pin.buildModelFromXML(xml_path)
data = model.createData()
`
When I run the program, it shows on terminal as
**ValueError: The XML stream does not contain a valid URDF model.**
For the UR5.xml file, I can make sure there is no problem with that, and I can run it on Mujoco directly. Also, in another project, I can import the XML file through python. I don't know what's the problem with this function. Is there something I did wrong or   Pinocchio 2.6.4 didn't support this function?

wxmerkt, 2022-08-26:
Can you provide the `UR5.xml` file for debugging?

JackTony123, 2022-08-26:
> 

Hi, May I send you through email? The file is a little bit large. My email: jackshine97@yahoo.com
Thanks for your help!
Anyone who would like to see the XML file can send email to me and I will send you. Thanks!

JackTony123, 2022-08-26:
> Can you provide the `UR5.xml` file for debugging?

Hi, May I send you through email? The file is a little bit large. My email: [jackshine97@yahoo.com](mailto:jackshine97@yahoo.com)
Thanks for your help!
Anyone who would like to see the XML file can send email to me and I will send you. Thanks!

wxmerkt, 2022-08-26:
Closing the issue as replied via email - the file in question is a MuJoCo MJCF file and not a URDF. Cf. https://github.com/stack-of-tasks/pinocchio/issues/1680 for MJCF support

closed 2022-08-26 by wxmerkt