stack-of-tasks/pinocchio #1740: ImportError: libpinocchio.so.2.6.9: cannot open shared object file and ModuleNotFoundError: No module named 'pinocchio.pinocchio_pywrap'

alexandre-blanc, 2022-09-01:
Hi, I seem to be facing an issue similar to #625, maybe also related to #1528.

I already described the issue in a comment in #625 but decided to file a new issue instead for visibility/traceability and also because the problem seem to be different.

I run ubuntu 18.04 (bionic) and primarily use python 3.8.

I installed pinocchio from robotpkg following the instructions on the stack-of-tasks website. I selected the py36 version.
I set up the environment variables accordingly in my .zshrc :
```
#setup paths for openrobots
export PATH=/opt/openrobots/bin:$PATH
export PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig:$PKG_CONFIG_PATH
export LD_LIBRARY_PATH=/opt/openrobots/lib:$LD_LIBRARY_PATH
export PYTHONPATH=/opt/openrobots/lib/python3.6/site-packages:$PYTHONPATH
export CMAKE_PREFIX_PATH=/opt/openrobots:$CMAKE_PREFIX_PATH
```

Then, I tried to import the pinocchio package from python 3.6 and python 3.8 interpreters, with the following result :

```
➜  python3  
Python 3.6.9 (default, Jun 29 2022, 11:45:57) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/openrobots/lib/python3.6/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ImportError: libpinocchio.so.2.6.9: cannot open shared object file: No such file or directory
>>> 
➜  python3.8
Python 3.8.13 (default, Apr 19 2022, 00:53:22) 
[GCC 7.5.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/openrobots/lib/python3.6/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ModuleNotFoundError: No module named 'pinocchio.pinocchio_pywrap'
>>> 
```

I checked the paths and the presence of the pinocchio_pywrap file :
```
➜  echo $PYTHONPATH
/opt/openrobots/lib/python3.6/site-packages:
➜  echo $LD_LIBRARY_PATH 
/opt/openrobots/lib:/opt/openrobots/lib:
➜  ls /opt/openrobots/lib/python3.6/site-packages/pinocchio 
deprecated.py   explog.py    pinocchio_pywrap.cpython-36m-x86_64-linux-gnu.so        __pycache__       romeo_wrapper.py  utils.py
deprecation.py  __init__.py  pinocchio_pywrap.cpython-36m-x86_64-linux-gnu.so.2.6.9  robot_wrapper.py  shortcuts.py      visualize
```

I also tried to look for a `libpinocchio.so.2.6.9` file to no avail :
```
➜  find / -path "/mnt/*" -prune -o -name "*pinocchio*" -print 2>/dev/null
/usr/share/doc/robotpkg-py36-pinocchio
/usr/share/doc/robotpkg-pinocchio
/var/lib/dpkg/info/robotpkg-py36-pinocchio.+SIZE_PKG
/var/lib/dpkg/info/robotpkg-py36-pinocchio.list
/var/lib/dpkg/info/robotpkg-py36-pinocchio.+BUILD_VERSION
/var/lib/dpkg/info/robotpkg-py36-pinocchio.+DESC
/var/lib/dpkg/info/robotpkg-pinocchio.+BUILD_INFO
/var/lib/dpkg/info/robotpkg-py36-pinocchio.md5sums
/var/lib/dpkg/info/robotpkg-pinocchio.+SIZE_PKG
/var/lib/dpkg/info/robotpkg-pinocchio.+SIZE_ALL
/var/lib/dpkg/info/robotpkg-pinocchio.md5sums
/var/lib/dpkg/info/robotpkg-pinocchio.+DESC
/var/lib/dpkg/info/robotpkg-py36-pinocchio.+SIZE_ALL
/var/lib/dpkg/info/robotpkg-py36-pinocchio.+CONTENTS
/var/lib/dpkg/info/robotpkg-pinocchio.list
/var/lib/dpkg/info/robotpkg-pinocchio.+CONTENTS
/var/lib/dpkg/info/robotpkg-py36-pinocchio.+BUILD_INFO
/var/lib/dpkg/info/robotpkg-pinocchio.+BUILD_VERSION
/var/lib/dpkg/info/robotpkg-py36-pinocchio.+COMMENT
/var/lib/dpkg/info/robotpkg-pinocchio.+COMMENT
/opt/openrobots/lib/python3.6/site-packages/pinocchio
/opt/openrobots/lib/python3.6/site-packages/pinocchio/pinocchio_pywrap.cpython-36m-x86_64-linux-gnu.so
/opt/openrobots/lib/python3.6/site-packages/pinocchio/pinocchio_pywrap.cpython-36m-x86_64-linux-gnu.so.2.6.9
/opt/openrobots/share/doc/debian/py36-pinocchio
```
My installation seems fairly clean to me and I don't see any obvious problems. I am not fluent in the build/linking process used to develop python bindings, and I can't figure out what my next step could be. Any help would be much appreciated.

jcarpent, 2022-09-01:
You can see that you are using Python 3.8 while your default system is 3.6.
It seems your system is a bit messy. Could you rather use conda, unset the env variables and everything will go fine.

jcarpent, 2022-09-01:
Could you show the content of /opt/openrobots/lib and /opt/openrobots/lib64?

alexandre-blanc, 2022-09-01:
I'd rather not use conda if an alternative exists. I typically use virtualenv and pip instead.
As you can see in the initial report, the import fails with python 3.6 as well (first promp, `python3`). I'll update with the contents of the lib and lib64 folders as soon as possible.

nim65s, 2022-09-01:
Hi @alexandre-blanc ,

I just tested an installation and use of robotpkg-py36-pinocchio on ubuntu 18.04, and it works for me:
```Dockerfile
FROM ubuntu:bionic

# Setup Robotpkg apt binary repository
ADD http://robotpkg.openrobots.org/packages/debian/robotpkg.key /
RUN --mount=type=cache,sharing=locked,target=/var/cache/apt --mount=type=cache,sharing=locked,target=/var/lib/apt \
    apt-get update -qqy && DEBIAN_FRONTEND=noninteractive apt-get install -qqy gnupg2 \
 && apt-key add /robotpkg.key \
 && echo "deb [arch=amd64] http://robotpkg.openrobots.org/packages/debian/pub bionic robotpkg" \
 >> /etc/apt/sources.list.d/robotpkg.list

# Install Pinocchio
RUN --mount=type=cache,sharing=locked,target=/var/cache/apt --mount=type=cache,sharing=locked,target=/var/lib/apt \
    apt-get update -qqy && DEBIAN_FRONTEND=noninteractive apt-get install -qqy \
    robotpkg-py36-pinocchio

# Configure python path ; this is equivalent to your PYTHONPATH=…
RUN echo "/opt/openrobots/lib/python3.6/site-packages" >> /usr/lib/python3/dist-packages/robotpkg.pth

# Test everything
RUN python3 -c "import pinocchio"
```

Can you check `apt update && apt upgrade && apt install robotpkg-py36-robotpkg` ? Maybe something went wrong during your installation.

jcarpent, 2022-09-02:
> I'd rather not use conda if an alternative exists. I typically use virtualenv and pip instead.

Honestly, I do think conda is much more suited, simpler and compatible with pip.

alexandre-blanc, 2022-09-02:
> Could you show the content of /opt/openrobots/lib and /opt/openrobots/lib64?

There is no directory named lib64 :
```
➜  ~ ls /opt/openrobots/lib
cmake                          libgepetto-viewer.so           libomniORB4.so.2            libOpenThreads.so.21      libosgManipulator.so.161     libosgSim.so.161        libosgUtil.so.161
gepetto-gui-plugins            libomniCodeSets4.a             libomniORB4.so.2.4          libOpenThreads.so.3.3.1   libosgManipulator.so.3.6.5   libosgSim.so.3.6.5      libosgUtil.so.3.6.5
libcollada-dom2.4-dp.so        libomniCodeSets4.so            libomnithread.a             libosgAnimation.so        libosgParticle.so            libosg.so               libosgViewer.so
libcollada-dom2.4-dp.so.0      libomniCodeSets4.so.2          libomnithread.so            libosgAnimation.so.161    libosgParticle.so.161        libosg.so.161           libosgViewer.so.161
libcollada-dom2.4-dp.so.2.4.0  libomniCodeSets4.so.2.4        libomnithread.so.4          libosgAnimation.so.3.6.5  libosgParticle.so.3.6.5      libosg.so.3.6.5         libosgViewer.so.3.6.5
libCOS4.a                      libomniConnectionMgmt4.a       libomnithread.so.4.1        libosgDB.so               libosgPresentation.so        libosgTerrain.so        libosgVolume.so
libCOS4.so                     libomniConnectionMgmt4.so      libomniZIOP4.a              libosgDB.so.161           libosgPresentation.so.161    libosgTerrain.so.161    libosgVolume.so.161
libCOS4.so.2                   libomniConnectionMgmt4.so.2    libomniZIOP4.so             libosgDB.so.3.6.5         libosgPresentation.so.3.6.5  libosgTerrain.so.3.6.5  libosgVolume.so.3.6.5
libCOS4.so.2.4                 libomniConnectionMgmt4.so.2.4  libomniZIOP4.so.2           libosgFX.so               libosgQt5.so                 libosgText.so           libosgWidget.so
libCOSDynamic4.a               libomniDynamic4.a              libomniZIOP4.so.2.4         libosgFX.so.161           libosgQt5.so.145             libosgText.so.161       libosgWidget.so.161
libCOSDynamic4.so              libomniDynamic4.so             libomniZIOPDynamic4.a       libosgFX.so.3.6.5         libosgQt5.so.3.5.7           libosgText.so.3.6.5     libosgWidget.so.3.6.5
libCOSDynamic4.so.2            libomniDynamic4.so.2           libomniZIOPDynamic4.so      libosgGA.so               libosgShadow.so              libosgUI.so             omniidl
libCOSDynamic4.so.2.4          libomniDynamic4.so.2.4         libomniZIOPDynamic4.so.2    libosgGA.so.161           libosgShadow.so.161          libosgUI.so.161         osgPlugins-3.6.5
libeigenpy.so                  libomniORB4.a                  libomniZIOPDynamic4.so.2.4  libosgGA.so.3.6.5         libosgShadow.so.3.6.5        libosgUI.so.3.6.5       pkgconfig
libgepetto-viewer-corba.so     libomniORB4.so                 libOpenThreads.so           libosgManipulator.so      libosgSim.so                 libosgUtil.so           python3.6
➜  ~ ls /opt/openrobots/lib64
ls: cannot access '/opt/openrobots/lib64': No such file or directory
```

alexandre-blanc, 2022-09-02:
> Can you check apt update && apt upgrade && apt install robotpkg-py36-robotpkg ? Maybe something went wrong during your installation.
I assume you meant `apt install robotpkg-py36-pinocchio`, the package `robotpkg-py36-robotpkg` does not seem to exit. Without copying the whole output, here is the result :
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
robotpkg-py36-pinocchio is already the newest version (2.6.9).
0 upgraded, 0 newly installed, 0 to remove and 23 not upgraded.
```

jcarpent, 2022-09-02:
Donc mix pip and robotpkg, this is not really possible due to pip.
Use meshcat instead of gepetto-viewer if you absolutely need a viewer.

I will close this issue as it seems to be solved.

PS: you should force reinstall with apt.

closed 2022-09-02 by jcarpent

alexandre-blanc, 2022-09-02:
I ended up solving the issue by :
- manually desinstalling all the robotpkg packages previously installed
- deleting the /opt/openrobots directory
- following _de_novo_ the instruction installations on stack-of-tasks
- using a virtual environment with python 3.6 and a .pth file pointing to /opt/openrobots/lib/python3.6/site-packages

All of this seems to work so far. I cannot make it work with python 3.8 however, but at this point I give up and will go with 3.6 since I do not really care.

**not actually requesting an answer to these, just food for thought :**
Does it mean that there is no support for python 3.8 ? Or that the preferred installation methods now are pip and conda ?
What about the tutorials in the documentation ? They still mention the use of robotpkg or assets installed through it such as the UR5 URDF.

Thank you for your availability !

jcarpent, 2022-09-02:
APT only supports the official Python version provided by the Ubuntu version
We do support all the python versions, and if you need a more recent one, you SHOULD switch to conda.

We should update the doc accordingly, but no time yet. @alexandre-blanc Could you help us by making a PR to update the doc? Your contribution will be nice for the whole community.

alexandre-blanc, 2022-09-02:
> but no time yet
Understandable !

> Could you help us by making a PR to update the doc?
I want to say yes, but cannot make any promises...

> Use meshcat instead of gepetto-viewer if you absolutely need a viewer.
My issue with gepetto-gui was that I was trying follow the tutorial in a jupyter notebook, like all the cool kids. It works fine when following the tutorial from a python prompt.

nim65s, 2022-09-02:
You'll get python 3.8 with robotpkg by upgrading your ubuntu 18.04 to ubuntu 20.04, since this is the version provided by the system :)