stack-of-tasks/pinocchio #1741: Mesh fails to load when filename attribute doesn't start with package://

stephane-caron, 2022-09-06:
The description [``cf2p.urdf``](https://github.com/utiasDSL/gym-pybullet-drones/blob/master/gym_pybullet_drones/assets/cf2p.urdf) for the Crazyflie 2.0 drone does not load properly with pinocchio 2.6.3:

```
ValueError: Could not load resource ./cf2.dae
Unable to open file "./cf2.dae".
Hint: the mesh directory may be wrong.
```

I believe I have configured the mesh path properly. Here is why:

- The URDF loads the mesh file using a relative path: ``<mesh filename="./cf2.dae" scale=" 1 1 1"/>``
- If we add the ``package://`` prefix: ``<mesh filename="package://./cf2.dae" scale=" 1 1 1"/>``, then the mesh loads properly and can e.g. be visualized in MeshCat:

[image]

Is there indeed an inconsistency in the way pinocchio treats the optional ``package://`` prefix?

(Relevant [part of the URDF spec](https://wiki.ros.org/urdf/XML/link).)

jcarpent, 2022-09-06:
I think relative paths are not supported by the convention.
Package relates to some known placement for the ROS ecosystem.

At first glance, we should be able to robustify to also handle relative path wrt the current file. @fabinsch Could you handle this issue?

stephane-caron, 2022-09-07:
Thanks for checking :slightly_smiling_face: Unfortunately there are several URDFs out there that use relative paths, I just took that one as an example.

From the sentence you quote, what the spec says is: "**if** you want to make the path relative to a package, **then** use ``package://``". Unfortunately this is incomplete as it does not specify whether relative paths without ``package://`` are allowed, and if so to what directory they are relative to.

> However, I added the ability in Pinocchio to handle this kind of relative path in https://github.com/stack-of-tasks/pinocchio/pull/1742.

Looks perfect :+1:

jcarpent, 2022-09-07:
Solved by #1742.

closed 2022-09-07 by jcarpent