stack-of-tasks/pinocchio #1743: issue for buildmodel function in gazebo simulator

Jaesoon-Lee, 2022-09-08:
I have a problem when i used buildmodel function, owing to build continously this function.

Can I use this function only once in a gazebo simulation?

thank you.

jcarpent, 2022-09-08:
Your issue is not very clear. Could you please provide a tiny reproducing example?

Jaesoon-Lee, 2022-09-13:
> Your issue is not very clear. Could you please provide a tiny reproducing example?

i'm using as below example code

`
void ROBOT_INFO::InverseDynamics(const Vector & q, const Vector & v, const std::string & filename, 
                                VectorXd & tau)
{
    
    Model model;
    
    pinocchio::urdf::buildModel(filename, model);
    VectorXd a = VectorXd::Zero(model.nq);

    Data data(model);

    nonLinearEffects(model, data, q, v);

    tau = data.nle;

}
`

The real time factor of the gazebo seems to be reduced as an effect of this function(buildModel).

As in this code, when this function is every called, buildModel is used.

The real time factor of the gazebo seems to be reduced as an effect of buildModel.

So, I want to declare and use buildModel only once like a global variable.

I want to know how to use buildModel.

fabinsch, 2022-09-13:
Hello, I think you should create the model and data object outside the function and then pass it to your custom function. something like:
```
Model model;
pinocchio::urdf::buildModel(filename, model);
Data data(model);

void ROBOT_INFO::InverseDynamics(const Model & model, Data & data, const Vector & q, const Vector & v, 
VectorXd & tau)
```

jcarpent, 2022-09-13:
Yes, you can create your model once and for all as suggested by @fabinsch. The same for the data (less costly).

And then you use your model as you want. Please look at the concept of singleton in C++ to help you define global variables safely.

I'm closing the issue as the suggestion seems to fulfil your need.