stack-of-tasks/pinocchio #1746: ROS2 Humble Release

Benjamin-Tan, 2022-09-14:
The pinocchio humble release PR has been merged into rosdistro, however the build is failing as described [here](https://discourse.ros.org/t/humble-pinoccio-package-missing-from-latest-sync/27288). Is there any ongoing work to get this fix?

olivier-stasse, 2022-09-14:
Dear @Benjamin-Tan  we are very committed to make it work on ROS-2 Humble.

From the log it looks like this is still the python bindings which are consuming too much memory. This is a recurring problem on which @wxmerkt, @jcarpent and @nim65s have been working a lot.
It created problems as well on our buildfarm. AFAIK it was fixed on our buildfarms, and we were hoping that it would also work on ROS-2 buildfarm.

Benjamin-Tan, 2022-09-14:
Dear @olivier-stasse , it is good to know that. Looking forward to the fix! Really appreciate the great work and commitment.

wxmerkt, 2022-10-16:
Hi @Benjamin-Tan 
The builds for Humble and Rolling are now building and should be available with the next sync of the respective distributions and available from now on the testing/staging server

closed 2022-10-16 by wxmerkt

olivier-stasse, 2022-10-16:
Thank you very much @wxmerkt for finding that `-g1` was enough.
Out of curiosity how did you check that it was passing apart trying it on a local docker ?
On my side I was trying with `-fvisibilty=hidden` but handling properly the exported symbol seems to be not so simple.

olivier-stasse, 2022-10-16:
I guess this is through the pre-release tests.

wxmerkt, 2022-10-16:
I compiled locally many times measuring the memory consumption for different build options, with different linkers, etc. - in the end the only thing that measurably had an impact was `-g1` (or `-g0` of course). The options to encourage the linker to reduce memory use did not help. For what it's worth, we currently need about ~1.8GB for the largest individual compilation unit outside of linking using `clang`