stack-of-tasks/pinocchio #1752: Difference() behavior not consistent between joint types

stephane-caron, 2022-09-26:
As of Pinocchio 2.6.4, ``pin.difference`` doesn't behave the same for different joint types when infinity is involved:

### Free flyers:

```python
>>> import pinocchio as pin
>>> model = pin.Model()
>>> model.addJoint(0, pin.JointModelFreeFlyer(), pin.SE3.Identity(), "root_joint")
1
>>> model.upperPositionLimit
array([1.79769313e+308, 1.79769313e+308, 1.79769313e+308, 1.79769313e+308,
       1.79769313e+308, 1.79769313e+308, 1.79769313e+308])
>>> pin.difference(model, pin.neutral(model), model.upperPositionLimit)
array([nan, nan, nan, nan, nan, nan])
```

### Revolute joints:

```python
>>> import pinocchio as pin
>>> model = pin.Model()
>>> model.addJoint(0, pin.JointModelRZ(), pin.SE3.Identity(), "foo")
1
>>> model.upperPositionLimit
array([1.79769313e+308])
>>> pin.difference(model, pin.neutral(model), model.upperPositionLimit)
array([1.79769313e+308])
```

### Expected behavior:

I think the second behavior is the right one, as it is expected that "+infinity - finite = +infinity". (Also, that's the one that would trivialize computing joint limits in IK, which is my current use case :wink:)

cmastalli, 2022-09-26:
I believe it should report `inf` for undefined limits. Moreover, the result of `diff` should be also `inf`.

jcarpent, 2022-09-27:
If you have an unbounded joint, then you should instead use the unbounded joint type, which is done for that.
The default value for the joint limits is +inf: the results you obtained come from internal computations.
We cannot guarantee a fixed output, as it really depends on the compiler in use.

I would say that it is the role of the end-user to properly use the correct input and output of the difference operations (which act on Lie groups).

jcarpent, 2022-09-27:
For the case of FreeFlyer, the bounds on the rotation part have no meaning in general. The NaN comes from there.

stephane-caron, 2022-09-28:
OK, I see the stakes better. I agree it would make sense to have well-formed configurations as a pre-requisite for ``difference``. Then the next thing to do would be to document this assumption, e.g. with a ``@note`` stating that the function's behavior is unspecified otherwise. (Ideally the type system could actually check this assumption for us, rather than delegating the task to end users.)

By (ii), do you mean changing the type of position limits to be something more structured than a configuration vector?

The use case I have in mind are joint position limits in a velocity-based IK. Currently it relies on

```python
Delta_q_max = pin.difference(configuration.model, q_act, q_max)
```

And then some ad-hoc parsing on NaNs/big floats to figure out which tangent coordinates are not proper limits. One way to get rid of this would be to add something like a ``JointModel.hasLimits()`` function, so that we can easily construct the slice of tangent coordinates to look at.

jcarpent, 2022-09-28:
Good idea to add the notion of hasConfigurationLimits() for each joint.
@fabinsch Could you handle this feature and extend the documentation accordingly?

jcarpent, 2022-09-28:
About this notion of limits, a FreeFlyer will have a bound on the translation but not on the rotation.

stephane-caron, 2022-09-29:
> One way to get rid of this would be to add something like a JointModel.hasLimits() function, so that we can easily construct the slice of tangent coordinates to look at.

I should have made it explicit, but what I have in mind are runtime limits, *i.e.* whether the user/URDF specified a limit explicitly.

stephane-caron, 2022-09-29:
That works with configurations, the question is how to do the same with tangent vectors?

```python
slice = [joint.idx_v for joint in model.joints if joint.has_runtime_configuration_limits]
```

Exposing ``min_config`` and ``max_config`` would do the job :+1:

jcarpent, 2022-09-29:
This is in the Model, no?

jcarpent, 2022-09-29:
I don't understand your approach in fact. @stephane-caron Do you think is the right one?

jcarpent, 2022-09-29:
What we can do for sure, is to set the real bounds on the configuration for non-Euclidian manifolds.

jcarpent, 2022-09-30:
@stephane-caron To move forward in the current issue, I would suggest you clamp the limit values in the set [-1e20, 1e20] for instance to get correct output from `difference`.

stephane-caron, 2022-09-30:
Yep, that's in essence how [joint limits are implemented in Pink v0.5](https://github.com/tasts-robots/pink/blob/510a702996411a4528b274b4351d53a20760a4be/pink/limits.py#L30). What's not great about this approach is that in relies on how Pinocchio is implemented, rather than on its API.

To keep things in scope, let me spin a separate issue on this point: https://github.com/stack-of-tasks/pinocchio/issues/1758

From my side, the initial point of this issue about the API for ``difference`` has been addressed https://github.com/stack-of-tasks/pinocchio/pull/1753 and we could close it. :ok_hand:

jcarpent, 2022-09-30:
> What's not great about this approach is that in relies on how Pinocchio is implemented, rather than on its API.

I don't understand this point.

stephane-caron, 2022-09-30:
Arh, sorry you meant clamping the _inputs_ to ``difference`` to get correct outputs, while I was looking at clamping its outputs.

Clamping the inputs is not sufficient in itself because ``difference`` is not meant to handle out-of-manifold rotation coordinates. For example:

```python
>>> model = pin.Model()
>>> model.addJoint(0, pin.JointModelFreeFlyer(), pin.SE3.Identity(), "root_joint")
1
>>> pin.difference(model, pin.neutral(model), np.array([1e20] * 7))
array([9618.35346861, 9618.35346861, 9618.35346861,    0.        ,
          0.        ,    0.        ])
```

But if we make sure unbounded rotations have proper coordinates, then we get correct outputs for the other coordinates:

```python
>>> pin.difference(model, pin.neutral(model), np.array([1e20]*3 + [0, 0, 0, 1]))
array([1.e+20, 1.e+20, 1.e+20, 0.e+00, 0.e+00, 0.e+00])
```

No worries re. my issue which found a pleasant solution in https://github.com/tasts-robots/pink/pull/12. I hope our exchange here was not too unstructured and somehow useful feedback for Pinocchio :smiley:

wxmerkt, 2022-09-30:
@stephane-caron The first example does not have the quaternion normalised which is likely why things break in unexpected ways. It's assumed that the quaternion being passed in has been normalised before

jcarpent, 2022-10-01:
We need to fix in Pinocchio the proper value for the bounds limit for unbounded joints.
We will do that with the neutral element of the given joint type.