stack-of-tasks/pinocchio #1755: Calculate the velocity at the center of mass

huihuishen, 2022-09-29:
Is there a python function to get the robot's velocity at the center of mass  ？

jcarpent, 2022-09-29:
```python
In [18]: pin.centerOfMass?
Docstring:
centerOfMass( (Model)model, (Data)data, (numpy.ndarray)q [, (bool)compute_subtree_coms=True]) -> numpy.ndarray :
    Compute the center of mass, putting the result in context::Data and return it.If compute_subtree_coms is True, the algorithm also computes the center of mass of the subtrees.

centerOfMass( (Model)model, (Data)data, (numpy.ndarray)q, (numpy.ndarray)v [, (bool)compute_subtree_coms=True]) -> numpy.ndarray :
    Computes the center of mass position and velocity by storing the result in context::Data. It returns the center of mass position expressed in the WORLD frame.
    If compute_subtree_coms is True, the algorithm also computes the center of mass of the subtrees.

centerOfMass( (Model)model, (Data)data, (numpy.ndarray)q, (numpy.ndarray)v, (numpy.ndarray)a [, (bool)compute_subtree_coms=True]) -> numpy.ndarray :
    Computes the center of mass position, velocity and acceleration by storing the result in context::Data. It returns the center of mass position expressed in the WORLD frame.
    If compute_subtree_coms is True, the algorithm also computes the center of mass of the subtrees.

centerOfMass( (Model)model, (Data)data, (int)kinematic_level [, (bool)compute_subtree_coms=True]) -> None :
    Computes the center of mass position, velocity and acceleration of a given model according to the current kinematic values contained in data and the requested kinematic_level.
    If kinematic_level = 0, computes the CoM position, if kinematic_level = 1, also computes the CoM velocity and if kinematic_level = 2, it also computes the CoM acceleration.

centerOfMass( (Model)model, (Data)data, (KinematicLevel)kinematic_level [, (bool)compute_subtree_coms=True]) -> numpy.ndarray :
    Computes the center of mass position, velocity or acceleration of a given model according to the current kinematic values contained in data and the requested kinematic_level.
    If kinematic_level = POSITION, computes the CoM position, if kinematic_level = VELOCITY, also computes the CoM velocity and if kinematic_level = ACCELERATION, it also computes the CoM acceleration.
    If compute_subtree_coms is True, the algorithm also computes the center of mass of the subtrees.

centerOfMass( (Model)model, (Data)data [, (bool)compute_subtree_coms=True]) -> numpy.ndarray :
    Computes the center of mass position, velocity and acceleration of a given model according to the current kinematic values contained in data.
    If compute_subtree_coms is True, the algorithm also computes the center of mass of the subtrees.
Type:      function
```

closed 2022-09-29 by jcarpent

huihuishen, 2022-09-29:
> ```python
> In [18]: pin.centerOfMass?
> Docstring:
> centerOfMass( (Model)model, (Data)data, (numpy.ndarray)q [, (bool)compute_subtree_coms=True]) -> numpy.ndarray :
>     Compute the center of mass, [...]

Thank you very much. I'll try that. By the way, where can I get the doc of full python api about 
pinocchio？

jcarpent, 2022-09-29:
Locally, you have it. With @fabinsch, we plan to release it online along with the C++ doc (https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/) and this [cheat sheet](https://github.com/stack-of-tasks/pinocchio/raw/master/doc/pinocchio_cheat_sheet.pdf).