stack-of-tasks/pinocchio #1757: Visualize texture from Dae file

ddliu365, 2022-09-29:
Hi, folks,

I am using meshcat to visualize the Pinocchio robot and found the texture is missing. My visual files are using `*.dae`.

I checked API "pinocchio: mesh cat-viewer-dae" and it seems only the geometry shape has been imported to the viewer.

https://docs.ros.org/en/melodic/api/pinocchio/html/meshcat__visualizer_8py_source.html

Is there any way to show the texture from dae file? I might miss some API functions.

jcarpent, 2022-09-30:
It has been fixed in Pinocchio 3.x. We hope to be able to release it soon.

closed 2022-09-30 by jcarpent

ddliu365, 2022-09-30:
Hi,

I built the pinocchio3-preview from source and install it, but it seems this issue still exist.

Should I try it by myself?

I used this function in `pinocchio` to visualize my dae file, maybe different API?

```
pin_robot = pin.RobotWrapper.BuildFromURDF(urdf, package_dirs)
self.viz = viz = pin.visualize.MeshcatVisualizer(
            pin_robot.model, pin_robot.collision_model, pin_robot.visual_model
        )
        viz.initViewer(viewer)
        viz.loadViewerModel()
```