stack-of-tasks/pinocchio #1759: Bullet Velocity vs Pinocchio Velocity

iBims1JFK, 2022-09-30:
Hello and thank you for your work and support,
I am using PyBullet for the simulation of a Barret WAM and it seems that the velocity of Pinocchio and Bullet are not the same.

What you can see is just the Y axis. It is the axis along which the endeffector falls.  
[image]
As you can see, both estimate the same values for the position. But the velocity differs.

The configuration is as follows: 
```
p.resetJointState(bodyUniqueId=BW, jointIndex=1, targetValue=np.pi/2)
p.resetJointState(bodyUniqueId=BW, jointIndex=4, targetValue=np.pi/2)
```
[image]
There is no influence from any motor. It just falls to the ground. What's really interesting is, at the beginning apparently Pinocchio thinks that the ee is falling to the wrong side.

Here is the code:
```
q = np.empty(modelWAM.nq)
dq = np.empty(modelWAM.nq)

for i in range(0, 4):
    q[i] = p.getJointState(BW, i + 1)[0]
    dq[i] = p.getJointState(BW, i + 1)[1]

pinocchio.forwardKinematics(modelWAM, data, q, dq)
pinocchio.updateFramePlacements(modelWAM, data)

J = pinocchio.computeJointJacobians(modelWAM, data)

endPos = data.oMf[frame_id_end_effector].translation
linkState = p.getLinkState(
    BW, 6, computeForwardKinematics=True, computeLinkVelocity=True)

pinocchioPos.append(endPos.copy())
bulletPos.append(np.asarray(linkState[0]))
pinocchioVel.append((J @ dq)) # or 
# bulletVel.append(pinocchio.getFrameVelocity(
#     modelWAM, data, frame_id_end_effector, pinocchio.WORLD).linear)
# leads to the same result
bulletVel.append(np.asarray(linkState[6]))

p.stepSimulation()
```

Do you have any idea what I have done wrong? Thank you in advance.

stevenjj, 2022-09-30:
Check which frame J is computed in from pinocchio. If you request the frame LOCAL_WORLD_ALIGNED, the values should look similar.

iBims1JFK, 2022-09-30:
Thank you, it seems to be working. Could you tell me, that is the difference between WORLD and LOCAL_WORLD_ALIGNED? Do I also need to get the dJ in that specific Frame than?

stevenjj, 2022-09-30:
The frame WORLD has the spatial velocity described in fixed-frame. Very often, this is not what you want LOCAL_WORLD_ALIGNED is the same as the LOCAL body Jacobian except the Jacobian has been rotated to be align with the world frame. 
A simple relationship between the LOCAL jacobian and LOCAL_WORLD_ALIGNED is the following:
`J_local_world_aligned = [R, 0; 0, R] * J_local ` where `J_local` has been pre-multiplied by a 6x6 matrix where R is the orientation of the link with respect to the world.

A good reference to understand the difference between WORLD and LOCAL is available in section 3.3.2 with a discussion on how to convert between the two in 3.3.2.1  here:
http://hades.mech.northwestern.edu/images/7/7f/MR.pdf

iBims1JFK, 2022-10-01:
Thank you very much

closed 2022-10-01 by iBims1JFK

jcarpent, 2022-10-04:
Thanks @stevenjj for nicely answering this issue.