stack-of-tasks/pinocchio #1760: Removing boost-container-limits.hpp or Boost in the future? Suggestions for modularity?

stevenjj, 2022-09-30:
Hi all, this library has been really great to use. 

However, one issue with integrating large code bases with pinocchio is that to use pinocchio with other Boost dependent libraries (e.g. ROS 1),  the forward declaration needs to be included first e.g.: `#include <pinocchio/fwd.hpp>` Specifically, `#include "pinocchio/container/boost-container-limits.hpp"` needs to occur before any Boost related headers. 

So suppose I have library A that depends on pinocchio, and then I have library B that depends on library A. When I compile library B and library B happens to include boost before including library A's header files, the compilation would fail as pinocchio would require the boost-container-limits.hpp to be included first.

To fix this problem, now library B will need to depend on pinocchio so that these forward declarations can be called first. By transitivity, a library C that depends on library B would also need pinocchio if C happens to call Boost headers first before B's headers that call on A. Here we have an issue that library C does not even require pinocchio, yet it has to depend on pinocchio so that it can include and ensure that the forward declarations happen first.

With this issue, pinocchio is not a very modular library.

A similar issue has been discussed previously:  https://github.com/stack-of-tasks/pinocchio/issues/849
With the primary suggestion to include the following in cmake (https://github.com/stack-of-tasks/pinocchio/pull/852):
````
# This information is redundant with the content of include/pinocchio/container/boost-container-limits.hpp
# but it avoids any compilation issue.
ADD_DEFINITIONS(-DBOOST_MPL_LIMIT_LIST_SIZE=30)
ADD_DEFINITIONS(-DBOOST_MPL_LIMIT_VECTOR_SIZE=30)
# Variable containing all the cflags definition relative to optional dependencies
SET(CFLAGS_DEPENDENCIES)
LIST(APPEND CFLAGS_DEPENDENCIES "-DBOOST_MPL_LIMIT_LIST_SIZE=30")
LIST(APPEND CFLAGS_DEPENDENCIES "-DBOOST_MPL_LIMIT_VECTOR_SIZE=30")
````
In most cases these global definitions work fine with other libraries. 

However, adding these definitions make some libraries that use Boost to fail. For example, the [hpipm catkin interface in the OCS2 library](https://github.com/leggedrobotics/ocs2/tree/main/ocs2_sqp/hpipm_catkin) would fail complaining about invalid vector sizes. In other libraries, the Boost related failure can be hard to debug. In essence, the global macro definitions is not a good general solution. 

Moreover, the definitions being global goes against modern cmake standards: https://cliutils.gitlab.io/modern-cmake/chapters/intro/dodonot.html

We have also been using the ROS debian version made by @wxmerkt , with the corresponding nice example https://github.com/wxmerkt/pinocchio_ros_example . However, this package also does not resolve the Boost include related issues.

Something we have attempted is to use modern cmake linking rules that make pinocchio header files private per target that actually use pinocchio:
```
target_link_libraries(${PROJECT_NAME} 
  PRIVATE ${catkin_LIBRARIES} ${pinocchio_LIBRARIES}
)

target_include_directories(
  ${PROJECT_NAME}
  PUBLIC $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>
               $<INSTALL_INTERFACE:include>          
  PRIVATE
   ${catkin_INCLUDE_DIRS}
    ${pinocchio_INCLUDE_DIRS}
)
````
However despite trying to make the libraries' dependency on pinochio private, ultimately when `#include "pinocchio/container/boost-container-limits.hpp` gets included after a Boost include on a separate library, the following error will get thrown
````
      # error "You should include pinocchio before the Boost headers (e.g. #include <pinocchio/fwd.hpp>)"
```

I'm wondering if there are plans to address this boost container issue? Despite pinocchio being header-only, the boost change requirement makes the library intrusive to other libraries.

Apologies for the long post and thank you for an otherwise excellent library and for your time.

jcarpent, 2022-10-01:
The quick answer to this long detailed issue is that:
1. Pinocchio needs Boost variant to be both efficient and flexible. (std::variant is not as mature as boost variant).
2. We can find an easy way to get everything working smoothly. Importantly, we can try to robustify how we check the desired length for MPL vectors. And we can check whether we can use variadic templates to properly use boost variant without relying on preprocessed macros.

stevenjj, 2023-02-08:
I may have a working solution for others facing similar issues. The idea is to wrap pinocchio under an interface and use the [pointer-to-implementation (pimpl) technique](https://en.cppreference.com/w/cpp/language/pimpl)

The header file is barebones while the cpp file has the forward declarations and all of the pinocchio related includes. When other user-made libraries want to use pinocchio, they only have to include the wrapper header file and never see the pinocchio-required forward declarations or pinocchio related objects.

Example implementation:

pinocchio_wrapper.hpp
````
#include <memory>
class PinocchioWrapper {
public:
    PinocchioWrapper();
    ~PinocchioWrapper();
    PinocchioWrapper(PinocchioWrapper const& other);

    void somePinocchioFunction();

private:
  struct Impl;
  std::unique_ptr<Impl> pimpl;
};
````
pinocchio_wrapper.cpp
````
// Forward declaration. Must be declared before any boost call.
#include "pinocchio/fwd.hpp"
// Wrapper header
#include "pinocchio_wrapper.h"

// Pinocchio related functions
#include "pinocchio/parsers/sample-models.hpp"

struct PinocchioWrapper::Impl { 
  Impl() {}
  ~Impl() {}
  void somePinocchioFunction()
  {
    pinocchio::Model model;
    pinocchio::buildModels::humanoidRandom(model);
    /** Other pinocchio functions here */
  }
};

PinocchioWrapper::PinocchioWrapper(): pimpl(std::make_unique<Impl>()) {} // constructor
PinocchioWrapper::PinocchioWrapper(PinocchioWrapper const& other): pimpl(std::make_unique<Impl>(*other.pimpl)) {} // copy constructor
PinocchioWrapper::~PinocchioWrapper(){} // default destructor

void PinocchioWrapper::somePinocchioFunction() // wrapper function uses the pointer implementation
{
  pimpl->somePinocchioFunction();
}
````

other_object_that_uses_pinocchio.hpp
````
#include "pinocchio_wrapper.hpp"
class OtherObject;
````

During compile time the wrapper implementation is hidden from other libraries.

jcarpent, 2023-02-09:
Thanks @stevenjj for this very nice suggestion.
Hope that everything will be simpler in the near future with Pinocchio 3.

stevenjj, 2023-02-10:
Thank you @jcarpent! The workaround has been great, so I'll be closing the issue. Otherwise, I'm looking forward to Pinocchio 3!

closed 2023-02-10 by stevenjj