stack-of-tasks/pinocchio #1761: Jacobian and ReferenceFrame understanding prolem

huihuishen, 2022-10-01:
My robot is floating, I now want to calculate the Jacobian matrix of the end of the robot arm in the world coordinate system (fixed), I use pinocchio::computeFrameJacobian(Model, Data, q, eeFrameId_, pinocchio::ReferenceFrame::LOCAL, J); I noticed similar questions before, like confused Jacobian computation #1222, but combined with the explanation of ReferenceFrame in the help documentation, I am still a little confused about the definition of ReferenceFrame. How many coordinate systems exist in a simulation process, and how is the position of each coordinate system determined? What is the difference between the specified LOCAL, WORLD, and LOCAL_WORLD_ALIGNED when calculating a frame or joint? The three specified methods mentioned in the help document move with the joint or frame, but in my understanding, the world The coordinate system is fixed. I don't know if I made the question clear, can you give a clear and detailed introduction to the coordinate system that exists in the simulation scene and the relevant definitions of the ReferenceFrame specified when calculating the Jacobian matrix? Are there any relevant description graphics to help understand?
Looking forward to your reply. Thanks!

huihuishen, 2022-10-01:
> My robot is floating, I now want to calculate the Jacobian matrix of the end of the robot arm in the world coordinate system (fixed), I use pinocchio::computeFrameJacobian(Model, Data, q, [...]

> My robot is floating, I now want to calculate the Jacobian matrix of the end of the robot arm in the world coordinate system (fixed), I use pinocchio::computeFrameJacobian(Model, Data, q, [...]

huihuishen, 2022-10-01:
> My robot is floating, I now want to calculate the Jacobian matrix of the end of the robot arm in the world coordinate system (fixed), I use pinocchio::computeFrameJacobian(Model, Data, q, [...]

Is world/WORLD frame not fixed from innitial?

jcarpent, 2022-10-01:
Dear @huihuishen,

Your issue is not crystal clear. I suggest looking at the [cheat sheet](https://github.com/stack-of-tasks/pinocchio/blob/master/doc/pinocchio_cheat_sheet.pdf) and to clearly simplify your wording to help us to provide you a clear and precise answer.

huihuishen, 2022-10-02:
In the doc:
The WORLD frame convention corresponds to the frame concident with the Universe/Inertial frame but moving with the moving part (Joint, Frame, etc.).
It said that the WORLD cs is moving with the  joint. Is the world frame moving?  What's the fixed frame? Universe/Inertial frame?

huihuishen, 2022-10-02:
> In the doc: The WORLD frame convention corresponds to the frame concident with the Universe/Inertial frame but moving with the moving part (Joint, Frame, etc.). It said that the WORLD cs is moving with the joint. Is the world frame moving? What's the fixed frame? Universe/Inertial frame?

If the world is moving with joint, I changed the X of root_joint, the data shouldn't change, but the actual result is that the positon data changed. It's incompatible. Where did I misunderstand？Looking forward to your reply.

jcarpent, 2022-10-02:
No, the universe is fixed, but the coincident point moves when the body moves.
Please look at Featherstone's book if you need a reference.

huihuishen, 2022-10-03:
Sorry, I don't get your point. I use the following: robot1 = RobotWrapper.BuildFromURDF(urdf_model_path, mesh_dir, pin.JointModelFreeFlyer())  to load my robot. I just want to know the position of the root_joint of my robot in world frame. As you said, the world frame is moving with robot's body. I use the code : idx= robot1.index('root_joint')
print(robot1.placement(q0, idx)). In the doc, it said it return the position of idx in world. If the world is moving with robot, I think the positon data will not change, since the relative position doesn't change. It's confusing!!!

huihuishen, 2022-10-03:
> No, the universe is fixed, but the coincident point moves when the body moves. Please look at Featherstone's book if you need a reference.

By the way, what's is the Featherstone's book? Which one should I read? Thank you very much!

huihuishen, 2022-10-03:
> Sorry, I don't get your point. I use the following: robot1 = RobotWrapper.BuildFromURDF(urdf_model_path, mesh_dir, pin.JointModelFreeFlyer()) to load my robot. I just want to know the position of [...]

Sorry, I don't get your point. I use the following: robot1 = RobotWrapper.BuildFromURDF(urdf_model_path, mesh_dir, pin.JointModelFreeFlyer()) to load my robot. I just want to know the position of the root_joint of my robot in world frame. As you said, the world frame is moving with robot's body. I use the code : idx= robot1.index('root_joint')
print(robot1.placement(q0, idx)). In the doc, it said it return the position of idx in world. If the world is moving with robot, I think the positon data will not change, since the relative position doesn't change. It's confusing!!!

huihuishen, 2022-10-03:
> No, the universe is fixed, but the coincident point moves when the body moves. Please look at Featherstone's book if you need a reference.

Sorry, I don't get your point. I use the following: robot1 = RobotWrapper.BuildFromURDF(urdf_model_path, mesh_dir, pin.JointModelFreeFlyer()) to load my robot. I just want to know the position of the root_joint of my robot in world frame. As you said, the world frame is moving with robot's body. I use the code : idx= robot1.index('root_joint')
print(robot1.placement(q0, idx)). In the doc, it said it return the position of idx in world. If the world is moving with robot, I think the positon data will not change, since the relative position doesn't change. It's confusing!!!

fabinsch, 2022-10-03:
Hi @huihuishen , the book of Featherstone can be found [here](https://bcourses.berkeley.edu/files/70847609/download?download_frd=1). You can find some additional material about Pinocchio in the [memmo summer school](https://memory-of-motion.github.io/summer-school/materials).

jcarpent, 2022-10-03:
Starting from simple. @huihuishen What do you want to do? What is your goal?

huihuishen, 2022-10-04:
> Starting from simple. @huihuishen What do you want to do? What is your goal?

Now, I have a floating robot. I want apply a force on the robot. The force is described in the fixed frame.To get the corresponding Torque, I should calculate the Jocabian in a fixed frame. As we discussed above, if I use "computeFrameJacobian(Model, Data, q, eeFrameId_, pinocchio::ReferenceFrame::WORLD, J)", It seems  not correct. Because the world frame is moving. Do I understand right?

jcarpent, 2022-10-04:
Use the `LOCAL_WORLD_ALIGNED` frame.

closed 2022-10-04 by jcarpent