stack-of-tasks/pinocchio #1765: Ubuntu22 installation problem

flilqhu, 2022-10-09:
The environment is the ubuntu22

'https://stack-of-tasks.github.io/pinocchio/download.html'
I follow the instruction "build from source"
but when I  enter the follow code

'cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local'

The error message is 
'
-- Package version (ROS package.xml): 2.6.10
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.1") found components: doxygen dot 
-- Since urdfdom >= 1.0.0, the default C++ standard is C++11. The project is then compiled with C++11 standard.
-- The Python bindings of Pinocchio will be compiled along the main library. If you want to disable this feature, please set the option BUILD_PYTHON_INTERFACE to OFF.
CMake Error at cmake/package-config.cmake:110 (find_package):
  By not providing "Findeigenpy.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "eigenpy", but
  CMake did not find one.

  Could not find a package configuration file provided by "eigenpy"
  (requested version 2.7.10) with any of the following names:

    eigenpyConfig.cmake
    eigenpy-config.cmake

  Add the installation prefix of "eigenpy" to CMAKE_PREFIX_PATH or set
  "eigenpy_DIR" to a directory containing one of the above files.  If
  "eigenpy" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:161 (ADD_PROJECT_DEPENDENCY)

-- Configuring incomplete, errors occurred!
'

so I try to install the eigenpy 

'sudo apt install robotpkg-py35-eigenpy'

it shows me that

'E: Unable to locate package robotpkg-py35-eigenpy'

Can you teach me how to fix it, or I can not install on the ubuntu22

Thank you so much

jcarpent, 2022-10-09:
Please use apt directly instead: `sudo apt-get install robotpkg-py310-pinocchio` or conda.

closed 2022-10-09 by jcarpent

nim65s, 2022-10-10:
Ubuntu 22.04 has python 3.10, so you should install `robotpkg-py310-eigenpy` instead, but as @jcarpent says, binary installs will lead to less troubles.