stack-of-tasks/pinocchio #1766: Assimp library linking error using pinocchio in ROS

shbang91, 2022-10-11:
Hello, 

Thank you for your great effort in maintaining this awesome repo.

I have a question regarding the assimp library linking error.

I created my own library (let's say Library A) which is ROS-independent by using pinocchio installed from source (/usr/local/lib). I did several testing but the Library A works fine without any problem. 

However, when I installed the Library A in the system (/usr/local/lib) and tried to use the Library A within ROS by target linking Library A with my ROS nodelet library (let's say Library B), for instance, target_link_libraries(Library B Library A),  the ROS keeps complaining  `undefined symbol: _ZNK6Assimp8IOSystem12ComparePathsEPKcS2_`. It seems that when integrated within ROS. pinocchio is not able to find the assimp library correctly. Do you have any suggestions or guidance to debug this issue?

Another thing to ask is how I could build the pinocchio with the assimp library. Even though there is a cmake module for the assimp within pinocchio repo, any CMakeLists.txt in the pinocchio repo does not have any options to be built with the assimp. Maybe I missed but I would really appreciate any guidance if possible.

Thank you,
Seung Hyeon

wxmerkt, 2022-10-12:
I am afraid I cannot debug based on the information given. It's a bit unclear how Pinocchio was installed (from source, then globally isntalled) but with which version of EigenPy and HPP-FCL? And whether ROS-Pinocchio is installed at the same time? Where is assimp from? Is Pinocchio linked _publicly_ based on the targets to the library or with pkg-config or explicit name?

jcarpent, 2022-10-12:
Thanks @wxmerkt for the quick feedback. @shbang91 Could you provide the missing info?

shbang91, 2022-10-13:
@jcarpent In my machine, I used assimp version 4.0 and I also looked into [https://github.com/dartsim/dart/issues/549](url). It seems same error but mine is runtime error instead of build error.

@wxmerkt Since I only use cpp version of pinocchio and some functions like (getjacobian, getmassmatrix, forwardkinematics, etc), I did not install EigenPy and HPP-FCL. Also, I did not install ROS-Pinocchio. 
Library_A is built as a static library and `target_link_libraries(Library_A PUBLIC ${pinocchio_LIBRARIES})` and this Library_A is ROS independent. 
Library_B is a shared library and `target_link_libraries(Library_B PUBLIC Library_A)` then I installed Library_B and Library_A globally (in /usr/local/lib) and tried to use Library_B as an external library in ROS by using custom-made CMake module to find Library_B. 
I made a Library_C which is a ROS nodelet but `target_link_libraries(Library_C Library_B)` and when I roslaunch the Library_C, I got the above runtime error which is `undefined symbol: _ZNK6Assimp8IOSystem12ComparePathsEPKcS2_`. 

Since the pinocchio that I globally installed is not dependent on the HPP-FCL, I believe it should have not complained about the assimp-related runtime error. But one thing I doubted is whether I should make the Library_A as a shared library. Or, should I install HPP-FCL in order for assimp library to be dependent on the pinocchio library?

Do you have any thoughts or suggestions?

shbang91, 2022-10-13:
I just tried that but it gives me the same error.

wxmerkt, 2022-10-13:
My hunch is that the assimp requirement comes from urdfdom which is used by Pinocchio's URDF parser. As such, I wonder if the dependency has not been correctly exported. However, without access to a minimally reproducible example (e.g. a Docker image) I am afraid I wont be able to provide further insight

shbang91, 2022-10-13:
Thank you for your comment. One more question from my side is that do you think ROS environment somehow messes up the assimp dependency to Pinocchio? When I used Library B (robot controller) that I mentioned above without ROS, it works perfectly without any error (no assimp-related runtime error).

jcarpent, 2022-10-13:
assimp is linked to pinocchio via hpp-fcl.

shbang91, 2022-10-13:
@jcarpent Do you think hpp-fcl is the only place where the assimp is linked to Pinocchio or there are somewhere else?

jcarpent, 2022-10-13:
This is the only place. But you have a too-old version of assimp apparently as I mentioned this morning.

shbang91, 2022-10-13:
@wxmerkt Yes, I did not install Eigen-py and HPP-FCL.

shbang91, 2022-10-13:
@wxmerkt One another information I would like to provide is that when I do `ldd Library_A, Library_B, Libray_C` I don't see any libassimp.so dependency at all.

jcarpent, 2022-10-13:
@shbang91 The best would be to share a docker image which replicates your issue.
This minimal effort will help us to find your issue.

shbang91, 2022-10-13:
@jcarpent Thank you for the suggestion. I'll try to share the docker image and get back to you soon.

jcarpent, 2022-10-24:
I will close this issue as there is no more activity on it.

closed 2022-10-24 by jcarpent