stack-of-tasks/pinocchio #1767: Segmentation fault (core dumped) when calculating Jacobian time variation

hh2564, 2022-10-13:
Hi, 

When I tried to calculate the dJ using the following lines 
```
t = pin.getJointJacobianTimeVariation(model, data, controlled_frame, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED)
```
I got the following error: 
```
Segmentation fault (core dumped)
```
However when I tried to change reference frame to local or world, this error disappear. However I'll need to use the Jacobian with the reference frame of local_world_aligned. Is there anyway I can solve this problem? 

Many thank in advance! 

Best,
Heming

fabinsch, 2022-10-13:
Hi @hh2564 ,

we have a cpp unittest for this function which is using the frame LOCAL_WORLD_ALIGNED ([here](https://github.com/stack-of-tasks/pinocchio/blob/master/unittest/joint-jacobian.cpp#L120)) and our ci is passing. So the function should be working, can you check the unittest joint-jacobian on your machine?

I see that you are using the python bindings, but this function is just calling the underlying cpp function [here](https://github.com/stack-of-tasks/pinocchio/blob/master/bindings/python/algorithm/expose-jacobian.cpp#L44) which is supposed to be working.

fabinsch, 2022-10-13:
Also, did you use the function computeJointJacobiansTimeVariation before calling getJointJacobianTimeVariation ?

hh2564, 2022-10-13:
Hi Fabian, 

Thank you very much for your response! I'm not sure about the unittest joint-jacobian you mentioned about. Here's the code I implemented: 
```
pin.framesForwardKinematics(model,data,q)
pin.computeJointJacobians(model,data)
J = pin.getFrameJacobian(model, data, controlled_frame, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED)
print(J)
pin.computeJointJacobiansTimeVariation(model,data,q,v)
Jt = pin.getJointJacobianTimeVariation(model, data, controlled_frame, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED)
print(Jt)
```

And attaining J doesn't have any issue.

fabinsch, 2022-10-13:
Your example is not self-containted. I did a quick check on my machine running the following code I had open anyway and it is working without any segfault.
```
import os

import numpy as np
import pinocchio as pin

# load romeo robot
filename = str(os.path.dirname(os.path.abspath(__file__)))
path = filename + '/benchmark_tsid/romeo'
urdf = path + '/urdf/romeo.urdf'
srdf = path + '/srdf/romeo_collision.srdf'

model, collision_model, visual_model = pin.buildModelsFromUrdf(urdf, path, pin.JointModelFreeFlyer())
pin.loadReferenceConfigurations(model, srdf, False)
nv = model.nv

q = model.referenceConfigurations["half_sitting"]
v = np.random.random(nv)

# modify initial robot configuration so that foot is on the ground (z=0)
frame_id = model.getFrameId("RAnkleRoll")
data = model.createData()

pin.framesForwardKinematics(model,data,q)
pin.computeJointJacobians(model,data)
J = pin.getFrameJacobian(model, data, frame_id, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED)
print(f"J: {J}")
pin.computeJointJacobiansTimeVariation(model,data,q,v)
Jt = pin.getJointJacobianTimeVariation(model, data, frame_id, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED)
print(f"Jt: {Jt}")
```

the output is 
```
(master_pinocchio) ➜  python git:(add-MSVC-ci) ✗ ipython -i test_JointJac.py    
Python 3.9.12 | packaged by conda-forge | (main, Mar 24 2022, 23:22:55) 
Type 'copyright', 'credits' or 'license' for more information
IPython 8.3.0 -- An enhanced Interactive Python. Type '?' for help.
J: [[ 1.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
  -8.02212513e-01  9.60000000e-02  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00 -6.02172513e-01 -2.88551208e-01
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00]
 [ 0.00000000e+00  1.00000000e+00  0.00000000e+00  8.02212513e-01
   0.00000000e+00  3.46224950e-02  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   3.46224950e-02  6.02172513e-01  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00]
 [ 0.00000000e+00  0.00000000e+00  1.00000000e+00 -9.60000000e-02
  -3.46224950e-02  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00 -3.46224950e-02  2.89516908e-02
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  1.00000000e+00
[... 48 lines cut ...]
   0.4248128   0.19043714 -0.06859184 -0.00846526  0.          0.
   0.        ]
 [ 0.          0.          0.          0.         -0.32259179  0.46007683
   0.          0.          0.          0.          0.          0.
   0.          0.          0.          0.          0.          0.
   0.46007683  0.          0.          0.          0.          0.
   0.          0.          0.          0.          0.          0.
  -0.61801197 -0.28271093 -1.17299038 -0.45965533  0.          0.
   0.        ]
 [ 0.          0.          0.          0.32259179  0.         -0.53776933
   0.          0.          0.          0.          0.          0.
   0.          0.          0.          0.          0.          0.
  -0.53776933  0.          0.          0.          0.          0.
   0.          0.          0.          0.          0.          0.
  -0.40685962  0.80553114  0.84630947  0.71822818  0.          0.
   0.        ]
 [ 0.          0.          0.         -0.46007683  0.53776933  0.
   0.          0.          0.          0.          0.          0.
   0.          0.          0.          0.          0.          0.
   0.          0.          0.          0.          0.          0.
   0.          0.          0.          0.          0.          0.
   0.67327166 -1.01037573 -0.40310389 -0.77585442  0.          0.
   0.        ]]

```

wxmerkt, 2022-10-13:
@hh2564 how do you retrieve `controlled_frame` - and is it perhaps greater than `nframes` or `njoints`?

hh2564, 2022-10-13:
I tried to take a look at the sample model and the following code works: 

```
import os

import numpy as np
import pinocchio as pin

model = pin.buildSampleModelManipulator()
data = model.createData()
 
q = pin.neutral(model)
v = pin.utils.zero(model.nv)
a = pin.utils.zero(model.nv)
controlled_frame = 6 

pin.computeJointJacobians(model,data)
J = pin.getFrameJacobian(model, data, controlled_frame, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED)
print(J)
pin.computeJointJacobiansTimeVariation(model,data,q,v)
Jt = pin.getJointJacobianTimeVariation(model, data, controlled_frame, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED)
print(Jt)
```
and the output
```
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for boost::shared_ptr<hpp::fcl::CollisionGeometry> already registered; second conversion method ignored.
[[0. 0. 0. 0. 0. 0.]
 [0. 0. 0. 0. 0. 0.]
 [0. 0. 0. 0. 0. 0.]
 [1. 0. 0. 0. 0. 0.]
 [0. 1. 0. 0. 0. 0.]
 [0. 0. 1. 0. 0. 0.]]
[[0. 0. 0. 0. 0. 0.]
 [0. 0. 0. 0. 0. 0.]
 [0. 0. 0. 0. 0. 0.]
 [0. 0. 0. 0. 0. 0.]
 [0. 0. 0. 0. 0. 0.]
 [0. 0. 0. 0. 0. 0.]]
```

However when I turns to the case with my customized urdf: 
```
import os

import numpy as np
import pinocchio as pin

urdf_filename = "/home/ubuntu/catkin_ws/src/my_rrr/urdf/arm.urdf"; 
model, collision_model, visual_model = pin.buildModelsFromUrdf(urdf_filename) 
data = model.createData()
controlled_frame = model.getFrameId("endEffector"); 
q = np.array([1.0,1.0,1.0])
v = np.array([0.0,0.0,0.0])

pin.framesForwardKinematics(model,data,q)
pin.computeJointJacobians(model,data)
J = pin.getFrameJacobian(model, data, controlled_frame, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED)
print(J)
pin.computeJointJacobiansTimeVariation(model,data,q,v)
Jt = pin.getJointJacobianTimeVariation(model, data, controlled_frame, pin.ReferenceFrame.LOCAL_WORLD_ALIGNED)
print(Jt)
```
it's still gives me a segfault
```
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for boost::shared_ptr<hpp::fcl::CollisionGeometry> already registered; second conversion method ignored.
[[ 0.03992233  0.07171423 -0.06975558]
 [ 0.06206657 -0.04612787  0.04486803]
 [ 0.          0.07379737  0.18199181]
 [ 0.         -0.54097222 -0.54097222]
 [ 0.         -0.84104046 -0.84104046]
 [ 1.          0.          0.        ]]
Segmentation fault (core dumped)
```
and @wxmerkt  it's true that when I looked at the controlled_frame, it's greater than njoints and less than nframes. Do you have any suggestion for fixing this? Thanks!

hh2564, 2022-10-13:
I'm using 
```
controlled_frame = model.getFrameId("endEffector"); 
```

jcarpent, 2022-10-14:
> @wxmerkt it's true that when I looked at the controlled_frame, it's greater than njoints and less than nframes. Do you have any suggestion for fixing this? Thanks!

This means that your "endEffector" does not exist in your model (you can use `model.existFrame`).

hh2564, 2022-10-14:
> > @wxmerkt it's true that when I looked at the controlled_frame, it's greater than njoints and less than nframes. Do you have any suggestion for fixing this? Thanks!
> 
> This means that your "endEffector" does not exist in your model (you can use `model.existFrame`).
In my urdf the "endEffector" is connecting to one of the existing frame with a fixed joint. Is there any way I can add the frame manually to the model, or is there anyway I can modifty the urdf to make the function buildModelsFromUrdf being able to identify this frame? Thanks a lot!

jcarpent, 2022-10-15:
You can have the method `model.addFrame` to perform what you need. Please refer to the doc or the cheat sheet for more information which are referenced in the readme.

closed 2022-10-15 by jcarpent