stack-of-tasks/pinocchio #1768: Add cone support in meshcat visualizer

whtqh, 2022-10-15:
Cone shape is supported in HPP-FCL, however, the cone is replaced by a cylinder in [ Meshcat-python](https://github.com/rdeits/meshcat-python/blob/master/src/meshcat/geometry.py)
We can still transform cone shape to cylinder in [meshcat_visualizer.py](https://github.com/stack-of-tasks/pinocchio/blob/2a4953eca3886d7cfcc3037c6944fe2b892d70fb/bindings/python/pinocchio/visualize/meshcat_visualizer.py#L235) by add the code below:
```
elif isinstance(geom, hppfcl.Cone): 
    obj = meshcat.geometry.Cylinder(2. * geom.halfLength, 0, 0.0, geom.radius)
```

Hope it will be updated in the near future, and more basic shapes like arrow and line will be supported, too.

Thanks!

jcarpent, 2022-10-15:
Could you open a PR and provides the requested feature?
On our side, we have limited resources to handle such kind of request.

Thanks in advanced for your contribution

cmastalli, 2022-10-15:
We have implemented the display of cones in Crocoddyl (using Meshcat). This PR is under review, however, I could point out to the code if this helps you.

jcarpent, 2022-10-15:
@cmastalli Why did you put that in Croco and not Pinocchio?

jcarpent, 2022-10-15:
@cmastalli Could you transfer it to Pinocchio then?

whtqh, 2022-10-15:
haha~Thanks a lot! I am amazed at such a quick response and it's my pleasure to make any tiny contribution.
Btw, I haven't implemented an arrow shape cause I am still a fresh bird for pinocchio~
Pinocchio and Crocoddyl are both great repos to learn something!

cmastalli, 2022-10-15:
> @cmastalli Why did you put that in Croco and not Pinocchio?

We have to take friction cone information via Crocoddyl API. I believe it could potentially remove it with the new contact API from Pinocchio v3.

We also have implemented other display options that could be potentially moved to Pinocchio, e.g., swing trajectory, contact forces. Note that we have done this for both: GV and Meshcat.

cmastalli, 2022-10-15:
> @cmastalli Could you transfer it to Pinocchio then?

Could you let me know where I should include this? (I could try to find a time for doing)

cmastalli, 2022-10-15:
> Haha, thanks a lot! I am amazed at such a quick response and it's my pleasure to make any tiny contribution.
> Btw, I haven't implemented an arrow shape cause I am still a fresh bird for pinocchio~
> Pinocchio and Crocoddyl are both great repos to learn something!

You might be faster than me as I have limited time. This is how we do it in Crocoddyl. Here, we create a cone:
https://github.com/cmastalli/crocoddyl/blob/dbe58bd83c8acb6c8835ca6d0925f5a933eec4c8/bindings/python/crocoddyl/__init__.py#L383-L393. And then, we update its information here: https://github.com/cmastalli/crocoddyl/blob/dbe58bd83c8acb6c8835ca6d0925f5a933eec4c8/bindings/python/crocoddyl/__init__.py#L354-L361.

I am not familiar with the Pinocchio API. But, it shouldn't be that hard to do it in Pinocchio.

whtqh, 2022-10-15:
> > @cmastalli Could you transfer it to Pinocchio then?
> 
> Could you let me know where I should include this? (I could try to find a time for doing)

I personally think that cone shape is easy to add(2 lines should be ok), but actually I want to realize an arrow(just like what in rviz) in meshcat, so I need some time to find a proper way, too.

jcarpent, 2022-10-15:
@ManifoldFR Do you have such tools in your toolbox?

ManifoldFR, 2022-10-15:
> @ManifoldFR Do you have such tools in your toolbox?

I don't, though it would be good to add

whtqh, 2022-10-15:
Copy display-shapes.py to display-shapes-meshcat.py with a tiny changes, and the result looks like ↓
[image]
And a real arrow should be another problem, thus not discussed here anymore.

jcarpent, 2022-10-15:
So, seems to work properly, no?

whtqh, 2022-10-17:
> So, seems to work properly, no?

I think cone could be resolved by my latest PR, arrow should be added later.

closed 2022-10-17 by whtqh