stack-of-tasks/pinocchio #1771: Check that joint and link names are unique

wxmerkt, 2022-10-18:
We recently noticed the assertion in model.hxx triggering that indicates that we have two equally named frames:
https://github.com/stack-of-tasks/pinocchio/blob/1127321ea77e99876eb3cb8293847f785f41880d/src/multibody/model.hxx#L264-L266

The issue came from a URDF that had the name of a joint match the name of a link - which I believe is valid under the URDF specification which specifies that name of a joint has to be unique and that the name of a link has to be unique -- but not that a link may not be named the same as a joint is. In contrast, when constructing the Pinocchio model, frames for both the joint and the link are named the same way.

**Suggestion:** Raise a warning / error when building the model -- especially since the assertion is not available when Pinocchio or the Python bindings are compiled in Release mode.

wxmerkt, 2022-10-18:
I am quite busy right now but might get a chance at some point next week

jcarpent, 2022-10-18:
Perfect. If this is not urgent for you, it can wait next week ;)

jcarpent, 2022-11-01:
Solved via #1781.

PR #1781 (2022-11-01): Elevate check for ambiguous input argument to an exception
A sufficient solution is to elevate the assertion to an exception 

Fixes #1771
closed 2022-11-01 by jcarpent