stack-of-tasks/pinocchio #1773: Creating C++ methods with Python binding

Gregwar, 2022-10-19:
Hello,

I am writing C++ methods that I would like to bind to Python (using Boost Python).
Is there a way to make argument types `pinocchio.SE3` converted automatically from numpy, the same way it works with Eigen types calling `eigenpy::enableEigenPy();` ?

Gregwar, 2022-10-19:
Ex. if I do:

```C++
void some_func(pinocchio::SE3 arg);
```

And bind it:
```C++
def("some_func", &something::some_func);
```

I get:
```python
In [3]: something.some_func(np.eye(4))
---------------------------------------------------------------------------
ArgumentError                             Traceback (most recent call last)
<ipython-input-3-2362f16be1bf> in <module>
----> 1 something.some_func(np.eye(4))

ArgumentError: Python argument types in
    something.some_func(numpy.ndarray)
did not match C++ signature:
    some_func(pinocchio::SE3Tpl<double, 0>)
```

While I see that it works with:
```python
In [5]: pin.log6(np.eye(4))
Out[5]: 
  v = 0 0 0
  w = 0 0 0
```

Gregwar, 2022-10-19:
Ok, I see!
I missed that you exposed `log6` with a `Matrix4d` as well, I thought it was some automatic conversions done by Boost bindings

Thanks

jcarpent, 2022-10-19:
> Ok, I see! I missed that you exposed `log6` with a `Matrix4d` as well, I thought it was some automatic conversions done by Boost bindings
> 
> Thanks

We could add it, but it might be dangerous.

jmirabel, 2022-10-20:
You can also wrap the function in Python. Depending on what you want to do, it is sometimes easier.