stack-of-tasks/pinocchio #1778: Python 3.11 compatability

JonathanKuelz, 2022-10-25:
Dear pinocchio-Team,

just today, Python 3.11 was released -- and it seems there have been updates to PyBoost, s.t. the pinocchio-bindings are not longer compatible. Steps to reproduce:
```
$ conda create --name tmp python=3.11
$ conda activate tmp
$ pip install pin
$ python
>>> import pinocchio
```
raises
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jonathan/anaconda3/envs/tmp/lib/python3.11/site-packages/cmeel.prefix/lib/python3.11/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
SystemError: type Boost.Python.enum has the Py_TPFLAGS_HAVE_GC flag but has no traverse function
```

just wanted to report this issue - keeping an older Python version is no problem, but of course it would be nice to being able to use pinocchio with the latest Python version again.

jcarpent, 2022-10-25:
Thanks @JonathanKuelz, for reporting this issue.
We will wait for Conda updates to properly update all the packages related to Pinocchio for Python 3.11.
It is an automatic process normally ;).

nim65s, 2022-10-25:
Thanks for the report. This issue is independent of conda, as I can reproduce it with this Dockerfile:

```Dockerfile
FROM python:3.11

RUN python -m pip install -U pip
RUN python -m pip install pin
RUN python -c "import pinocchio"
```

I'll try to fix this this week.

jcarpent, 2022-10-25:
Seems indeed related to https://github.com/boostorg/python/pull/385

jcarpent, 2022-10-29:
We are making progress on this line of work with https://github.com/conda-forge/boost-feedstock/pull/156.

nim65s, 2022-11-02:
Thanks @jcarpent for these refs. I applied that in https://github.com/cmake-wheel/cmeel-boost/commit/d9c611e99cea23abe270b5dec2daceeffcd007ae, and it looks like this is fixed now. 

Can you confirm for your side, @JonathanKuelz, and close the issue ?

closed 2022-11-06 by jcarpent

JonathanKuelz, 2022-11-07:
Sorry for the late replay, I can also confirm it working in my environment again now - thanks for the fast fix!