stack-of-tasks/pinocchio #1779: Question about external force and impedance control

Betancourt20, 2022-10-25:
Hi,

First of all, thank you so much for the framework. It is really interesting.
I'm trying to implement an simple impedance control but my force seems to be in another frame, which I don't understand in which one. I'm using the kinova Jaco as a manipulator. The manipulator moves but not as defined in the documentation of the robot. I'm pretty sure that I'm not doing in the best way the transformation of the frames in order to have the good one.
```
pin.forwardKinematics(model,data,q)
pin.updateFramePlacements(model,data)
oMdes = data.oMf[model.getFrameId("j2s6s200_end_effector")]
wrench = np.array([nx,ny,nz,nx,fy,fz])
wrench_ = pin.Force.Zero()
wrench_.linear = np.array([fx,fy,fz])
wrench_.angular = np.array([nx,ny,nz])
ft = wrench_

for k in range(model.njoints):
    faux = pin.Force.Zero()
    if k == (model.njoints-1):
        faux.linear = ft.linear
        faux.angular =ft.angular
    fext.append(faux)
```
Then, I have the simulation part with the control algorithm
```
for k in range(N):
  t += dt
  tic = time.time()
  pin.forwardKinematics(model,data,q)
  pin.updateFramePlacements(model,data)
  pin.computeJointJacobians(model, data, q)
  J=pin.getFrameJacobian(model,data,frameIndex,pin.LOCAL)
  
  xdot = J@qdot
  velerror = -xdot
  
  u = Dd@velerror + wrench
  a =  J.T@(np.linalg.pinv(Md)@u)   
  tau = pin.rnea(model, data, q, qdot, a,fext)
  qddot = pin.aba(model, data, q, qdot, tau, fext)
  
  qdot += qddot * dt 
  q = pin.integrate(model,q,qdot*dt)
  
  toc = time.time() 
  ellapsed = toc - tic
  dt_sleep = max(0,dt - (ellapsed))
  time.sleep(dt_sleep)
```
I know that the frames
```
oMdes= data.oMf[model.getFrameId("j2s6s200_end_effector")]
oMdes = data.oMi[jointIndex] # corresponds to the parent of the end_effector frame
```
are differents but it is not clear to me from where to where I need to transform the force and based on which idea. I tried to follow the same ideas as described in https://fr.mathworks.com/help/robotics/ug/compute-joint-torques-to-balance-an-endpoint-force-and-moment.html but I did not succeed. 
Any hint ? Thanks in advance ;)

jcarpent, 2022-10-25:
`model.frames[model.getFrameId("j2s6s200_end_effector")].placement` contains the relative placement between the LOCAL frame associated to the Frame itself, and its parent joint.
In your case, `J=pin.getFrameJacobian(model,data,frameIndex,pin.LOCAL)` is already taking things into account, meaning that the Jacobian is already expressed in the LOCAL frame of the Frame.
The question is in which frame wrench is expressed in your case?

You could also simplify your code with:
```python
for k in range(N):
  t += dt
  tic = time.time()
  pin.forwardKinematics(model,data,q)
  pin.updateFramePlacements(model,data)
  pin.computeJointJacobians(model, data, q)
  J=pin.getFrameJacobian(model,data,frameIndex,pin.LOCAL)
  
  xdot = J@qdot
  velerror = -xdot
  
  u = Dd@velerror + wrench
  a =  J.T@(np.linalg.pinv(Md)@u)   
  
  qdot += a * dt 
  q = pin.integrate(model,q,qdot*dt)
  
  toc = time.time() 
  ellapsed = toc - tic
  dt_sleep = max(0,dt - (ellapsed))
  time.sleep(dt_sleep)
```

Betancourt20, 2022-10-25:
Thanks a lot for your fast answer and for the simplification of the code. You are totally right.
Well, following the ideas of the impedance control, I believe the wrench should be described in the end-effector frame. I implemented this control on MATLAB and the results that I'm looking for are the following: If I remember well, the force (different from the wrench) control was described in the global frame (or inertial frame). 
Thanks in advance for your help.

jcarpent, 2022-10-25:
So, if your `wrench` quantity is expressed in the LOCAL frame, nothing needs to be changed.
Otherwise, you can use `oMdes= data.oMf[model.getFrameId("j2s6s200_end_effector")].actInv(wrench)` to transform a wrench expressed at the origin of the world aligned with the world coordinate systems to the LOCAL frame of the Frame.

I hope this answer helps you. Could we close this issue then?

closed 2022-10-25 by Betancourt20