stack-of-tasks/pinocchio #1783: import pinocchio seg fault with ros-humble-pinocchio 2.6.11-1jammy

Benjamin-Tan, 2022-11-04:
Hi, I have installed ros-humble-pinocchio on the humble base docker image. However, as i tried to import pinocchio in a python script, it results in segmentation fault.

The following are the gdb backtrace logs with import pinocchio.
```
Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x0000555555676958 in _Py_Dealloc ()
(gdb) bt
#0  0x0000555555676958 in _Py_Dealloc ()
#1  0x00007fff73df46bf in ?? () from /opt/ros/humble/lib/python3.10/site-packages/pinocchio/pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so
#2  0x00007fff73dea756 in pinocchio::python::exposeSE3() () from /opt/ros/humble/lib/python3.10/site-packages/pinocchio/pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so
#3  0x00007fff73b4871d in init_module_pinocchio_pywrap() () from /opt/ros/humble/lib/python3.10/site-packages/pinocchio/pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so
#4  0x00007fff72de56e3 in boost::python::handle_exception_impl(boost::function0<void>) () from /usr/lib/x86_64-linux-gnu/libboost_python310.so.1.74.0
#5  0x00007fff72de6c69 in boost::python::detail::init_module(PyModuleDef&, void (*)()) () from /usr/lib/x86_64-linux-gnu/libboost_python310.so.1.74.0
```

I have install ros-humble-eigenpy, ros-humble-hpp-fcl. import eigenpy and import hpp-fcl has no issue.
Anyone has any idea how to resolve this?

jcarpent, 2022-11-04:
Hi @Benjamin-Tan,

Could you provide a docker image to reproduce the bug you have?
Do you have a unique version of Pinocchio installed?

@wxmerkt Have you been able to reproduce this bug on your side?

Best,
Justin

Benjamin-Tan, 2022-11-04:
@jcarpent, below is the docker image to reproduce the bug. It is only the ros humble version of pinocchio is installed.

```
FROM ros:humble

RUN apt-get update && apt-get upgrade -y && apt-get install -y \
      ros-humble-pinocchio && \
    rm -rf /var/lib/apt/lists/*

CMD ["bash"]
```

wxmerkt, 2022-11-04:
Thank you very much for reporting the issue and the Dockerfile to reproduce @Benjamin-Tan. I can reproduce the issue but do not know the source of it - and I dont have the bandwidth to debug right now. @jcarpent, the change introduced to make it build on humble was to add the `-g1` flag to reduce the amount of debug symbols. Otherwise it's a standard build with clang.

jcarpent, 2022-11-04:
I can debug if you can provide me with the command lines to build Pinocchio from source in the docker image ...

wxmerkt, 2022-11-04:
This should be: 
```
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_INSTALL_LIBDIR=lib/x86_64-linux-gnu -DCMAKE_INSTALL_PREFIX=/opt/ros/rolling -DCMAKE_PREFIX_PATH=/opt/ros/rolling ..
```

Note that we have extra arguments exported in the `ENV` such that the compile commands show (per [this log](https://build.ros2.org/job/Rbin_uJ64__pinocchio__ubuntu_jammy_amd64__binary/101/consoleFull)):
```
Building CXX object src/CMakeFiles/pinocchio.dir/parsers/urdf/model.cpp.o
22:25:59 cd /tmp/binarydeb/ros-rolling-pinocchio-2.6.11/.obj-x86_64-linux-gnu/src && /usr/lib/ccache/clang++ -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DBOOST_MPL_LIMIT_VECTOR_SIZE=30 -DHPP_FCL_HAS_OCTOMAP -DHPP_FCL_HAVE_OCTOMAP -DOCTOMAP_MAJOR_VERSION=1 -DOCTOMAP_MINOR_VERSION=9 -DOCTOMAP_PATCH_VERSION=8 -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_URDFDOM_USE_STD_SHARED_PTR -DPINOCCHIO_WITH_HPP_FCL -DPINOCCHIO_WITH_URDFDOM -Dpinocchio_EXPORTS -I/tmp/binarydeb/ros-rolling-pinocchio-2.6.11/.obj-x86_64-linux-gnu -I/tmp/binarydeb/ros-rolling-pinocchio-2.6.11/.obj-x86_64-linux-gnu/include -I/tmp/binarydeb/ros-rolling-pinocchio-2.6.11/include -isystem /usr/include/eigen3 -isystem /opt/ros/rolling/include -g -O2 -ffile-prefix-map=/tmp/binarydeb/ros-rolling-pinocchio-2.6.11=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -g1 -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -std=gnu++11 -MD -MT src/CMakeFiles/pinocchio.dir/parsers/urdf/model.cpp.o -MF CMakeFiles/pinocchio.dir/parsers/urdf/model.cpp.o.d -o CMakeFiles/pinocchio.dir/parsers/urdf/model.cpp.o -c /tmp/binarydeb/ros-rolling-pinocchio-2.6.11/src/parsers/urdf/model.cpp
22:26:10 clang: warning: optimization flag '-ffat-lto-objects' is not supported [-Wignored-optimization-argument]
```

Note that EigenPy and HPP-FCL are built with g++ whereas Pinocchio is built with clang. This has not been an issue in the past on other distros, but may factor in here.

Note, the regular `devel` builds pass all tests fine, so the extra arguments used during packaging might have to do with it: https://build.ros2.org/job/Rdev__pinocchio__ubuntu_jammy_amd64/30/consoleFull

The extra arguments are specified here: https://github.com/ros2-gbp/pinocchio-release/blob/0f3019446ec4fe2d0bc42284ffa863b1015b0893/debian/rules.em

jcarpent, 2022-11-04:
@olivier-stasse Could you try on your side the building from source?

jcarpent, 2022-11-05:
I've been able to identify clang as the faulty guy.
@wxmerkt Would it be possible to switch back to GCC?

wxmerkt, 2022-11-05:
We could try, but are likely to hit out of memory again - the reason for switching gcc to clang was for (a) reduce out of memory compilation issues and (b) higher performance of Pinocchio when compiled with clang. Is there a workaround for the issue with clang or could we change the clang version used? Do you know what the issue is / is there an upstream bug report?

jcarpent, 2022-11-06:
@wxmerkt I've found the bug, which is a bit hard to diagnosis, but #1784 has been tested successfully on the docker image.
Could you make a patch on it, or maybe I can make a new release?

Let me know what is the best solution for you.

wxmerkt, 2022-11-06:
Great find, although it's hard to follow why this would have caused a bug with clang. Thank you for creating the release - I'll release them now to the buildfarm.

jcarpent, 2022-11-06:
Perfect. Thanks for your help.
Yes, this is a bizarre bug which occurs sometimes.

jcarpent, 2022-11-09:
Should be fixed now.

Benjamin-Tan, 2022-11-09:
@wxmerkt @jcarpent really appreciate for the quick update and fix for such a bug! Thanks a lot! 

I’m wondering how did you manage to find this bug?

Benjamin-Tan, 2022-11-24:
@wxmerkt @jcarpent After the latest ros humble sync today, using the ros-humble-pinocchio 2.6.12-1jammy, the segfault issue is still there. Does this happen on your side too?

```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007fe02c4fc25b in __cxa_begin_catch () from /lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007fe02c4fc25b in __cxa_begin_catch () from /lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007fe02d4b5e86 in ?? () from /opt/ros/humble/lib/python3.10/site-packages/pinocchio/pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so
#2  0x00007fe02d9117d0 in ?? () from /opt/ros/humble/lib/python3.10/site-packages/pinocchio/pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so
#3  0x00007fe02d90c49b in ?? () from /opt/ros/humble/lib/python3.10/site-packages/pinocchio/pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so
#4  0x00007fe02d790310 in init_module_pinocchio_pywrap() () from /opt/ros/humble/lib/python3.10/site-packages/pinocchio/pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so
#5  0x00007fe02ca086e3 in boost::python::handle_exception_impl(boost::function0<void>) () from /lib/x86_64-linux-gnu/libboost_python310.so.1.74.0
#6  0x00007fe02ca09c69 in boost::python::detail::init_module(PyModuleDef&, void (*)()) () from /lib/x86_64-linux-gnu/libboost_python310.so.1.74.0
```

wxmerkt, 2022-11-24:
Hi @Benjamin-Tan,
Thank you for the update - and sorry to hear the issue is still present. I can reproduce it with the Docker image but don't know the source of it. @jcarpent do you have any insights on where it might be coming from?

To reproduce:
```
docker pull ros:humble
docker run -it --rm ros:humble /bin/bash
```

Inside the Docker image:
```
apt update && apt install -y ros-humble-pinocchio gdb
gdb -ex run --args python3 -c "import pinocchio"
```

jcarpent, 2022-11-24:
Could you try using conda and rostack instead?
If you need just Python bindings, it might be sufficient.

jcarpent, 2022-11-24:
@wxmerkt What is the version of clang used for the build?

wxmerkt, 2022-11-24:
Per [the last build log](https://build.ros2.org/job/Hbin_uJ64__pinocchio__ubuntu_jammy_amd64__binary/lastBuild/consoleFull) it is `clang: 1:14.0-55~exp2`, so clang-14?

jcarpent, 2022-11-24:
For me, this docker image works:
```docker
FROM ros:humble

RUN apt-get update && apt-get upgrade -y && apt-get install -y \
      ros-humble-pinocchio && \
    rm -rf /var/lib/apt/lists/*

RUN apt-get update
RUN apt-get install -y python3-pip build-essential cmake git clang gdb
RUN pip install ipython
RUN cd /tmp && git clone https://github.com/stack-of-tasks/pinocchio.git --recursive
RUN mkdir -p /tmp/pinocchio/build_gcc
RUN cd /tmp/pinocchio/build_gcc && echo $(pwd) && cmake .. -DCMAKE_INSTALL_PREFIX=/opt/ros/humble -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_INSTALL_LIBDIR=lib/x86_64-linux-gnu -DCMAKE_INSTALL_PREFIX=/opt/ros/humble -DCMAKE_PREFIX_PATH=/opt/ros/humble -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=OFF ..
RUN cd /tmp/pinocchio/build_gcc && make -j3
RUN mkdir -p /tmp/pinocchio/build_clang
RUN cd /tmp/pinocchio/build_clang && CXX=clang cmake -DCMAKE_INSTALL_PREFIX=/opt/ros/humble -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_INSTALL_LIBDIR=lib/x86_64-linux-gnu -DCMAKE_INSTALL_PREFIX=/opt/ros/humble -DCMAKE_PREFIX_PATH=/opt/ros/humble -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=OFF ..
RUN cd /tmp/pinocchio/build_clang && make -j3
RUN apt-get install -y vim

CMD ["bash"]
```
When you do: `cd /tmp/pinocchio/build_clang/bindings/python && python -c "import pinocchio"` it works like a charm

jcarpent, 2022-11-24:
@wxmerkt Could get into the build process of ROS at some point?
Meaning I would like to reproduce exactly what happens on ROS buildfarm similarly to what conda provides when building feedstocks?

jcarpent, 2022-11-24:
It is weird, because this is not the same error message as before!

Before:

> Hi, I have installed ros-humble-pinocchio on the humble base docker image. However, as i tried to import pinocchio in a python script, it results in segmentation fault.
> 
> The following are the gdb backtrace logs with import pinocchio.
> 
> ```
> Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
> 0x0000555555676958 in _Py_Dealloc ()
> (gdb) bt
> #0  0x0000555555676958 in _Py_Dealloc ()
> #1  0x00007fff73df46bf in ?? () from /opt/ros/humble/lib/python3.10/site-packages/pinocchio/pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so
> #2  0x00007fff73dea756 in pinocchio::python::exposeSE3() () from /opt/ros/humble/lib/python3.10/site-packages/pinocchio/pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so
> #3  0x00007fff73b4871d in init_module_pinocchio_pywrap() () from /opt/ros/humble/lib/python3.10/site-packages/pinocchio/pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so
> #4  0x00007fff72de56e3 in boost::python::handle_exception_impl(boost::function0<void>) () from /usr/lib/x86_64-linux-gnu/libboost_python310.so.1.74.0
> #5  0x00007fff72de6c69 in boost::python::detail::init_module(PyModuleDef&, void (*)()) () from /usr/lib/x86_64-linux-gnu/libboost_python310.so.1.74.0
> ```
> 
> I have installed ros-humble-eigenpy, ros-humble-hpp-fcl. import eigenpy and import hpp-fcl has no issue. Anyone has any idea how to resolve this?

Now:

> @wxmerkt @jcarpent After the latest ros humble sync today, using the ros-humble-pinocchio 2.6.12-1jammy, the segfault issue is still there. Does this happen on your side too?
>
> ```
> Program terminated [...]

jcarpent, 2022-11-27:
@wxmerkt @Benjamin-Tan Could you confirm that the docker file that I've provided works fine on your side?

Benjamin-Tan, 2022-11-28:
@jcarpent I just tried the docker file that you have provided, I can confirm it is working fine.

wxmerkt, 2022-11-28:
@jcarpent Many thanks for your investigations and the Docker image. For the build steps, I would extract the arguments from the build log. Additionally, the following settings are applied:
```
export LDFLAGS=-Wl,-s
export DEB_CXXFLAGS_MAINT_APPEND=-DNDEBUG -g1
```

I'm quite busy right now but could look into this further after December 7th

Benjamin-Tan, 2023-01-28:
@wxmerkt for your information, I notice from the build logs that the unit tests related to python bindings are failing due to segfault. However, the build still passes even with failing tests. 

https://build.ros2.org/job/Hbin_uJ64__pinocchio__ubuntu_jammy_amd64__binary/71/consoleFull

wxmerkt, 2023-01-28:
Thank you for the analysis @Benjamin-Tan - the corresponding log entry gives some more insight:
```
09:13:41  24/128 Test  #24: test-cpp-srdf ......................................   Passed    0.34 sec
09:13:41 test 25
09:13:41         Start  25: test-cpp-python_parser
09:13:41 
09:13:41 25: Test command: /tmp/binarydeb/ros-humble-pinocchio-2.6.14/.obj-x86_64-linux-gnu/unittest/test-cpp-python_parser
09:13:41 25: Environment variables: 
09:13:41 25:  PYTHONPATH=/tmp/binarydeb/ros-humble-pinocchio-2.6.14/.obj-x86_64-linux-gnu/bindings/python:/opt/ros/humble/lib/python3.10/site-packages:/tmp/ros_buildfarm:
09:13:41 25: Test timeout computed to be: 1500
09:13:41 25: Running 1 test case...
09:13:41 25: unknown location(0): [4;31;49mfatal error: in "python_parserTest/buildModel": memory access violation at address: 0x00000000: no mapping at fault address[0;39;49m
09:13:41 25: ./unittest/python_parser.cpp(15): [1;36;49mlast checkpoint: "buildModel" test entry[0;39;49m
09:13:41 25: 
09:13:41 25: [1;31;49m*** 1 failure is detected in the test module "python_parserTest"
09:13:41 25: [0;39;49m
09:13:41  25/128 Test  #25: test-cpp-python_parser .............................***Failed    0.21 sec
09:13:41 Running 1 test case...
09:13:41 unknown location(0): [4;31;49mfatal error: in "python_parserTest/buildModel": memory access violation at address: 0x00000000: no mapping at fault address[0;39;49m
09:13:41 ./unittest/python_parser.cpp(15): [1;36;49mlast checkpoint: "buildModel" test entry[0;39;49m
09:13:41 
09:13:41 [1;31;49m*** 1 failure is detected in the test module "python_parserTest"
09:13:41 [0;39;49m
09:13:41 
09:13:41 test 26
09:13:41         Start  26: test-cpp-all-joints
```

Unfortunately, I do not have the bandwidth to debug these issues right now

Benjamin-Tan, 2023-01-29:
@wxmerkt @jcarpent I might have found the culprit that is causing a reproducible segmentation fault, which is the enabling of link-time optimization. I simply add `CFLAGS=-flto=auto CXXFLAGS=-flto=auto` just before the cmake command. The segfault happens only for clang but not gcc build. This flag seems to be introduced by default in dpkg-buildflags as mentioned in  https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1324388618

The following is the gdb backtrace while performing `gdb -ex run --args python3 -c 'import pinocchio'` with clang build. Hope this finding could help in troubleshooting this issue.

```
Starting program: /usr/bin/python3 -c import\ pinocchio
warning: Error disabling address space randomization: Operation not permitted
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007faa2ee4f25b in __cxa_begin_catch () from /lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007faa2ee4f25b in __cxa_begin_catch () from /lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007faa2fa9ca76 in __clang_call_terminate () from /tmp/pinocchio/build_clang/bindings/python/pinocchio/pinocchio_pywrap.cpython-310-x86_64-linux-gnu.so
#2  0x00007faa2fec7d30 in boost::python::api::object_base::~object_base (this=0x7fffecfa4128) at /usr/include/boost/python/object_core.hpp:423
#3  boost::python::detail::name_space_def<bool (*)(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&), boost::python::default_call_policies, boost::python::class_<pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>, boost::python::detail::not_specified, boost::python::detail::not_specified, boost::python::detail::not_specified> > (name_space=..., kw={...}, name=<optimized out>, f=<optimized out>, policies=..., doc=<optimized out>) at /usr/include/boost/python/detail/defaults_def.hpp:60
#4  boost::python::detail::define_stub_function<0>::define<pinocchio::python::existFrame_overload::non_void_return_type::gen<boost::mpl::vector4<bool, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::FrameType const&> >, boost::python::default_call_policies, boost::python::class_<pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>, boost::python::detail::not_specified, boost::python::detail::not_specified, boost::python::detail::not_specified> > (kw={...}, name_space=..., name=<optimized out>, policies=..., 
    doc=<optimized out>) at /usr/include/boost/python/detail/defaults_def.hpp:281
#5  boost::python::detail::define_with_defaults_helper<0>::def<pinocchio::python::existFrame_overload::non_void_return_type::gen<boost::mpl::vector4<bool, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::FrameType const&> >, boost::python::default_call_policies, boost::python::class_<pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>, boost::python::detail::not_specified, boost::python::detail::not_specified, boost::python::detail::not_specified> > (kw={...}, name_space=..., name=<optimized out>, stubs=..., policies=..., 
    doc=<optimized out>) at /usr/include/boost/python/detail/defaults_def.hpp:196
#6  boost::python::detail::define_with_defaults_helper<1>::def<pinocchio::python::existFrame_overload::non_void_return_type::gen<boost::mpl::vector4<bool, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::FrameType const&> >, boost::python::default_call_policies, boost::python::class_<pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>, boost::python::detail::not_specified, boost::python::detail::not_specified, boost::python::detail::not_specified> > (kw={...}, name_space=..., name=<optimized out>, stubs=..., policies=..., 
    doc=<optimized out>) at /usr/include/boost/python/detail/defaults_def.hpp:178
#7  boost::python::detail::define_with_defaults<pinocchio::python::existFrame_overload, boost::python::class_<pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>, boost::python::detail::not_specified, boost::python::detail::not_specified, boost::python::detail::not_specified>, boost::mpl::vector4<bool, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::FrameType const&> > (name_space=..., name=<optimized out>, overloads=...) at /usr/include/boost/python/detail/defaults_def.hpp:251
#8  boost::python::class_<pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>, boost::python::detail::not_specified, boost::python::detail::not_specified, boost::python::detail::not_specified>::def_maybe_overloads<pinocchio::python::existFrame_overload, bool (pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>::*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::FrameType const&) const> (this=0x7fffecfa45c8, sig=<optimized out>, name=<optimized out>, overloads=...) at /usr/include/boost/python/class.hpp:551 [... cut ...]
#61 0x0000556e5e1bbc35 in Py_RunMain ()
#62 0x0000556e5e191bbd in Py_BytesMain ()
#63 0x00007faa32279d90 in __libc_start_call_main (main=main@entry=0x556e5e191b80, argc=argc@entry=3, argv=argv@entry=0x7fffecfa6a68) at ../sysdeps/nptl/libc_start_call_main.h:58
#64 0x00007faa32279e40 in __libc_start_main_impl (main=0x556e5e191b80, argc=3, argv=0x7fffecfa6a68, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffecfa6a58) at ../csu/libc-start.c:392
#65 0x0000556e5e191ab5 in _start ()
```

wxmerkt, 2023-01-30:
Thank you very much for the investigation and pointer @Benjamin-Tan. I've made a change to the release flags and re-released (https://github.com/ros/rosdistro/pull/35982) - once merged we should be able to verify whether this fixed it

wxmerkt, 2023-02-01:
I can confirm that this is now fixed with the binaries available on the testing repository for `rolling`. The same modifications for `humble` will land with https://github.com/ros/rosdistro/pull/36013. I believe this should be resolved now. Many thanks for the debugging and pointers @Benjamin-Tan

closed 2023-02-01 by wxmerkt