stack-of-tasks/pinocchio #1788: pin3.6.12 fails to import

JonathanKuelz, 2022-11-10:
Dear pinocchio team,

since the latest update to `pin2.6.12` on PyPi, it fails to import with the following message:
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.10/site-packages/cmeel.prefix/lib/python3.10/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .pinocchio_pywrap import *
ImportError: liboctomap.so.1.9: cannot open shared object file: No such file or directory
```

The error can be reproduced with the following Dockerfile:
```
FROM python:3.10

RUN python -m pip install -U pip
RUN python -m pip install pin
RUN python -c "import pinocchio"
```
and `docker build .`

nim65s, 2022-11-10:
HI,

This package is currently beeing rebuild: https://github.com/cmake-wheel/pinocchio/actions/runs/3435153904

JonathanKuelz, 2022-11-10:
Hello,

thanks for the fast reply -- just wanted to correct myself: It seems to be a general problem atm, not pin2.6.12 - related. I will wait for the rebuild then.

JonathanKuelz, 2022-11-10:
The new build seems to be pushed to PyPi, but unfortunately I still get the same error.

nim65s, 2022-11-10:
Thanks for the report. I wasn't expecting another issue, but we have the same in hpp-fcl. It is rebuilding in https://github.com/cmake-wheel/hpp-fcl/actions/runs/3439607126 ; I hope we won't have to re-rebuild pinocchio, but I'm not sure yet.

JonathanKuelz, 2022-11-11:
Problem seems to be resolved, thanks a lot for the quick support!

closed 2022-11-11 by JonathanKuelz