stack-of-tasks/pinocchio #1794: pinocchio 2.9.1 build failure on osx

chenrui333, 2022-11-22:
👋 trying to build the latest release, but run into some build issue. The error log is as below:

<details>
<summary>test error</summary>

```
  /tmp/pinocchio-20221120-71841-1cqh73s/pinocchio-2.9.1/build/include/pinocchio/multibody/geometry.hxx:47:34: error: no matching conversion for functional-style cast from 'const pinocchio::GeometryObject::CollisionGeometryPtr' (aka 'const shared_ptr<fcl::CollisionGeometry>') to 'fcl::CollisionObject'
        collisionObjects.push_back(fcl::CollisionObject(geom_object.geometry));
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  In file included from /tmp/pinocchio-20221120-71841-1cqh73s/pinocchio-2.9.1/src/parsers/urdf/geometry.cpp:5:
  In file included from /tmp/pinocchio-20221120-71841-1cqh73s/pinocchio-2.9.1/build/include/pinocchio/parsers/urdf.hpp:10:
  In file included from /tmp/pinocchio-20221120-71841-1cqh73s/pinocchio-2.9.1/build/include/pinocchio/multibody/geometry.hpp:448:
  /tmp/pinocchio-20221120-71841-1cqh73s/pinocchio-2.9.1/build/include/pinocchio/multibody/geometry.hxx:47:34: error: no matching conversion for functional-style cast from 'const pinocchio::GeometryObject::CollisionGeometryPtr' (aka 'const shared_ptr<fcl::CollisionGeometry>') to 'fcl::CollisionObject'
        collisionObjects.push_back(fcl::CollisionObject(geom_object.geometry));
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

</details>

full build log, https://github.com/Homebrew/homebrew-core/actions/runs/3509282025/jobs/5878306372
relates to Homebrew/homebrew-core#116238

jcarpent, 2022-11-22:
@chenrui333 Thanks for raising this issue.
Unfortunately, we do not yet support 2.9.1 for deployment.

chenrui333, 2022-11-22:
Interesting, I forgot how I got that release. Closing the issue for now as it is not the release list.

closed 2022-11-22 by chenrui333