stack-of-tasks/pinocchio #1795: End-Effector Velocity in Bullet vs Pinocchio

Berk-Tosun, 2022-11-23:
Hello,

I have a similar issue like https://github.com/stack-of-tasks/pinocchio/issues/1759: I have a simulation in PyBullet and I am computing same terms in Pinocchio to check.

I have a system with a floating base (red), revolute joint and a fixed joint for the feet (blue):
[image]
I have created it by modifying a double pendulum urdf. The urdf is attached:
[acrobot_floating.zip](https://github.com/stack-of-tasks/pinocchio/files/10073915/acrobot_floating.zip)

The feet or end-effector velocities from pinocchio do not match those from pybullet. However, the end-effector positions (both linear and angular) and angular velocities match perfectly. 

The first two plots in the left side show the linear velocity of the end effector in the world frame, in them we see the the velocity mismatch. The other two plots in the left side show that the end effector's linear position and angular velocity match. The right side plots shows end-effector angular position as a quaternion, I have plotted each component. A regular simulation looks like following:
[image]

I have noticed the deviation is caused by the angular velocity of the link. The deviation increases with the angular velocity. If I simulate for example a free fall, nothing rotates and everything matches:
[image]

I get the frame velocity as following:
```python
...

self.robot.forwardKinematics(q, v)
pin.computeJointJacobians(self.robot.model, self.robot.data)
pin.updateFramePlacements(self.robot.model, self.robot.data)

...

vel = pin.getFrameVelocity(
            self.robot.model,
            self.robot.data,
            self.frame_id_ee_com,
            pin.ReferenceFrame.LOCAL_WORLD_ALIGNED,
        ).linear.copy()
```

Please note, I am using a frame attached to the center of mass of the feet. As explained in other issues, pinocchio replaces fixed joints with frames. I have noticed the frame corresponding to the feet was not placed as I expected by pinocchio. Pinocchio did not take the offset to CoM into account. So I added a new frame while creating the model:

```python
len_link1 = 1.0
radius_feet = 0.1
len_com = len_link1 + radius_feet
len_tip = len_link1 + 2 * radius_feet

frame_name_ee_com = "PointFootCoM"
frame_ee_com = pin.Frame(
    name=frame_name_ee_com,
    parent_joint=2,
    parent_frame=6,
    placement=pin.SE3(
        rotation=np.eye(3),
        translation=np.array([0, 0, -len_com]),
    ),
    type=pin.FrameType.BODY,
)
self.frame_id_ee_com = self.robot.model.addFrame(frame_ee_com)

self.robot.data = self.robot.model.createData()
```

Could you please tell me what I am missing here? Sorry I could not simplify the script to share it here at the moment.

jcarpent, 2022-11-23:
Bullet uses a reference frame at the centre of mass of the links, while Pinocchio uses for a Frame the placement wrt the parent joint supporting the frame in question. 
You can add frames in Pinocchio to match the convention of Bullet, as you have started to do.

You can recover the inertia of the foot expressed in the joint frame as `I = model.inertias[2]`
And use `I.lever` as the offset to position the frame wrt the supporting parent joint frame.

Is it clear?

Berk-Tosun, 2022-11-23:
Not clear.

Just to be on the same page, regarding this simple robot:
  red link: thigh
  green link: calf
  blue link: feet

Thigh is the floating base. I believe pinocchio treats calf + feet as a single link since they are connected by a fixed_joint. When I use `I.lever` as you specify, it gives me the CoM offset for the combined link. Since feet has very little inertia, it is very close to middle of the calf.

I am interested in the CoM velocity of the feet. Bullet treats it as a link, and all the velocities I have shown so far belong to the feet CoM. How can I get this? I think the way I placed the frame is conceptually the same as you describe already. 

I believe, I have already verified the `translation` of the frame is correct as shown by the plots above. I think there is a problem with the rotation that I do not understand.

Thanks for the quick reply!

jcarpent, 2022-11-23:
So, you can recover the inertia of the foot in model.frames[6].inertia. This inertia is expressed in the frame of the Frame itself. You need to do `model.frames[6].placement.act(model.frames[6].inertia)` to get it expressed in the parent joint frame. From that, you should be able to get what you need.

jcarpent, 2022-11-23:
For the orientation, it depends in which frame you aim to express the quantity. In the current settings, the axes of the frame are aligned with the parent joint frame.

Berk-Tosun, 2022-11-23:
Sorry, I do not fully follow. How do I use the inertia expressed in the joint frame, to get the my target, the (feet) classical frame velocity in world coordinates?

jcarpent, 2022-11-23:
I think we have misunderstandings. 
Could tell me what is not working on your side (shortly)? Why do you want to follow the Bullet convention (which is not the one used for robot controlà?

Berk-Tosun, 2022-11-23:
Of course.

I do not want to follow the Bullet convention, I use it as my simulation environment at the moment. I have traced the problems in my controller to the mismatch in the end-effector velocities/accelerations. I am just trying to make sure Bullet and Pinocchio agree.

---

My problem is that the linear velocity of the foot expressed in the world frame is wrong.

I am running a simulation in which the feet slides in the x-direction and never leaves the ground. Pinocchio does not agree with Bullet, the groundtruth:
[image]

Since the foot never leaves the ground and moves only in the x-direction, the linear velocity in z-direction should be zero for the whole duration. Again bullet is the groundtruth and as expected it is zero at all times. Pinocchio does not agree again.
[image]

However, some other quantities match perfectly: Feet position, Feet orientation and Feet angular velocity all in world frame. You can check my first post for the plots over time.

jcarpent, 2022-11-23:
How do you compute the feed vel?

Berk-Tosun, 2022-11-23:
self.robot.forwardKinematics(q, v)
pin.computeJointJacobians(self.robot.model, self.robot.data)
pin.updateFramePlacements(self.robot.model, self.robot.data)

...

vel = pin.getFrameVelocity(
            self.robot.model,
            self.robot.data,
            self.frame_id_ee_com,
            pin.ReferenceFrame.LOCAL_WORLD_ALIGNED,
        ).linear.copy()

jcarpent, 2022-11-23:
The len_com used does not rever to the com position of the foot.
You should use the spatial inertia of the fixed joint stored in frame.inertia field.
In the inertia, the com position is stored in the `lever` quantity. This translation should then be expressed wrt the parent joint frame (`frame.placement.act(lever)`)

Berk-Tosun, 2022-11-23:
Is this what you meant:

```python        
frame_id_ankle_joint = self.robot.model.getFrameId("AnkleJoint")
frame_fixed_joint = self.robot.model.frames[frame_id_ankle_joint]
frame_parent = self.robot.model.frames[frame_fixed_joint.parent]
translation_com = frame_parent.placement.act(
    frame_fixed_joint.inertia.lever  # = [0, 0, -0.1]
)  # same value returns, = [0, 0, -0.1]

frame_name_ee_com = "PointFootCoM"
frame_ee_com = pin.Frame(
    name=frame_name_ee_com,
    parent_joint=2,
    parent_frame=6,
    placement=pin.SE3(
        rotation=np.eye(3),
        translation=translation_com,
    ),
    type=pin.FrameType.BODY,
)
self.frame_id_ee_com = self.robot.    type=pin.FrameType.BODY,
)
self.frame_id_ee_com = self.robot.model.addFrame(frame_ee_com)
model.addFrame(frame_ee_com)
```
I keep running the same simulation (feet slides in x-direction, no change in height). Still velocities disagree, with the modification now the feet positions disagree as well.

[image]

Thanks a lot for the support

jcarpent, 2022-11-23:
For your info, I've never mentioned this line in the previous comments: `frame_parent = self.robot.model.frames[frame_fixed_joint.parent]`
This line is not in line with my previous comments.

jcarpent, 2022-11-23:
```
frame_id_ankle_joint = self.robot.model.getFrameId("AnkleJoint")
frame_fixed_joint = self.robot.model.frames[frame_id_ankle_joint]
frame_parent = self.robot.model.frames[frame_fixed_joint.parent]
translation_com = frame_parent.placement.act(
    frame_fixed_joint.inertia.lever  # = [0, 0, -0.1]
)  # same value returns, = [0, 0, -0.1]
```
should be replaced by:
```
frame_id_ankle_joint = self.robot.model.getFrameId("AnkleJoint")
frame_fixed_joint = self.robot.model.frames[frame_id_ankle_joint]
translation_com = frame_fixed_joint.placement.act(
    frame_fixed_joint.inertia.lever  # = [0, 0, -0.1]
)  # same value returns, = [0, 0, -0.1]
```

Berk-Tosun, 2022-11-23:
Bullet's official doc: https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.3v8gjd1epcrt

It is not stated in the doc very clearly; you can have a look on this issue too: https://github.com/bulletphysics/bullet3/issues/2429

jcarpent, 2022-11-23:
So, use the link velocity expressed at the joint level directly for the Bullet side.
In the case of Pinocchio, this means at the frame level.

jcarpent, 2022-11-27:
I will close this issue as there is no more activity on it.

closed 2022-11-27 by jcarpent

Berk-Tosun, 2022-11-27:
I could not fully locate the source of the issue. I believe the problem is due to faulty urdf, and/or bullet and pinocchio do not agree on how to parse it.

Anyway, thanks for the support.

For future readers, my recommendation: getJointStates in bullet to match directly with pinocchio.

jcarpent, 2022-11-27:
Thanks @Berk-Tosun for the clear notification. Should be helpful for people.